espressif/esp-idf #2234: truncate function not defined for SPIFFS filesystem

davctv, 2018-07-27:
Hello everybody,

truncate() function seems to be implemented for spiffs components. If called, it returns -1
and errno is 88 which description in "\esp-idf\components\newlib\include\sys\errno.h" is 

`#define ENOSYS 88	/* Function not implemented */`

Is it a problem of ESP-IDF or is it a problem of  SPIFFS component itself?

As a reference I link a similar issue which is solved (for FATfs):
[(https://github.com/espressif/esp-idf/issues/1980)]

Thanks

negativekelvin, 2018-07-28:
https://github.com/pellepl/spiffs/issues/107

davctv, 2018-07-28:
Thanks @negativekelvin,

Do you suggest me to try implement the function and then pull request on esp-idf or to pull request to spiffs native repository (pelleel/spiffs)?

closed 2022-03-22 by espressif-bot