espressif/esp-idf #2235: ADC question

nene8, 2018-07-27:
Hi
i want to read a voltage between 0-3.3v.
I’m setting in this way the adc1 converter:

  adc_gpio_init(ADC_UNIT_1,ADC1_CHANNEL_6);
adc1_config_width(ADC_WIDTH_BIT_10);
adc1_config_channel_atten(ADC1_CHANNEL_6,ADC_ATTEN_DB_11);
  adc_power_on();

in a function called every 10 seconds i do:

   u16 val = adc1_get_raw(ADC1_CHANNEL_6);
   printf("ADC value :%d",val);

I read a different value, expecially I set on the GPIO34 500mV, and I read 118 (about 0.380V)
(3.3/1024)*118=0.380V

The reading should be 155…
I don't know why
thanks in advance

krzychb, 2018-07-27:
Hi  @nene8,

The "raw" value is a result of comparison of the attenuated voltage against internal ESP32 ADC voltage reference. Per design the ADC reference voltage is 1100mV, however the true reference voltage can range from 1000mV to 1200mV amongst different ESP32s.

Check [ADC Calibration](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html#adc-calibration) how to convert raw reading into real voltage taking into account true reference voltage value in your particular ESP32.

nene8, 2018-07-28:
Thanks for the answer, i have already try to insert the adc calibration but with the same result. Can you post a example that read a voltage between 0 to 3.3v ?? Many thanks

krzychb, 2018-07-29:
Try slightly updated [peripherals/adc](https://github.com/espressif/esp-idf/blob/561884b62fac66fab29a8c9619556ddd27577d7b/examples/peripherals/adc/main/adc1_example_main.c) example below.
1. Compile and upload it to an ESP32 board
2. Apply some voltage in range from 0 to 3.9V (or from 150 to 2450mV to obtain more accurate results) between GND and GPIO34
3. Check the voltage reading by the application
4. Measure V_REF between GND and GPIO25
5. Enter measured value in line `#define DEFAULT_VREF` instead of `1100`, that represents1100mV
6. If measured V_REF is different than default 1100mV, then voltage reading by the application will be closer to the actual value applied between GND and GPIO34

```c
/* ADC1 Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_adc_cal.h"

#define DEFAULT_VREF            1100     //Use adc2_vref_to_gpio() to obtain a better estimate
#define VREF_OUT_GPIO   (GPIO_NUM_25)    //Check Vref at this GPIO pin
#define NO_OF_SAMPLES             64     //Multisampling

static esp_adc_cal_characteristics_t *adc_chars;
static const adc_channel_t channel = ADC_CHANNEL_6;     //GPIO34 if ADC1, GPIO14 if ADC2
static const adc_atten_t atten = ADC_ATTEN_DB_11;       //11dB attenuation, gives full-scale voltage of 3.9V
                                                        //see note in adc1_config_channel_atten() docs
static const adc_unit_t unit = ADC_UNIT_1;

static void check_efuse()
{
    //Check TP is burned into eFuse
    if (esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_TP) == ESP_OK) {
        printf("eFuse Two Point: Supported\n");
    } else {
        printf("eFuse Two Point: NOT supported\n");
    }

    //Check Vref is burned into eFuse
    if (esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_VREF) == ESP_OK) {
        printf("eFuse Vref: Supported\n");
    } else {
        printf("eFuse Vref: NOT supported\n");
    }
}

static void print_char_val_type(esp_adc_cal_value_t val_type)
{
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP) {
        printf("Characterized using Two Point Value\n");
    } else if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        printf("Characterized using eFuse Vref\n");
    } else {
        printf("Characterized using Default Vref\n");
    }
}

void app_main()
{
    //Check if Two Point or Vref are burned into eFuse
    check_efuse();

    //Route Vref to VREF_OUT_GPIO
    esp_err_t status = adc2_vref_to_gpio(VREF_OUT_GPIO);
    if (status == ESP_OK) {
        printf("Vref routed to GPIO%d\n", VREF_OUT_GPIO);
    } else {
        printf("Failed to route Vref\n");
    }

    //Configure ADC
    if (unit == ADC_UNIT_1) {
        adc1_config_width(ADC_WIDTH_BIT_12);
        adc1_config_channel_atten(channel, atten);
    } else {
        adc2_config_channel_atten((adc2_channel_t)channel, atten);
    }

    //Characterize ADC
    adc_chars = calloc(1, sizeof(esp_adc_cal_characteristics_t));
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(unit, atten, ADC_WIDTH_BIT_12, DEFAULT_VREF, adc_chars);
    print_char_val_type(val_type);

    //Continuously sample ADC1
    while (1) {
        uint32_t adc_reading = 0;
        //Multisampling
        for (int i = 0; i < NO_OF_SAMPLES; i++) {
            if (unit == ADC_UNIT_1) {
                adc_reading += adc1_get_raw((adc1_channel_t)channel);
            } else {
                int raw;
                adc2_get_raw((adc2_channel_t)channel, ADC_WIDTH_BIT_12, &raw);
                adc_reading += raw;
            }
        }
        adc_reading /= NO_OF_SAMPLES;
        //Convert adc_reading to voltage in mV
        uint32_t voltage = esp_adc_cal_raw_to_voltage(adc_reading, adc_chars);
        printf("Raw: %d\tVoltage: %dmV\n", adc_reading, voltage);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

closed 2018-08-20 by FayeY