espressif/esp-idf #2238: can bus driver speed issue

snahmad, 2018-07-28:
Have you tested using this latest can driver.

https://github.com/espressif/esp-idf/bl ... iver/can.c

I am reading can frame using task/thread every 10 ms. I am losing frames time to time. it cannot cope high traffic on can bus.

Dazza0, 2018-07-30:
What is your CAN driver configuration? Could you provide a code sample?

I would also suggest you use the [alert feature](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/can.html#alerts) to check if you are loosing frames due to a full RX queue (enable `CAN_ALERT_RX_QUEUE_FULL` )

snahmad, 2018-08-02:
ok, will try out. thanks.

snahmad, 2018-08-08:
Thanks for asking. I am busy doing some other stuff. I will come back to can bus.
Do you have any full example to enable alter alert feature of CAN bus. Do you have experience of receiving can bus packets and send them over TCP/UDP socket.

Dazza0, 2018-08-17:
See the [docs code snippets](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/can.html#reconfiguring-and-reading-alerts) for how to configure and use the CAN alerts.

For an application example using alerts, see the [alert and recovery example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/can/can_alert_and_recovery).

snahmad, 2018-08-17:
Thanks. I will try out.

snahmad, 2018-08-22:
Receive is fine. but 
esp_err_t esp_err = can_transmit(&send_message, pdMS_TO_TICKS(0));

gets timeout even after few reties.

snahmad, 2018-11-30:
Hi,
no more issue. close it