espressif/esp-idf #2242: read and ioctl decreases the speed

shervinkh88, 2018-07-30:
## Environment

- Development Kit:      [ESP32-Wrover]
- IDF version 3.0.2
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Operating System:		Ubuntu
- Power Supply:         external 3.3V

## Problem Description
hi
I want to send and receive fixed 8KBps with iwip with a timer, in the timer I have a variable that by making it one I know that I have to send data. And I create a task that continuously check the variable and if it's 1 it sends the data, this works perfect without a problem but if I add read() or ioctl(socket, FIONREAD, &FDReadCount); in another task or in the main app source the speed decreases to half or even less, internet speed is high and also one reason that shows that it's not related to speed is ioctl alone decreases the speed even if there's no data to read
if i have read or write in the timer I get this error :
Interrupt wdt timeout on CPU0
here's the code for read task:

```cpp
	while(1)
	{
		vTaskDelay(50/portTICK_RATE_MS);
		if(ConnectedToServer==1){
			if(ReadFromServerDone==0)
			{
				ioctl(s, FIONREAD, &FDReadCount);
				if(FDReadCount>0)
				{
					read(s, write_raw_arr, 1);
					ReadFromServerDone=1;
				}
			}
		}
}
```
code for timer:
```cpp
static void example_tg0_timer_init(int timer_idx, 
    bool auto_reload, double timer_interval_sec)
{
    /* Select and initialize basic parameters of the timer */
    timer_config_t config;
    config.divider = TIMER_DIVIDER;
    config.counter_dir = TIMER_COUNT_UP;
    config.counter_en = TIMER_PAUSE;
    config.alarm_en = TIMER_ALARM_EN;
    config.intr_type = TIMER_INTR_LEVEL;
    config.auto_reload = auto_reload;
    timer_init(TIMER_GROUP_0, timer_idx, &config);

    /* Timer's counter will initially start from value below.
       Also, if auto_reload is set, this value will be automatically reload on alarm */
    timer_set_counter_value(TIMER_GROUP_0, timer_idx, 0x00000000ULL);

    /* Configure the alarm value and the interrupt on alarm. */
    timer_set_alarm_value(TIMER_GROUP_0, timer_idx, timer_interval_sec * TIMER_SCALE);
    timer_enable_intr(TIMER_GROUP_0, timer_idx);
    timer_isr_register(TIMER_GROUP_0, timer_idx, timer_group0_isr, 
        (void *) timer_idx, NULL, NULL);

    timer_start(TIMER_GROUP_0, timer_idx);
}

void timer_group0_isr(void *para)
{
    uint64_t timer_counter_value = 
        ((uint64_t) TIMERG0.hw_timer[TIMER_1].cnt_high) << 32
        | TIMERG0.hw_timer[TIMER_1].cnt_low;
		
	timer_event_t evt;
    evt.timer_group = 0;
    evt.timer_idx = TIMER_1;
    evt.timer_counter_value = timer_counter_value;
	evt.type = TEST_WITH_RELOAD;
	TIMERG0.int_clr_timers.t1 = 1;

	TIMERG0.hw_timer[TIMER_1].config.alarm_en = TIMER_ALARM_EN;

	if(ConnectedToServer==1 && SendingToServer==0)
	{
		SendingToServer=1;
		towrite_arr[0]=*MyData*;
	}
}

```
and code for write :

```cpp
void SendToServerTask(void* arg)
{
	while(1){
			if(SendingToServer==1)
			{
				if (write(s, read_raw_arr, 1) < 0) {
					ConnectedToServer=0;
					close(s);
					ESP_LOGE(TAG, "... socket send failed");
				}
				SendingToServer=0;
			}
	}
}

```
Code for Socket Initialization:
```cpp
	StartConnecting:
	BufferCount=0;
	while(1){
		const struct addrinfo hints = {
			.ai_family = AF_INET,
			.ai_socktype = SOCK_STREAM,
		};
		struct addrinfo *res;
		struct in_addr *addr;

		int err = getaddrinfo(Address, Port, &hints, &res);

		if(err != 0 || res == NULL) {
			ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
			vTaskDelay(1000 / portTICK_PERIOD_MS);
			continue;
		}
		
        addr = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
        ESP_LOGI(TAG, "DNS lookup succeeded. IP=%s", inet_ntoa(*addr));

        s = socket(res->ai_family, res->ai_socktype, 0);
        if(s < 0) {
            ESP_LOGE(TAG, "... Failed to allocate socket.");
            freeaddrinfo(res);
            vTaskDelay(1000 / portTICK_PERIOD_MS);
            continue;
        }
        ESP_LOGI(TAG, "... allocated socket");

        if(connect(s, res->ai_addr, res->ai_addrlen) != 0) {
            ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
            close(s);
            freeaddrinfo(res);
            vTaskDelay(4000 / portTICK_PERIOD_MS);
            continue;
        }

        ESP_LOGI(TAG, "... connected");
        freeaddrinfo(res);

        if (write(s, "A", 1) < 0) {
            ESP_LOGE(TAG, "... socket send failed");
            close(s);
            vTaskDelay(4000 / portTICK_PERIOD_MS);
            continue;
        }
        break;
	}
	
		
	ConnectedToServer=1;

```

Is this a bug or I have to do it in another way
thx

shervinkh88, 2018-07-31:
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHFREQ="80m"
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
CONFIG_FREERTOS_HZ=1000

shervinkh88, 2018-08-03:
I changed CONFIG_TCP_WND_DEFAULT=5744 to 20000 still problem exists with even ioctl no read function, waiting for an answer, thanks

Alvin1Zhang, 2019-06-13:
@shervinkh88 Sorry for getting back to you so late. Thanks for reporting the issue, would you help share if any updates for this issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2019-08-21:
@shervinkh88  Thanks for reporting the issue, will close this due to short of feedback, feel free to reopen if the issue still exists. Thanks.

closed 2019-08-21 by Alvin1Zhang