espressif/esp-idf #2247: Need High Speed SPI

kovachjk, 2018-07-31:
I have a ESP32-WROOM-32U on a board of my design. I have the need to acquire a large amount of data at a fast date rate.  200KB-1MB @ minimum 100KSmpl/sec. I need this because I am building device for monitoring ultrasonic vibrations on machines.  I am using a 16 bit ADC.
This data will also needs to be streamed to an open file on an SD Card.
So I need two high speed SPI transactions.  I am also concerned with my understanding that the DMA only handles 32 bit data.  After an acquisition would I need to read out 32 bit data from the SD card and write it back on as 16 bit data?
To avoid any high speed SD Card issues, I had planned on writing to an area of flash that had been erased, but I understand that has issues and is slow also.
I am confused as to why a transaction can only be used once?
Is it because it is being copied or somehow mangled by the RTOS?
I have no need to share any time with any task while I am acquiring and writing data.
Do I have to use queues or mutexes to use the SPI?
Why can't I just run and stay in a task with a high priority and use for loops and a timer?
Most SPI's are simple. Set the send buf, run the SCLK and read the recv buf.  Why is this so complex?
Why aren't there a set of low level functions that allow simple access to the SPI  and other peripherals that can run completely inside a task or without a RTOS?
My last question is, should I even try with this platform?
I really hope this is possible, I really like the product and support. Plus I have already done much design work. This SPI issue caught me by surprise.

michprev, 2018-08-01:
Hi @kovachjk 
So to sum this up you are reading some data from external 16 bit ADC over SPI and then writing it on an SD card over SPI again, right?

You can take a look at https://github.com/espressif/esp-idf/issues/2233

When you need running a time critical task I recommend you doing following:

- make sure you are running wifi task on core 0 (Component config -> Wi-Fi -> WiFi Task Core ID in sdkconfig)

- pin your time critical task to core 1, all other tasks should be pinned to core 0

- as a temporal workaround change https://github.com/espressif/esp-idf/blob/1c7a8b3b712c4020562551f692d0ced8f7470d2c/components/lwip/port/freertos/sys_arch.c#L432 to `result = xTaskCreatePinnedToCore(thread, name, stacksize, arg, prio, &CreatedTask, 0);`

In case this does not help:
Can you provide a small example of your code with notes about expected behavior and behavior you are experiencing?

Even though SPI driver uses FreeRTOS it is reasonably fast.

kovachjk, 2018-08-01:
When you say that the FreeRTOS is reasonably fast.  I would need it to respond to a timer interrupt and run the code within 10uSec.
Your reference to 2233 is a little over my head.  This was why my question was focused on staying in a task until all my data was on the SD card, so that I might avoid the overhead of tasks, mutexes, semaphores.......
I am confused, if I  understand properly, that  you level 20 task is being preempted.  This is also a concern for me.  During my acquisition( up and over 1 sec) I cannot be interrupted or delayed, this would cause an anomaly in my data analysis.
Is ProjectGus saying that an ISR is preempting it?
I am relatively new to both the ESP32 and FreeRTOS, plus my core knowledge for the past 3 years has been Javascript so I have many holes in my knowledge.
I did not even know that the ESP32 could use threads or why they should.
My application serves a single page web app then listens for commands on a websocket connection, so the LWIP does not need to be available all the time.  Your code showing a simple web and socket server intrigues  me.
The user requests an operation, in this case a large waveform, and waits for the results. 
This, I hoped, would give the ESP32 much flexibility to focus on only getting and saving ADC data.
My only concern is loosing the webserver and websocket connections if I suspended the PRO core.
My question was about the SPI transaction object and I cannot just reuse it in a loop in a task without interrupted.  It looks to me like a strut holding SPI resister values.  Why does the RTOS need to be involved.
I am not sure I have the knowledge to implement this, I just want to understand it enough to help me find the right person to hire.

kovachjk, 2018-08-01:
Hi Michprev,
We have already implemented a ADS8866 16 bit ADC which work fine. It's just I need more faster.  
I am still waiting for someone to explain why I need to task switch or use mutexes to read and write to a SPI peripheral inside a high level task.
I guess I will need to implement a board with small ARM chip, ADC and flash IC with two SPI ports to buffer an acquisition, then write it back to the ESP32 using a UART, which will suck.

kovachjk, 2018-08-01:
To be clear to use an ADC I have to toggle the chip select line high every 10uSec.  This is considered a SPI transaction which takes 25uSec.   I need to acquire 16 bit every transaction. At 25uSec/transaction I am limited to less than 40KSamples/ sec or around 600KB/sec.  I would be interested in what your SPI data rate is without RTOS.

kovachjk, 2018-08-02:
I finally found the file with the mutexes. 
I assume this this the one I will need to learn of find someone who knows how to implement the SPI functions there without all the mutexes.

closed 2018-08-02 by kovachjk