espressif/esp-idf #2248: Posix time functions break compilation.

fake-name, 2018-08-01:
I'm attempting to update the micropython project to build against the latest micropython.

This fails with 

```
/home/durr/esp/esp-idf/components/newlib/time.c: In function 'clock_gettime':
/home/durr/esp/esp-idf/components/newlib/time.c:425:14: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
         case CLOCK_MONOTONIC:
              ^
/home/durr/esp/esp-idf/components/newlib/time.c:425:14: note: each undeclared identifier is reported only once for each function it appears in

```

Following this through, it appears that these functions were added in https://github.com/espressif/esp-idf/commit/e274940a46fb01a9421df1ceebe37293d61b6a2f. `CLOCK_MONOTONIC` is only defined if `_POSIX_MONOTONIC_CLOCK` is defined. If `_POSIX_MONOTONIC_CLOCK` is not defined, 

`_POSIX_MONOTONIC_CLOCK` seems to only be defined ever if either `__rtems__` is defined, or if `__CYGWIN__` is defined.

Neither of these are true in my case, and as a result, the file is functionally impossible to compile.

I'm not sure how this should be fixed, but it certainly seems like an issue. @KonstantinKondrashov?

igrr, 2018-08-01:
@fake-name please post the full gcc command line used to compile time.c.

fake-name, 2018-08-01:
Note that reverting e274940a46fb01a9421df1ceebe37293d61b6a2f appears to fix the issue,

fake-name, 2018-08-01:
Full command line: 

```
xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/durr/esp/esp-idf/components/bootloader_support/include -I/home/durr/esp/esp-idf/components/bootloader_support/include_bootloader -I/home/durr/esp/esp-idf/components/driver/include -I/home/durr/esp/esp-idf/components/driver/include/driver -I/home/durr/esp/esp-idf/components/nghttp/port/include -I/home/durr/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/durr/esp/esp-idf/components/esp32/include -I/home/durr/esp/esp-idf/components/soc/include -I/home/durr/esp/esp-idf/components/soc/esp32/include -I/home/durr/esp/esp-idf/components/ethernet/include -I/home/durr/esp/esp-idf/components/expat/include/expat -I/home/durr/esp/esp-idf/components/expat/port/include -I/home/durr/esp/esp-idf/components/heap/include -I/home/durr/esp/esp-idf/components/json/include -I/home/durr/esp/esp-idf/components/json/port/include -I/home/durr/esp/esp-idf/components/log/include -I/home/durr/esp/esp-idf/components/newlib/include -I/home/durr/esp/esp-idf/components/nvs_flash/include -I/home/durr/esp/esp-idf/components/freertos/include -I/home/durr/esp/esp-idf/components/tcpip_adapter/include -I/home/durr/esp/esp-idf/components/lwip/include/lwip -I/home/durr/esp/esp-idf/components/lwip/include/lwip/port -I/home/durr/esp/esp-idf/components/lwip/include/lwip/posix -I/home/durr/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/durr/esp/esp-idf/components/mbedtls/port/include -I/home/durr/esp/esp-idf/components/spi_flash/include -I/home/durr/esp/esp-idf/components/ulp/include -I/home/durr/esp/esp-idf/components/vfs/include -I/home/durr/esp/esp-idf/components/newlib/platform_include -I/home/durr/esp/esp-idf/components/xtensa-debug-module/include -I/home/durr/esp/esp-idf/components/wpa_supplicant/include -I/home/durr/esp/esp-idf/components/wpa_supplicant/port/include -I/home/durr/esp/esp-idf/components/ethernet/include -I/home/durr/esp/esp-idf/components/app_trace/include -I/home/durr/esp/esp-idf/components/app_update/include -I/home/durr/esp/esp-idf/components/pthread/include -I/home/durr/esp/esp-idf/components/smartconfig_ack/include -DIDF_VER=\"v3.2-dev-362-g1c7a8b3b\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_BTREE=1 -c -MD -o build//home/durr/esp/esp-idf/components/newlib/select.o /home/durr/esp/esp-idf/components/newlib/select.c
/home/durr/esp/esp-idf/components/newlib/time.c: In function 'clock_gettime':
/home/durr/esp/esp-idf/components/newlib/time.c:425:14: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
         case CLOCK_MONOTONIC:
              ^
/home/durr/esp/esp-idf/components/newlib/time.c:425:14: note: each undeclared identifier is reported only once for each function it appears in
../../py/mkrules.mk:47: recipe for target 'build//home/durr/esp/esp-idf/components/newlib/time.o' failed
make: *** [build//home/durr/esp/esp-idf/components/newlib/time.o] Error 1
make: *** Waiting for unfinished jobs....

```

igrr, 2018-08-01:
Yes, it's missing platform_include and several other directories. Please build any example project with IDF (`make V=1`) and compare the command lines for time.c.

fake-name, 2018-08-01:
> In IDF, platform_include directory should precede newlib include directory on the compiler command line. Please make sure this is the case in micropython build system as well.

Ah, yep, that's the problem.

fake-name, 2018-08-01:
> Yes, it's missing platform_include and several other directories. Please build any example project with IDF (make V=1) and compare the command lines for time.c.

```
-I/home/durr/esp/esp-idf/components/newlib/platform_include 
```

Huh?

fake-name, 2018-08-01:
Ok, yeah, it was an include issue.

Sorry about that.

closed 2018-08-01 by fake-name