espressif/esp-idf #2250: config wont disable uart flow control

Arthedian, 2018-08-01:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version  - I dont know
- IDF version 561884b6
- Development Env: other - sublime text
- Operating System: debian
- Power Supply: USB

## Problem Description

I am trying to disable flow control, but when installing driver it still creates interrupt.

### Expected Behavior

It wont use RTS and CTS pin.

### Actual Behavior
Its creating interrupt for it 
### Steps to repropduce

1. install uart driver

### Code to reproduce this issue

```cpp
uart_config_t uart_config = {
.baud_rate = 115200,
.data_bits = UART_DATA_8_BITS,
.parity = UART_PARITY_DISABLE,
.stop_bits = UART_STOP_BITS_1,
.flow_ctrl = UART_HW_FLOWCTRL_DISABLE
};
uart_param_config(UART_NUM_2, &uart_config);

uart_set_pin(UART_NUM_2, 33, 35, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```

## Debug Logs

```
D (34355) intr_alloc: Connected src 36 to int 3 (cpu 1)
```

michprev, 2018-08-01:
UART driver will create interrupt no matter if you use flow control or not. See https://github.com/espressif/esp-idf/blob/1c7a8b3b712c4020562551f692d0ced8f7470d2c/components/driver/uart.c#L1351

Arthedian, 2018-08-01:
Why it is creating the interrupt? How can I change the pin for the interrupt?

michprev, 2018-08-01:
Interrupts are in no way connected with pins. Registering interrupts mean that once a special event occurs in a peripheral, CPU gets interrupted to care about it (for example copy data to an user buffer).

What is the exact problem you are experiencing?

Arthedian, 2018-08-01:
And what does the source mean?

negativekelvin, 2018-08-01:
https://github.com/espressif/esp-idf/blob/1c7a8b3b712c4020562551f692d0ced8f7470d2c/components/soc/esp32/include/soc/soc.h#L350

Arthedian, 2018-08-01:
Does that mean, that if I have uart2 enabled, I can't have anything on pin36?

negativekelvin, 2018-08-01:
No, interrupt numbers have nothing to do with pin numbers.  All gpio interrupts are combined together to make       
https://github.com/espressif/esp-idf/blob/1c7a8b3b712c4020562551f692d0ced8f7470d2c/components/soc/esp32/include/soc/soc.h#L336
Move beyond "Arduino thinking"...

koobest, 2018-08-06:
Hi,@Arthedian 
    The interrupt source number has nothing to do with the pin number of gpio. The CPU handles many interrupts. Each interrupt has a unique number. For example, interrupt 36 is the interrupt of UART2. All GPIOs share an interrupt, Its number is 22. so, You don't have to worry about UART2 interrupts affecting GPIO interrupts.

xiongyumail, 2018-08-27:
Hi @Arthedian 
The UART must turn on interrupts, but if you don't use flow control, the flow control pins can be used safely by other peripherals.

closed 2018-09-20 by Arthedian