espressif/esp-idf #2251: i2s in camera mode + DMA skips samples even at relatively low clock speed

easyvolts, 2018-08-01:
## Environment

- Development Kit:      WEMOS-LOLIN32
- Kit version: v1.0.0
- Core (if using chip or module): ESP-WROOM32
- IDF version: v3.1-dev-661-gf586f5e6
- Development Env:		Arduino IDE
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

i2s in parallel camera mode with DMA skips(looses) some samples.
i2s is configured in parallel camera mode with DMA and gets WS signal with clock speed 20 MHz. Sampling is continuous with data stored in a circular buffer, the digital signal generator provides 3-bit continuously increasing signal (pattern 0, 1, 2, 3, 4, 5, 6, 7, 0, 1, 2....). Data accumulated in the DMA buffer looks like 0,2,2,3,3,5,5,7,7,0... Looks like every second 
[image]
i2s sample is repeated twice in the data buffer. According to the documentation i2s should work with clock up to 40MHz.

### Expected Behavior
WS signal with clock speed up to 40 MHz should result in proper capturing of all data from the parallel bus input. Data accumulated in the DMA buffer should contain all samples without skipped samples (according to the documentation i2s should work with clock up to 40MHz).
### Actual Behavior
Even at WS clock speed 20MHz not all samples from the parallel input bus are captured and collected in RAM buffer, every second sample is the same as previous instead of real data.

### Steps to reproduce

1. flash Arduino scetch from attachment.
2. attach digitag 3-bit wide parallel signal that changes with 20MHz speed to pins 18,23,19. Connect 20MHz clock to pin 15.
3. run the firmware and check output in serial port (115200 8N1). Data in printed buffer should be an exact decimal representation of the sequence provided to the parallel input bus of the i2s, but in my tests it indicates skipped/lost samples.

### Code to reproduce this issue

```cpp
#include "soc/soc.h"
#include "soc/gpio_sig_map.h"
#include "soc/i2s_reg.h"
#include "soc/i2s_struct.h"
#include "soc/io_mux_reg.h"
#include "driver/gpio.h"
#include "driver/periph_ctrl.h"
#include "driver/i2s.h"
#include "rom/lldesc.h"

DMA_ATTR uint32_t dma_buff[2][100];
DMA_ATTR lldesc_t dma_descriptor[2];

void setup() {
    Serial.begin(115200);
    Serial.println(ESP.getSdkVersion());
  
    //buff 0
    dma_descriptor[0].length = 0; //number of byte written to the buffer
    dma_descriptor[0].size = sizeof(dma_buff[0]); //max size of the buffer in bytes
    dma_descriptor[0].owner = 1;
    dma_descriptor[0].sosf = 1;
    dma_descriptor[0].buf = (uint8_t*)&dma_buff[0][0];
    dma_descriptor[0].offset = 0;
    dma_descriptor[0].empty = 0;
    dma_descriptor[0].eof = 0;
    //pointer to the next descriptor
    dma_descriptor[0].qe.stqe_next = &dma_descriptor[1];

    //buff 1
    dma_descriptor[1].length = 0; //number of byte written to the buffer
    dma_descriptor[1].size = sizeof(dma_buff[1]); //max size of the buffer in bytes
    dma_descriptor[1].owner = 1;
    dma_descriptor[1].sosf = 1;
    dma_descriptor[1].buf = (uint8_t*)&dma_buff[1][0];
    dma_descriptor[1].offset = 0;
    dma_descriptor[1].empty = 0;
    dma_descriptor[1].eof = 0;
    //pointer to the next descriptor
    dma_descriptor[1].qe.stqe_next = &dma_descriptor[0];

    //data inputs
    pinMode(18, INPUT);
    gpio_matrix_in(18,    I2S0I_DATA_IN0_IDX, false);
    pinMode(23, INPUT);
    gpio_matrix_in(23,    I2S0I_DATA_IN1_IDX, false);
    pinMode(19, INPUT);
    gpio_matrix_in(19,    I2S0I_DATA_IN2_IDX, false);
    //for i2s in parallel camera input mode data is receiver only when V_SYNC = H_SYNC = H_ENABLE = 1. We don't use these inputs so simply set them High
    gpio_matrix_in(0x38, I2S0I_V_SYNC_IDX, false);
    gpio_matrix_in(0x38, I2S0I_H_SYNC_IDX, false);  //0x30 sends 0, 0x38 sends 1
    gpio_matrix_in(0x38, I2S0I_H_ENABLE_IDX, false);

    pinMode(15, INPUT);
    gpio_matrix_in(15/*XCLK*/, I2S0I_WS_IN_IDX, false);
    
    // Enable and configure I2S peripheral
    periph_module_enable(PERIPH_I2S0_MODULE);

    // Enable slave mode (sampling clock is external)
    I2S0.conf.rx_slave_mod = 1; 
    // Enable parallel mode
    I2S0.conf2.lcd_en = 1;

    // Use HSYNC/VSYNC/HREF to control sampling
    I2S0.conf2.camera_en = 1;
    
    // Configure clock divider
    I2S0.clkm_conf.clkm_div_a = 1;
    I2S0.clkm_conf.clkm_div_b = 0;
    I2S0.clkm_conf.clkm_div_num = 2;
    I2S0.sample_rate_conf.rx_bck_div_num = 2; //i2s clk is divided before reaches BCK output 
       
    // FIFO will sink data to DMA
    I2S0.fifo_conf.dscr_en = 1;
    I2S0.fifo_conf.tx_fifo_mod_force_en = 1;
    I2S0.fifo_conf.rx_fifo_mod_force_en = 1;
    // FIFO configuration
    //two bytes per dword packing
    I2S0.fifo_conf.rx_fifo_mod = 1;
    I2S0.conf_chan.rx_chan_mod = 1;
    // Clear flags which are used in I2S serial mode
    I2S0.sample_rate_conf.rx_bits_mod = 0;
    I2S0.conf.rx_right_first = 0;
    I2S0.conf.rx_msb_right = 0;
    I2S0.conf.rx_msb_shift = 0;
    I2S0.conf.rx_mono = 0;
    I2S0.conf.rx_short_sync = 0;
    I2S0.timing.val = 0;

    //start i2s
    I2S0.rx_eof_num = 0xFFFFFFFF;
    I2S0.in_link.addr = (uint32_t)&dma_descriptor[0];
    I2S0.in_link.start = 1;
    I2S0.int_clr.val = I2S0.int_raw.val;
    I2S0.int_ena.val = 0;

    //start i2s + dma
    I2S0.conf.rx_start = 1;

    delay(10000);

    //stop i2s + dma
    I2S0.conf.rx_start = 0;
    
    for(uint8_t i = 0; i<sizeof(dma_buff[0])/2; i++) {
      Serial.println(*(((uint16_t *)&dma_buff[0][0]) + i));
      Serial.print(' ');
      delay(10);
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```cpp

## Other items : 
The attached picture shows an irregular graph instead of an expected even ramp signal.

igrr, 2018-08-01:
@easyvolts IIRC  `rx_fifo_mod` needs to be set to zero at 20MHz frequency. There will be two samples per 32 bit word, 2nd sample of each word is the same as 1st sample of the next one (so you would get 00000001, 00010002, 00020003, 00030004 words in the buffer). Frequencies above 20MHz do not work in I2S mode.

Also, FYI, I am going to close this since this is not an esp-idf issue. Please use esp32.com forum or technical support section of espressif.com if you need help troubleshooting this further. Thanks.

closed 2018-08-01 by igrr

Gustavomurta, 2020-01-24:
Links of reference: 
**DMA descriptor’s linked lists** = [(https://github.com/espressif/arduino-esp32/blob/a59eafbc9dfa3ce818c110f996eebf68d755be24/tools/sdk/include/esp32/rom/lldesc.h)]

    **ESP32-cam driver** = [https://github.com/espressif/esp32-camera/blob/1b8fdfe335b3c8c8b9e091e3615b0aee16946ef0/driver/camera.c]

Gustavomurta, 2020-01-24:
Links of Reference:
**DMA descriptor’s linked lists**
**ESP32-cam driver**

Gustavomurta, 2020-01-24:
Links of reference:
[https://github.com/espressif/esp-idf/blob/9f3b550ff4df63d0787f5ae6499eb97e13b954b9/components/esp_rom/include/esp32/rom/lldesc.h](url)

aguaviva, 2021-04-14:
@igrr could you give a pointer to the solution of this issue?