espressif/esp-idf #2254: ESP32 Bluetooth modem sleep mode

nicolaerosia, 2018-08-02:
We are in the process of deciding whether to choose ESP32 for our new project which needs BLE and is battery powered.
After doing a little bit of digging, I noticed people complaining that Bluetooth modem sleep mode was not implemented and is planned for v3.1.

I went over the release notes for v3.1-beta1 and I see it is disabled by default and has stability issues.
Is there a timeline for fixing the issues? Can these issues be fixed with the current hardware revision ? (are the issues generated by software?)

negativekelvin, 2018-08-02:
Probably not a good choice, it can't match power performance of dedicated BLE chipsets that can execute from internal flash and are optimized for battery powered designs.

Weijian-Espressif, 2018-08-08:
@nicolaerosia The issues have be fixed , please update to the latest IDF and retry, thanks.

closed 2018-08-08 by FayeY