espressif/esp-idf #2257: BLE gatt client. Unable to establish connection for the second time

KryvosheyaIvan, 2018-08-03:
Hello!
I use standard gatt_client example and it works well. But there the principle "open BLE connection and wait for notify" is used. In my case it is needed to open connection, write necessary characteristics and close connection. I have problems with it: I couldnt establish connection for the second time.
I use the following sequence of API:
1. esp_ble_gattc_open() --> wait for open event
2. esp_ble_gap_update_conn_params()
3. esp_ble_gattc_send_mtu_req() --> wait for MTU_CFG
4. all necessary functions to write chars (that works well)
5. esp_ble_gattc_close()--> wait for close event

Could someone tell me what is wrong?

Weijian-Espressif, 2018-08-23:
@KryvosheyaIvan  please provide the test log

FayeY, 2018-09-25:
See #2167

closed 2018-09-25 by FayeY