espressif/esp-idf #2260: AP and mesh networking

QuentinGuillet, 2018-08-03:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM32
- IDF version : v3.2-dev-179-g04caff2f
- Operating System:	Ubuntu
- Power Supply:      external 5V

## Problem Description

Hello, I would like that my ESP32 creates an access point while being part of a mesh network. When I create the AP, I cannot see it from my computer. I wanted to ask whether the wifi settings were changed by the mesh libraries, and if it is possible to create an access point while being part of a mesh network.

Thanks a lot for the hints

Quentin

dongdong004, 2018-08-07:
The difference in mesh is that root  will working on `WIFI_MODE_APSTA`.

May you need to set ip info by `tcpip_adapter_set_ip_info` on your AP to make sure root has different gateway on interface AP&STA of root. And it's also necessary to bind the ip of station when you create socket on root.

QuentinGuillet, 2018-08-07:
Hello @dongdong004. Thanks for the answer.
I already made sure that the gateway of both interdaces were not on the same subnet. What do you mean with "bind the ip of station" ? Do I have to use static IPs ?

QuentinGuillet, 2018-08-10:
I understood my issue. The mesh network disconnects my AP to create its own which is hidden. Is there a way to have the APs of the mesh network not hidden or to known their SSID ?

dongdong004, 2018-08-10:
Oh, I'm sorry that I misunderstood what you said.

So the issue is you want to creat an AP in mesh node, right? Both softAP and station will be used if the mesh is started. So you could't use it as a simple softAP when you start mesh. Also meshAP is only for mesh nodes and will not allow non-mesh stations to connect now.

By the way you could use `esp_wifi_get_config()` to get SSID.

QuentinGuillet, 2018-08-10:
Ok thank you. 
I will find a way around then, if it is not possible to connect directly my phone to one of the node from the mesh network.

closed 2018-08-10 by QuentinGuillet

abpreen, 2019-04-01:
Hi Did you get any solution for this...
I am also planning to create a mesh network, and these should act as wifiAP for the users to connect and be used as AP.

dongdong004, 2020-02-27:
> Hi Did you get any solution for this...

@abpreen  https://github.com/espressif/esp-mdf/issues/43#issuecomment-483193928