espressif/esp-idf #2262: Whitelisting still does not allow connections

kealist, 2018-08-03:
## Environment

- Core : [ESP-WROOM32]
- IDF version: `1c7a8b3b712c4020562551f692d0ced8f7470d2c`
- Development Env:		[msys32]
- Operating System:		[Windows]
- Power Supply:         [external 3.3V]

## Problem Description

Whitelist does not work as expected, all connection attempts timeout when set to only allow whitelist connections.

I am trying to get the whitelist to work on my code which is based on Secure Gatt Server example.  When I set advertising parameters to be:

`data_adv_params.adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_WLST;`

or 

`data_adv_params.adv_filter_policy = ADV_FILTER_ALLOW_SCAN_WLST_CON_WLST;`

all connections timeout even from whitelisted devices.  No GAP events are handled.  When the device is bonded at `ESP_GAP_BLE_AUTH_CMPL_EVT` event I add it to the whitelist.

## Debug Logs

Here is a sample log.  "Pairing mode" is just a change in advertising to allow any scan any connection for a set period of time.   After the advertising is set back to whitelist, all Gatt connections timeout, even from the newly bonded / whitelisted device.

```
I (10330) GATT SERVER: Entering Pairing Mode
I (10334) GATT SERVER: Removed all bonded devices
I (10347) GATT SERVER: GAP_EVT, event 17
I (10347) GATT SERVER: GAP_EVT, event 6
I (10347) GATT SERVER: advertising start success
I (10349) GATT SERVER: GAP_EVT, event 23
I (10353) GATT SERVER: ESP_GAP_BLE_REMOVE_BOND_DEV
I (10358) GATT SERVER: -----ESP_GAP_BLE_REMOVE_BOND_DEV----
I (10363) GATT SERVER: 61 86 f0 64 a9 f7 
I (10367) GATT SERVER: ------------------------------------
I (13906) GATT SERVER: ESP_GATTS_CONNECT_EVT
I (13907) GATT SERVER: ESP_GATTS_CONNECT_EVT
E (13909) BT_APPL: earlier enc was not done for same device

I (14115) GATT SERVER: GAP_EVT, event 20
W (15705) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (15826) GATT SERVER: GAP_EVT, event 9
I (15827) GATT SERVER: key type = ESP_LE_KEY_LENC
I (15827) GATT SERVER: GAP_EVT, event 9
I (15827) GATT SERVER: key type = ESP_LE_KEY_PENC
I (15831) GATT SERVER: GAP_EVT, event 9
I (15835) GATT SERVER: key type = ESP_LE_KEY_LID
I (16002) GATT SERVER: GAP_EVT, event 9
I (16002) GATT SERVER: key type = ESP_LE_KEY_PID
I (16002) GATT SERVER: GAP_EVT, event 8
I (16003) GATT SERVER: remote BD_ADDR: 44c553f4daed
I (16007) GATT SERVER: address type = 1
I (16011) GATT SERVER: pair status = success
I (16016) GATT SERVER: Bonded devices number : 1
I (16019) GATT SERVER: Bonded devices list : 1
I (16023) GATT SERVER: 44 c5 53 f4 da ed 
I (16028) GATT SERVER: GAP_EVT, event 27
I (16031) GATT SERVER: Whitelist updated
I (16037) GATT SERVER: GAP_EVT, event 20
I (19306) GATT SERVER: ESP_GATTS_DISCONNECT_EVT
I (19306) GATT SERVER: ESP_GATTS_DISCONNECT_EVT
I (19310) GATT SERVER: GAP_EVT, event 6
I (19311) GATT SERVER: advertising start success
I (19312) GATT SERVER: GAP_EVT, event 6
I (19315) GATT SERVER: advertising start success
I (40323) GATT SERVER: bt_pairing_timer_callback
I (40331) GATT SERVER: Stopping Pairing Mode
I (40334) GATT SERVER: GAP_EVT, event 17
I (40334) GATT SERVER: GAP_EVT, event 6
I (40334) GATT SERVER: advertising start success
```

kealist, 2018-08-10:
I updated due to some recent changes and while I see some different output, it seems whitelisting doesn't work for servers still.   

```

I (101127) GATT SERVER: Entering Pairing Mode
I (101127) GATT SERVER: Removed all bonded devices
I (101134) GATT SERVER: GAP_EVT, event 17
I (101134) GATT SERVER: GAP_EVT, event 6
I (101138) GATT SERVER: advertising start success
I (107823) GATT SERVER: ESP_GATTS_CONNECT_EVT
I (107824) GATT SERVER: ESP_GATTS_CONNECT_EVT
E (107826) BT_APPL: earlier enc was not done for same device

I (107939) GATT SERVER: GAP_EVT, event 20
W (109532) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (109654) GATT SERVER: GAP_EVT, event 9
I (109654) GATT SERVER: key type = ESP_LE_KEY_LENC
I (109654) GATT SERVER: GAP_EVT, event 9
I (109654) GATT SERVER: key type = ESP_LE_KEY_PENC
I (109659) GATT SERVER: GAP_EVT, event 9
I (109663) GATT SERVER: key type = ESP_LE_KEY_LID
I (109799) GATT SERVER: GAP_EVT, event 9
I (109800) GATT SERVER: key type = ESP_LE_KEY_PID
I (109800) GATT SERVER: GAP_EVT, event 8
I (109800) GATT SERVER: remote BD_ADDR: 65fbf0ddd585
I (109805) GATT SERVER: address type = 1
I (109809) GATT SERVER: pair status = success
I (109814) GATT SERVER: Bonded devices number : 1
I (109817) GATT SERVER: Bonded devices list : 1
I (109821) GATT SERVER: 65 fb f0 dd d5 85 
I (109826) GATT SERVER: GAP_EVT, event 27
I (109829) GATT SERVER: Whitelist updated
I (109864) GATT SERVER: GAP_EVT, event 20
I (116553) GATT SERVER: ESP_GATTS_DISCONNECT_EVT
I (116554) GATT SERVER: ESP_GATTS_DISCONNECT_EVT
I (116558) GATT SERVER: GAP_EVT, event 6
I (116558) GATT SERVER: advertising start success
I (116559) GATT SERVER: GAP_EVT, event 6
I (116563) GATT SERVER: advertising start success
I (131117) GATT SERVER: bt_pairing_timer_callback
I (131125) GATT SERVER: Stopping Pairing Mode
I (131128) GATT SERVER: GAP_EVT, event 17
I (131128) GATT SERVER: GAP_EVT, event 6
I (131128) GATT SERVER: advertising start success
```

I do duplication of `ESP_GATTS_DISCONNECT_EVT`, which is probably unrelated if it is actually a problem.

MatejMaster, 2018-08-24:
Hi guys, I have similar problems with whitelist (establishing connection without whitelist works as expected, please refer to attachment1.png)... 

Once I set advertising filter policy on ADV_FILTER_ALLOW_SCAN_WLST_CON_WLST I'm still able to see in sniffer tool that scan response is send back to the smart device (android) and ESP is shown up as expected... But unfortunately establishing connection fails… Smart device (android) sends out connect request, but from ESP side there are no more packets seen in sniffer tool (please refer to attachment2.png)…

How this whitelist works in case that from user perspective there are stored just static address. Smart device address is changing quite often. Does the BLE stack take over for IRKs?

Is there any example available – I have found just whitelist example for scanning.

Thank you for info!

BR,
Matej

[image]
[image]

Weijian-Espressif, 2018-09-25:
hi all:
   Sorry, whitelist have a bug. We will find the cause and fix it.

jkhax0r, 2018-10-31:
@Weijian-Espressif Any updates on this?

Weijian-Espressif, 2018-11-01:
@jkhax0r Sorry, this is caused by a hardware problem, we are thinking about a solution, it will take some time.

rtrujillo28, 2018-12-11:
@jkhax0r any movement on this ticket? Has anyone else implemented a manual whitelisting in the meantime?

chegewara, 2019-02-09:
After short investigation i found that with device whitelisted and advertising set to connect whitelisted only i am getting `GATT ERROR 0x85` on android.
http://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/534737#

Maybe its not esp32 issue at the end, i will try to connect with another esp32.

onesbasket, 2019-02-23:
@Weijian-Espressif 
Do you have any movements? or any workaround to prohibit connection from annonymous?

Weijian-Espressif, 2019-03-14:
Sorry, the whitelist only support public address and static random address without encryption.

MatejMasat, 2019-04-05:
@Weijian-Espressif, I’m wondering how can I add device to whitelist without performing bonding? In [Gatt Security Server Example Walkthrough example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md#connecting-and-bonding-to-a-peer-device) is mentioned that pairing and bonding process is invoked with esp_ble_set_encryption()… In latest version of esp-idf, there is no option within esp_ble_sec_act_t for non-secure connection. I tried with 0, but then bonding is not triggered. That means no keys are exchanged – nothing to store for advertising with whitelist
I understand that there is a bug, but is there any workaround? What I would like to achieve is that once one central (smart device) connects to peripheral (esp32), none of other centrals (smart devices) won’t be able to connect to peripheral (esp32). But reconnecting for first connected central (smart device) should work. 

AFAIK direct advertising is no go… since smart devices changes their MACs after some time. Any idea, till waiting on fix? 

Thank you for info!

Weijian-Espressif, 2019-04-08:
@MatejMasat,  I think you do not need to use whitelist. Once your device is successfully bond, ESP32  always shows the first bond address, even if the peer device address has changed. The premise is that the peer device uses a resolvable address.

simfero, 2019-05-02:
@Weijian-Espressif, when can we expect to have a resolution to the random address whitelist issue? Were you able to find the bug and do you have a time frame for a fix?

simfero, 2019-05-10:
After further investigation, I see that esp_ble_gap_update_whitelist() in esp_gap_ble_api.h has two parameters unlike it's analog in bluedroid that has a third parameter esp_ble_wl_addr_type_t wl_addr_type. Why was the third parameter and associated enum dropped in the esp-idf? When can we expect the fix?

@Weijian-Espressif  @TianHao-Espressif @Yulong-espressif @Yulong-espressif

negativekelvin, 2019-05-11:
https://github.com/espressif/esp-idf/commit/c0bdea6aaee41d6a08897d21852fe03f2b19f6ad#diff-ef4b8f59d8290755270cc2d35c13b352

chegewara, 2019-05-11:
I think this is very important comment to note:
> /* Controller do not support resolvable address now, only support public address and static random address */
https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/btm/btm_ble_bgconn.c#L202

simfero, 2019-05-26:
@chegewara good point! How come the official idf doc has the third esp_ble_wl_addr_type_t parameter in esp_ble_gap_update_whitelist() ? I don't see it in the latest release source code in tools\sdk\include\bt\esp_gap_ble_api.h

See here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html?highlight=esp_ble_gap_update_whitelist#_CPPv428esp_ble_gap_update_whitelistb13esp_bd_addr_t22esp_ble_wl_addr_type_t

dracula5555, 2019-05-26:
I am also developing against esp32 and need this functionality, otherwise I'll need to move to another MCU vendor.

OPyshkin, 2019-07-04:
Any news about whitelisting? I saw you restructured bluetooth examples on your repo, maybe there are any updates?

AshUK, 2019-07-22:
+1 we really need this functionality to be able to use RPA with smartphones

Turbonic77, 2019-11-01:
Hi, is there any update on this please - this functionality is required for a project that I am working upon.  An indication of whether this can/will be addressed would be most beneficial, thanks.

BigSmileFace, 2019-12-05:
Mark!
i have the same requirments!

benjaminaigner, 2020-07-02:
:+1: 
I need the whitelist functionality as well.
Any updates? (last statement from Espressif is 04.2019 -> 15months ago)

squonk11, 2020-08-25:
For my project I also need the feature to switch pairing off and activate it only if the user presses a button. Unfortunately this seems to be not possible. Does anyone know a workaround?

chegewara, 2020-08-25:
> For my project I also need the feature to switch pairing off and activate it only if the user presses a button. Unfortunately this seems to be not possible. Does anyone know a workaround?

You can refuse pairing when button is/was not pressed.

squonk11, 2020-08-25:
Thank you for your fast reply.
How do I accomplish to refuse pairing? The callbacks

- bool onConfirmPIN(uint32_t pass_key)
- bool onSecurityRequest()

are never been called in my code. Probably because I am using static passkey?

chegewara, 2020-08-25:
IIRC you have to add 1 line of code after setiing up static pin:
`pSecurity->setAuthenticationMode(param);` with one of those as parameter, because by default static pin is setup to not bond:
- ESP_LE_AUTH_REQ_SC_BOND
- ESP_LE_AUTH_REQ_SC_MITM_BOND
- ESP_LE_AUTH_REQ_SC_MITM

But im not 100% sure right now and there may be something else in security connection setup.

squonk11, 2020-08-25:
I have this line `pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);` already in my code. But the relevant callbacks (mentioned above) are not called. The only callbacks which are called are `void onPassKeyNotify(uint32_t pass_key)` and `void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl)`. But with these I don't know how to deny paring.

chegewara, 2020-08-25:
I dont have good answer at this moment, i would have to test it, but there is hacky way to do it.
You can try to change esp32 capability to `ESP_IO_CAP_KBDISP` for the time when button is not pressed, which would require to enter or confirm pin on esp32 and from callbacks send false or pin 000000 for example.
For sure there is better way to do it, but this is something that came to my mind right now.

squonk11, 2020-08-25:
This workaround seems to work now. Although it is not nice to present the user of the smartphone a request to confirm a PIN which then results in a refusal to connect. But for the time being it could be acceptable until the issue in ESP-IDF if fixed (if it finally will).

SamyRICHET, 2021-07-09:
Hi all,
Do you have any update about the whitelist and RPA?
A solution software or hardware (with the eco-v3 chip for example) was founded ?

duarterr, 2021-07-20:
I'm still struggling to get this working...

xiewenxiang, 2021-07-21:
Parts of the whitelist feature are already available in ESP32,  it's need to enable `Update RPA to Controller` in menuconfig. 
First of all, add the Public address of a peer device to the whitelist, and then connect and pairing, After disconnect, call `esp_ble_gap_config_local_privacy(true)`  generate a resolvable random address. Set ADV parameters and start broadcasting, the `own_addr_type` should set to `BLE_ADDR_TYPE_RPA_RANDOM`, the `adv_filter_policy` set to `ADV_FILTER_ALLOW_SCAN_ANY_CON_WLST`.

Public addresses are not allowed after encryption, only resolvable addresses can be used.

SamyRICHET, 2021-08-04:
Thanks you @xiewenxiang for your answer.
This feature is only work on Bluedroid? 
I don't  @find this option on Nimble menuconfig setting?

rahult-github, 2021-08-09:
Hello @SamyRICHET ,  for Nimble,  if we do not enable BLE_HOST_BASED_PRIVACY, then by default it will go for controller based privacy. So, there is no explicit option "Update RPA to Controller" in menuconfig for Nimble.

rahult-github, 2023-04-25:
Closing this . Please feel free to reopen in case of any issue.

closed 2023-04-25 by rahult-github

MartinTrieu, 2023-12-19:
Having some issues to make it work. 

I'm running idf 5.0.2 on a GATT server initializing advertising with own_addr_type = BLE_ADDR_TYPE_RPA_PUBLIC. When pairing is successful I'm trying to whitelist the peer device (Android phone) and manage to trigger ESP_GAP_BLE_UPDATE_WHITELIST_COMPLETE_EVT. In that event i tried implementing the suggested solution by @xiewenxiang but can't get it to work. Is there any updates regarding the whitelist functionality?