espressif/esp-idf #2263: No `fmemopen` declared in stdio.h

v1993, 2018-08-03:
## Environment

- Development Kit:      noname
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 561884b6
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description

I want to use `fmemopen`, but it seems missing somehow in `stdio.h`.

But manual declaration using `extern "C" FILE* fmemopen(void*, size_t, const char*);` works.

### Expected Behavior
Include `stdio.h`, use `fmemopen`.

### Actual Behavior
Include `stdio.h`, can't use `fmemopen`.

### Steps to repropduce

1. `#include <stdio.h>`
2. Include call to `fmemopen`

### Code to reproduce this issue

```cpp
// I use C++
extern "C" {
	#include <stdio.h>
}

extern "C" void app_main()
{
    auto f = fmemopen(nullptr, 32, "w");
	fclose(f);
}

```

igrr, 2018-08-04:
Such issue usually happens when you override C++ standard for the component or the project, from the default `-std=gnu++11` to e.g. `-std=c++11`. This puts the compiler and the standard library into `__STRICT_ANSI__` mode (see e.g. [this page](https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html), look for `__STRICT_ANSI__`), hence non-standard `fmemopen` does not get defined.

Could you please run `make V=1` so that the full command line used to compile your project is visible? This would confirm whether this is due to C++ standard or something else.

v1993, 2018-08-04:
Thank you! I use C++14, so replacing `-std=c++14` with `-std=gnu++14` did solved the problem.

closed 2018-08-04 by v1993