espressif/esp-idf #2265: COMPONENT_PRIV_INCLUDEDIRS seems not working

v1993, 2018-08-04:
## Environment

- Development Kit:      noname
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 1c7a8b3b
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description

Compilation fails with "No such file or directory". No include in `make V=1` output

### Expected Behavior
Include file from `COMPONENT_PRIV_INCLUDEDIRS` will work.

### Actual Behavior
Include file from `COMPONENT_PRIV_INCLUDEDIRS` doesn't work.

### Steps to repropduce

1. Use `COMPONENT_PRIV_INCLUDEDIRS := config tremor ogg/include` (real code)
2. Include something from that dirs.
3. Run `make V=1 | grep tremor`, get error and no output from grep.

mahavirj, 2018-08-07:
@v1993 
Many of the existing components in ESP-IDF makes use of `COMPONENT_PRIV_INCLUDEDIRS` https://github.com/espressif/esp-idf/blob/b85e8e10b90ab7089ec7989ba513f7a2235f9ea4/components/spiffs/component.mk#L2
Could you please help to check verbose build for such components in your environment?

v1993, 2018-08-07:
Hmm… now I see my problem: error is caused by files which are included from outside.

I have a class containing structure from external library (tremor). So I have to include it in my exporting header, where it can cause naming issues. Worst part is file called "os.h",  because it is common name.

Is there a nice (C++) way to avoid this better than using pointer to void + `new`?

Thanks.

mahavirj, 2018-08-14:
@v1993 
I would recommend you esp32 forum for how-to questions. I will close this issue, if you still think that there is some issue with build system in ESP-IDF, please feel free to reopen.

closed 2018-08-14 by mahavirj