espressif/esp-idf #2266: make broekn after update - esptool write_flash: error: Detected overlap at address: 0xe000 for file: boot_app0.bin

qt1, 2018-08-05:
make flash stopped working.

this is the error:
```
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: /home/baruch/esp/rmt_sender/components/arduino/tools/partitions/boot_app0.bin
/
```

Partition table:
```
cat partitions.csv 
#default.csv from arduino config
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
eeprom,   data, 0x99,    0x290000,0x1000,
spiffs,   data, spiffs,  0x291000,0x16F000,
```

The command make suggests and it fails:
```
$ make
To flash all build output, run 'make flash' or:
python /home/baruch/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /home/baruch/esp/rs232_bridge/build/ota_data_initial.bin 0xe000 /home/baruch/esp/rs232_bridge/components/arduino/tools/partitions/boot_app0.bin 0x1000 /home/baruch/esp/rs232_bridge/build/bootloader/bootloader.bin 0x10000 /home/baruch/esp/rs232_bridge/build/rmt_sender.bin 0x8000 /home/baruch/esp/rs232_bridge/build/partitions.bin
```

negativekelvin, 2018-08-05:
https://github.com/espressif/arduino-esp32/issues/1724

qt1, 2018-08-05:
The above is posted in arduino-idf, it is probably the same issue but it should probably be addressed by the people that manage the make + python flash / partition tools of esp-idf.

Looking at the git history I can see some recent changes and merges that involve files like components/partition_table/Makefile.projbuild and friends, so I would guess this may have some relation to the problem.

Is there a way to know what was the master git branch some two weeks ago ?

qt1, 2018-08-05:
Reverting like below seems to help:

```
cd $IDF_PATH
git reset 8d5ec413d56975df2f893c98644a98a9c93b8e34 --hard
git submodule foreach --recursive git reset --hard
```

Now back to the esp project, do `make clean`, `make` and finally `make flash` works again.

negativekelvin, 2018-08-05:
https://github.com/espressif/esp-idf/commit/de2b1cb02a342d0a706a449cffe39fd16b1a2606

This is the conflict

projectgus, 2018-08-17:
This issue is because ESP-IDF also added functionality for clearing the OTA data partition on flash. This feature already existed in in Arduino, and when arduino-esp32 updated to newer ESP-IDF the command line was accidentally updated to flash this partition twice.

This has now been fixed in arduino-esp32, at the linked issue above.

closed 2018-08-17 by projectgus

qt1, 2018-08-18:
Confirmed fixed in current update.
Thanks !