espressif/esp-idf #2269: Suggestion: Implicit vTaskDelete() on task logic return

nkolban, 2018-08-05:
This issue is being posted as a suggestion for workflow and usability enhancement.  If we write an ESP32 task we are leveraging FreeRTOS xTaskCreate().  When the task runs, it is responsible for calling vTaskDelete() before termination.  Failure to call that function results in an unhandled exception panic.

While it is documented that vTaskDelete() should be called prior to task termination/return, I can't help but feel that we might be able to detect a stack unwinding from a clean return of the function started by a task creation and implicitly perform the work for task deletion behind the scenes.  The benefit of this is that it is one less source of error opportunity for programmers who either mistakenly failed to execute the task delete or otherwise didn't know that was a job to be performed.

Admittedly, I haven't looked at the semantics of FreeRTOS outside of the scope of ESP-IDF and it may very well be the case that FreeRTOS requires this semantic.  If that is the case, then it is unlikely we would want to step over FreeRTOS but rather continue to point the user at FreeRTOS rules.  However ... this will be a function of Espressif internal design intents and whether ESP-IDF wants to try and do more / better / different than the out of the box FreeRTOS.

negativekelvin, 2018-08-05:
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/August_2015/freertos_Return_from_task_31954668j.html

projectgus, 2018-09-12:
Hi @nkolban ,

After a lot of internal discussion, we plan to meet you half-way on this one.

By default, task functions will be in a wrapper so something more user-friendly happens if they return. However, to keep consistently and compatibility with FreeRTOS behaviour, that "something" will be logging an error message and aborting.

This behaviour can be disabled via a config option for cases where the absolute minimal stack size is required.

As it happens, adding these wrapper functions also fixes some other bugs with gdb stack unwinding and C++ exceptions (#2300) in the task function.

This issue will be updated when the fix reaches GitHub.

commit 3ba63a52 (2018-09-18): freertos: Add task function wrapper

This commit adds an option to enclose all FreeRTOS task functions within a
wrapper function. In the case that a task function returns, the wrapper function
will log an error and abort the application immediately.

Closes #2269
Closes #2300
closed 2018-09-18 by igrr