espressif/esp-idf #2271: Any examples with TCP server for LwIP?

sarpdaltaban, 2018-08-06:
PPPoS is working great and thank you for the example.

However, when I try to implement a TCP server over PPPoS, Incorrect ack number returns to the Clients who try to connect.

When I try to implement a TCP client over PPPoS, client can connect to a server but can send data until a TCP window size is reached. The window must be refreshed per send, but it doesn't and overloads..

I tried to use Raw API(LwIP), Netconn API and  BSD Socket API in a second thread and results are the same..

liuzfesp, 2018-08-21:
@sarpdaltaban we don't have example for TCP server yet.

Could you provide your code which is based on Socket API here, then we can check it.

Alvin1Zhang, 2018-09-17:
@sarpdaltaban Could you help provide some inputs? Thanks.

Alvin1Zhang, 2019-06-14:
@sarpdaltaban Thanks for reporting the issue, could you please help provide some more inputs? Thanks.

liuzfesp, 2019-07-10:
@@sarpdaltaban, we already have TCP examples in $IDF_PATH/examples/protocols/sockets/tcp_server, please check it please.

Alvin1Zhang, 2019-07-10:
@sarpdaltaban Thanks for reporting the issue, feel free to reopen the ticket if the issue still happens, thanks.

closed 2019-07-10 by Alvin1Zhang