espressif/esp-idf #2272: ESP32 Classic Bluetooth Initiating Connection Errors

my-abousamra, 2018-08-06:
## Environment

    Development Kit: [none]
    Core (if using chip or module): [ESP-WROOM32]
    IDF version 1c7a8b3b
    Development Env: [Eclipse]
    Operating System: [Ubuntu]
    Power Supply: [Battery]

## Problem Description

I'm trying to make classic Bluetooth connection from ESP32 module to a remote Bluetooth device which requires a pin code to complete the connection operation.

1- I scan for the device.
2- When I found it and get ESP_SPP_DISCOVERY_COMP_EVT event, I initiate a connection using 

`ESP_ERROR_CHECK( esp_spp_connect ( ESP_SPP_SEC_AUTHENTICATE /* or ESP_SPP_SEC_AUTHORIZE */ , ESP_SPP_ROLE_MASTER, param->disc_comp.scn[0], g_peer_bd_addr) );`

3- Then I got ESP_SPP_CL_INIT_EVT event and call

`ESP_ERROR_CHECK ( esp_bt_gap_ssp_passkey_reply ( g_peer_bd_addr, true, 401513  ) );`

where 401513 is the remote address pin code.
4- After a while I receive ESP_SPP_CLOSE_EVT event and these 2 errors

```
E (73975) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
E (73976) esp_bt_gap_cb: authentication failed, status:9
```

How to complete the connection successfully?

Thanks

my-abousamra, 2018-08-06:
@blueMoodBHD please help

blueMoodBHD, 2018-08-28:
@my-abousamra Sorry for later reply.

Maybe your should call` esp_bt_gap_ssp_passkey_reply()` when you got ESP_BT_GAP_KEY_REQ_EVT event. Or, call` esp_bt_gap_ssp_confirm_reply()` when you got ESP_BT_GAP_CFM_REQ_EVT event.

Which event you will got is depend on the IO capability if local device and remote device.

my-abousamra, 2018-08-29:
Note that I call esp_bt_gap_ssp_passkey_reply with remote device pin code when I got ESP_SPP_CL_INIT_EVT event. Is that true??
If not, What should I call?

blueMoodBHD, 2018-08-29:
@my-abousamra No, it is not true. Please try to do nothing  when you got ESP_SPP_CL_INIT_EVT event. Just wait ESP_BT_GAP_KEY_REQ_EVT or ESP_BT_GAP_CFM_REQ_EVT events, then call `esp_bt_gap_ssp_passkey_reply()` or `esp_bt_gap_ssp_confirm_reply()`

blueMoodBHD, 2018-08-29:
By the way, does the remote device support SSP?, or just support legacy pairing?

blueMoodBHD, 2018-08-29:
In SPP, there is no fixed pin code,  It will choose different ways according to the IO  capability.  So, maybe it used legacy pairing.

If it supported, Then, do nothing when you got ESP_SPP_CL_INIT_EVT event, Is there any other event did you got after?

my-abousamra, 2018-08-29:
If I didn't call any function after ESP_SPP_CL_INIT_EVT had arrived then no other event would come, it just waits for a while until the ESP_SPP_CLOSE_EVT event.

blueMoodBHD, 2018-08-29:
No ESP_BT_GAP_KEY_NOTIF_EVT neither? If not, it could well be that they used legacy pairing.

Could please tell me what the remote device is and it's version. I can do some test if I have one.

my-abousamra, 2018-08-29:
It's a medical device so I don't think you have one. It is http://www.nonin.com/OEMSolutions/WristOx23150-OEM
No ESP_BT_GAP_KEY_NOTIF_EVT event comes

blueMoodBHD, 2018-08-29:
The device created with Bluetooth 2.0 wireless technology,
[image]

But , SSP is supported after Bluetooth 2.1, so, same as I guess, they used  legacy pairing instead of SPP.

blueMoodBHD, 2018-09-07:
@my-abousamra I have added legacy pair into ESP-IDF(master) and merge it. The commit id is `8484b9cf` . Please try again.
Thanks.

blueMoodBHD, 2018-09-07:
When you try, after getting ESP_BT_GAP_PIN_REQ_EVT event, call esp_bt_gap_pin_reply() with the pin code of HC-06. Just like the example.

blueMoodBHD, 2018-09-07:
When you try, after getting ESP_BT_GAP_PIN_REQ_EVT event, call esp_bt_gap_pin_reply() with the pin code of your device. Just like the example.

blueMoodBHD, 2018-09-17:
Hi @my-abousamra,  Does it work well or not?

my-abousamra, 2018-09-17:
Hi @blueMoodBHD 
Sorry for the late reply, I will be trying it in 2 days and confirm with you!

my-abousamra, 2018-09-19:
Hi @blueMoodBHD 
I tried it with my device and it works fine although ESP_BT_GAP_PIN_REQ_EVT event with its handling never been called!    

 case ESP_BT_GAP_PIN_REQ_EVT:{
            ESP_LOGI("esp_bt_gap_cb", "ESP_BT_GAP_PIN_REQ_EVT min_16_digit:%d", param->pin_req.min_16_digit);
            if (param->pin_req.min_16_digit) {
                ESP_LOGI("esp_bt_gap_cb", "Input pin code: 0000 0000 0000 0000");
                esp_bt_pin_code_t pin_code = {0};
                esp_bt_gap_pin_reply(param->pin_req.bda, true, 16, pin_code);
            } else {
                ESP_LOGI("esp_bt_gap_cb", "Input pin code: XXXXXX");
                esp_bt_pin_code_t pin_code;
                pin_code[0] = 'X';
                pin_code[1] = 'X';
                pin_code[2] = 'X';
                pin_code[3] = 'X';
                pin_code[4] = 'X';
                pin_code[5] = 'X';
                esp_bt_gap_pin_reply(param->pin_req.bda, true, 6, pin_code);
            }
            break;

closed 2018-09-19 by my-abousamra

blueMoodBHD, 2018-09-20:
Hi @my-abousamra 
I'm happy to hear that  it works fine.
But it is abnormal that ESP_BT_GAP_PIN_REQ_EVT event with its handling never been called.  Did you get ESP_BT_GAP_AUTH_CMPL_EVT and the status is success or not? And did you change the param of `esp_bt_gap_set_pin()` in `app_main`.