espressif/esp-idf #2277: SNTP manual interval request in the application

Trubochkin, 2018-08-07:
The SNTP component [provides public methods](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/lwip/apps/sntp/sntp.h) sntp_init and sntp_stop, but does not provide the method sntp_request publicly. Also, I see that request interval is set by [define](https://github.com/espressif/esp-idf/blob/4480ab6c8ce70778df4948e587e712540de237eb/components/lwip/include/lwip/apps/sntp/sntp_opts.h#L129). I need to set the request interval manually in my application. How can I do this?

liuzfesp, 2018-08-21:
@Trubochkin currently LWIP SNTP doesn't provide API to manually set the polling interval. However, you can change the polling interval by changing SNTP_UPDATE_DELAY in lwipopts.h, the default polling interval is 3600000 milliseconds.

Alvin1Zhang, 2018-10-25:
@Trubochkin Could you help share if any updates for this issue? Thanks.

X-Ryl669, 2018-11-19:
@Trubochkin  You have an error in your code, you're busy waiting for sntp_enabled (probably a zealous ';' here). You probably need #2349 to be fixed too to avoid waiting 500ms for nothing, and instead be informed when the time has been changed.

KonstantinKondrashov, 2019-06-17:
Hi @Trubochkin! We will add the Kconfig option to change `SNTP_UPDATE_DELAY`.  It will be merged soon.

closed 2019-06-26 by igrr