espressif/esp-idf #2278: Heap Corruption

Ralim, 2018-08-08:
## Environment

- Development Kit:      [ESP32-Wrover-Kit| UBLOX NINA W102]
- Kit version (for WroverKit): [v3]
- Core (if using chip or module): [ESP32-Wrover-I| UBLOX NINA W102]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //b85e8e10b90ab7089ec7989ba513f7a2235f9ea4
   //8d3d3fdd4ae1a94daf599545c26848495b2dfcf1
- Development Env:		[Make]
- Operating System:		[Windows|MacOS]
- Power Supply:         [USB]

## Problem Description

Long running code caused CORRUPT HEAP issues in malloc calls after a period of runtime in our main codebase. Attempted diagnostics using the `heap_caps_check_integrity_all` and  the heap tracing instructions.

If heap safety checks are turned to minimum, the code runs for a varying amount of time between 10-45 seconds before crashing out (in that time around 200-400 malloc's work).

When trying to debug the firmware using the above checks `heap_caps_check_integrity_all` would cause a crash whenever it was called, even if called at the beginning of app_main()

Then went and tested on the `hello_world` demo example, by enabling full heap checking and calling heap_caps_check_integrity_all at the beginning of the app_main code. This reproduced the same issue.

Tested on master (as of time of writing) and issue was reproducible. So also tested using the UBLOX NINA module and the issue is reproducible on that module as well. Also tested on a mac computer as well, and issue was reproducible.

Not certain if this is a config issue for the `menuconfig` or possible bug in the heap implementation?

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. Compiled attached files
2. Flashed to different boards
3. Watch monitor for crash messages

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "esp_heap_caps.h"
#include "esp_heap_trace.h"
#include "esp_spi_flash.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
void app_main() {
  heap_caps_check_integrity_all(true);
  heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
  printf("Hello world!\n");

  /* Print chip information */
  esp_chip_info_t chip_info;
  esp_chip_info(&chip_info);
  printf("This is ESP32 chip with %d CPU cores, WiFi%s%s, ", chip_info.cores, (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "", (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");
}
```

## Debug Logs

```
```Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12564
entry 0x40078cb8
I (90) cpu_start: Pro cpu up.
I (90) cpu_start: Single core mode
I (91) heap_init: Initializing. RAM available for dynamic allocation:
D (94) heap_init: New heap initialised at 0x3ffae6e0
I (99) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (106) heap_init: New heap initialised at 0x3ffb37c8
I (110) heap_init: At 3FFB37C8 len 0002C838 (178 KiB): DRAM
I (116) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (123) heap_init: At 3FFE4350 len 00013CB0 (79 KiB): D/IRAM
I (129) heap_init: At 3FFFC000 len 00004000 (16 KiB): D/IRAM
D (138) heap_init: New heap initialised at 0x40089578
I (140) heap_init: At 40089578 len 00016A88 (90 KiB): IRAM
I (147) cpu_start: Pro cpu start user code
D (159) clk: RTC_SLOW_CLK calibration value: 3006144
D (168) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (169) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (170) stack_chk: Intialize random stack guard
D (174) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (179) cpu_start: Starting scheduler on PRO CPU.
D (185) heap_init: New heap initialised at 0x3ffe0440
D (185) heap_init: New heap initialised at 0x3ffe4350
D (185) heap_init: New heap initialised at 0x3fffc000
D (195) intr_alloc: Connected src 16 to int 12 (cpu 0)
CORRUPT HEAP: Block 0x3fffc010 is outside heap (last valid block 0x00000000)
Heap summary for capabilities 0x00001000:
  At 0x3ffae6e0 len 6432 free 24 allocated 5976 min_free 24
    largest_free_block 24 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3ffb37c8 len 182328 free 171564 allocated 10540 min_free 171564
    largest_free_block 171564 alloc_blocks 11 free_blocks 1 total_blocks 12
  At 0x3ffe0440 len 15296 free 15248 allocated 0 min_free 15248
    largest_free_block 15248 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 81072 free 81024 allocated 0 min_free 81024
    largest_free_block 81024 alloc_blocks 0 free_blocks 1 total_blocks 1
assertion "next > (intptr_t)block" failed: file "/Users/fred/esp/esp-idf/components/heap/./multi_heap.c", line 116, function: get_next_block
abort() was called at PC 0x400d203f on core 0
0x400d203f: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40087480:0x3ffb4bc0 0x4008757f:0x3ffb4be0 0x400d203f:0x3ffb4c00 0x40086db6:0x3ffb4c30 0x400871b9:0x3ffb4c50 0x400d1e11:0x3ffb4c70 0x400d1f7e:0x3ffb4cb0 0x400d07ba:0x3ffb4ce0
0x40087480: invoke_abort at /Users/fred/esp/esp-idf/components/esp32/./panic.c:572

0x4008757f: abort at /Users/fred/esp/esp-idf/components/esp32/./panic.c:572

0x400d203f: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40086db6: get_next_block at /Users/fred/esp/esp-idf/components/heap/./multi_heap.c:116
 (inlined by) multi_heap_get_info_impl at /Users/fred/esp/esp-idf/components/heap/./multi_heap.c:690

0x400871b9: multi_heap_get_info at /Users/fred/esp/esp-idf/components/heap/./multi_heap_poisoning.c:315

0x400d1e11: heap_caps_print_heap_info at /Users/fred/esp/esp-idf/components/heap/./heap_caps.c:389

0x400d1f7e: app_main at /Users/fred/Downloads/hello_world/main/./hello_world_main.c:21

0x400d07ba: main_task at /Users/fred/esp/esp-idf/components/esp32/./cpu_start.c:449

================= CORE DUMP START =================

```

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [X] coredump (This provides stacks of tasks.) 

Files in the zip

[hello-world.zip](https://github.com/espressif/esp-idf/files/2268964/hello-world.zip)

mahavirj, 2018-08-18:
@Ralim 
For first part, there is issue in trax feature (enabled in your sdkconfig `CONFIG_ESP32_TRAX`) with respect to single core mode of operation, it reserves incorrect memory bank and hence results in heap corruption. Fix for that should appear soon in ESP-IDF.

For original issue of heap corruption in longer run, I am not sure but if you had kept this feature always enabled (and not just for debugging purpose), then it might fix your problem. I will notify once fix gets merged and maybe you can retest your use-case.

mahavirj, 2018-08-24:
@Ralim 
Please try with https://github.com/espressif/esp-idf/commit/902344c5163ff2e72b2f98996d170c5c09be718e

Ralim, 2018-08-27:
Hi @mahavirj ,
Looks to be working well at the moment. 
I'll be back if it comes back up again.

closed 2018-08-27 by Ralim

PRitu21, 2019-03-14:
CORRUPT HEAP: multi_heap.c:429 detected at 0x3ffd5cb8
abort() was called at PC 0x4008ebb3 on core 0

Any suggestion???