espressif/esp-idf #2279: error MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY

Rajkumar181, 2018-08-08:
why will posting the data through mbedtls_ssl_write , i am getting this error MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY.

Debug output : 

I (1932868) mbedtls: ssl_tls.c:6904 => read

I (1932878) mbedtls: ssl_tls.c:3811 => read record

I (1932878) mbedtls: ssl_tls.c:2254 => fetch input

I (1932888) mbedtls: ssl_tls.c:2415 in_left: 0, nb_want: 5

I (1937668) mbedtls: ssl_tls.c:2439 in_left: 0, nb_want: 5

I (1937668) mbedtls: ssl_tls.c:2440 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (1937668) mbedtls: ssl_tls.c:2460 <= fetch input

I (1937678) mbedtls: ssl_tls.c:2254 => fetch input

I (1937678) mbedtls: ssl_tls.c:2415 in_left: 5, nb_want: 31

I (1937688) mbedtls: ssl_tls.c:2439 in_left: 5, nb_want: 31

I (1937688) mbedtls: ssl_tls.c:2440 ssl->f_recv(_timeout)() returned 26 (-0xffffffe6)

I (1937698) mbedtls: ssl_tls.c:2460 <= fetch input

I (1937708) mbedtls: ssl_tls.c:1621 => decrypt buf

I (1937718) mbedtls: ssl_tls.c:2094 <= decrypt buf

I (1937718) mbedtls: ssl_tls.c:4102 got an alert message, type: [1:0]

I (1937728) mbedtls: ssl_tls.c:4117 is a close notify message

W (1937728) mbedtls: ssl_tls.c:3833 mbedtls_ssl_handle_message_type() returned -30848 (-0x7880)

W (1937738) mbedtls: ssl_tls.c:6993 mbedtls_ssl_read_record() returned -30848 (-0x7880)

W (1937748) Mbedtls: ret read: -30848
W (1937758) Mbedtls: MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
is it configuration issue.

Alvin1Zhang, 2020-06-08:
@Rajkumar181 Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2020-06-15:
@Rajkumar181 We will need more information to comment on issue (server can close connection under multiple scenarios, error `MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY`). Please feel free to re-open with more information along with sample code.

closed 2020-06-15 by mahavirj