espressif/esp-idf #2280: issue with DNS resolve

Rajkumar181, 2018-08-08:
i am trying to use  the DNS resolver to try to resolve google.com. it is working fine when internet is there.
But when internet is got loss. esp32 getting rested. Why?

Debug logs : 

Get IP for URL: dns.google.com
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2e0  PS      : 0x00060930  A0      : 0x800d31f4  A1      : 0x3ffd1450  
A2      : 0x3ffc7b24  A3      : 0x00000000  A4      : 0x00000014  A5      : 0x3ffc7b24  
A6      : 0x00060f23  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffd1520  
A10     : 0x3ffc715c  A11     : 0x3ffd1540  A12     : 0x4013a16c  A13     : 0x00000000  
0x4013a16c: dns_timer at /home/iram14/esp/esp-idf/components/lwip/core/timers.c:224

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4000c2e0:0x3ffd1450 0x400d31f1:0x3ffd1460 0x4013758b:0x3ffd1480 0x40137a4c:0x3ffd14a0 0x40137adb:0x3ffd14c0 0x40138317:0x3ffd14e0 0x4013a16f:0x3ffd1500 0x4013a269:0x3ffd1520 0x40134378:0x3ffd1540
0x400d31f1: dns_found_cb at /home/iram14/Documents/RM3100_14_07_2018_GW_SENSOR/ESP32_D/main/main.c:58

0x4013758b: dns_call_found at /home/iram14/esp/esp-idf/components/lwip/core/dns.c:909

0x40137a4c: dns_check_entry at /home/iram14/esp/esp-idf/components/lwip/core/dns.c:1019

0x40137adb: dns_check_entries at /home/iram14/esp/esp-idf/components/lwip/core/dns.c:1063 (discriminator 3)

0x40138317: dns_tmr at /home/iram14/esp/esp-idf/components/lwip/core/dns.c:467

0x4013a16f: dns_timer at /home/iram14/esp/esp-idf/components/lwip/core/timers.c:227

0x4013a269: sys_timeouts_mbox_fetch at /home/iram14/esp/esp-idf/components/lwip/core/timers.c:575

0x40134378: tcpip_thread at /home/iram14/esp/esp-idf/components/lwip/api/tcpip.c:474

igrr, 2018-08-08:
The backtrace points to dns_found_cb in your main.c file. Are you trying to use LwIP raw API to do resolve a domain name?

Rajkumar181, 2018-08-10:
Yes i am tying to use LWIP raw API.

void dns_found_cb(const char *name, const ip_addr_t *ipaddr, void *callback_arg)
{
    ip_Addr = *ipaddr;
    bDNSFound = true;
}

chegewara, 2018-08-10:
Similar to this?
https://github.com/chegewara/esp32-ota-with-ble-setup/blob/master/main/ota_main.cpp#L109-L142
I did not test it lately but it worked before.

FayeY, 2018-08-20:
Any update? Is this issue solved now?

Rajkumar181, 2018-08-21:
i tried similar to this  -- https://github.com/chegewara/esp32-ota-with-ble-setup/blob/master/main/ota_main.cpp#L109-L142

but still the same problem.

liuzfesp, 2018-09-04:
@zhangyanjiao please help

zhangyanjiaoesp, 2018-09-06:
@Rajkumar181 please look like this https://github.com/espressif/esp-idf/issues/2113#issuecomment-405777313. Or you can refer to the implementation of [this function](https://github.com/espressif/esp-idf/blob/a0d2dd03856fec4242d0ac043898b404a5caff17/components/lwip/api/api_msg.c#L1957) to write your own `dns_found_cb`

Alvin1Zhang, 2018-10-25:
@Rajkumar181 Could you help share if any updates for this issue? Thanks.

FayeY, 2018-12-18:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-12-18 by FayeY