espressif/esp-idf #2285: esp_log_level_set("*", ESP_LOG_INFO) doesn't work

v1993, 2018-08-11:
## Environment

- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description

After executing `esp_log_level_set("*", ESP_LOG_INFO);` debug messages from `esp_timer` module is still here.

### Expected Behavior
1. Set `Default log vebosity` to debug.
2. Run `esp_log_level_set("*", ESP_LOG_INFO);`.
3. No more debug messages from modules.

### Actual Behavior
1. Set `Default log vebosity` to debug.
2. Run `esp_log_level_set("*", ESP_LOG_INFO);`.
3. Still getting debug messages from modules.

### Steps to repropduce
Listed above

v1993, 2019-05-04:
Are there any news aboun this?

projectgus, 2021-04-16:
Hi @v1993,

Sorry for the unacceptably long time it's taken someone to reply to you about this, and thanks for being patient.

The root cause for this is that some log lines have to be ISR-safe as they may be called with interrupts disabled, which means they use `ESP_EARLY_LOGx` macros instead of `ESP_LOGx`. Until now these macros aren't subject to any runtime filtering - if they're compiled in at the default log level then they're always going to log. The only current workaround is to define the `LOG_LOCAL_LEVEL` macro in the corresponding source files to change their compile-time log level.

We have a change in review that fixes this. It's still not possible to do tag-specific filtering on "early" log macros, but setting the default level with `*` will enable/disable them.

This issue will be updated when that fix is available (should be soon).

closed 2021-05-04 by espressif-bot