espressif/esp-idf #2286: BT_HCI: hci write adv params error 0x12

chegewara, 2018-08-11:
I have this error right after starting advertising but right before:
`ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT`

master commit.

chegewara, 2018-08-11:
There is something strange. After 10 or so app re-flashing this error disappeared and this is no longer issue:
https://github.com/espressif/esp-idf/issues/2185

Weijian-Espressif, 2018-08-23:
@chegewara Can you provide your test code? thanks.

chegewara, 2018-08-23:
@Weijian-Espressif  I can, but its code with c++ Kolban's library, if you dont mind to use it.

ArTyGill, 2018-08-30:
How to solve the problem? I need a stable job, at some point all the esp began to work with the dispatch interval of advertising packages 1270ms. The rollback to the old idf did not help.

Weijian-Espressif, 2018-09-07:
hi chegewara , I can't reproduce the issue?  Do you use other library? or Can you reproduce it with our IDF demo?

chegewara, 2018-09-07:
hi @Weijian-Espressif as far as i remember it was simple example code to test new wrover module, no other libraries used except Kolban's BLE library for arduino. As you can see it was 4 weeks ago, so i will try to test it one more time and post results here (when i find some time).

closed 2018-10-02 by chegewara