espressif/esp-idf #2289: esp_ble_gatts_delete_service and 0x2a05 indicate

chegewara, 2018-08-12:
Hi, maybe its my test app issue or my understanding problem, but is not that after deleting services ble should send indication in Service changed characteristic? (service 0x1801, characteristic 0x2a05)

I would send it myself but its generic service created by ble stack and we dont have access to it.

Here is log when i delete service:
```
D (24639) BLEService: >> stop(): Stopping service (esp_ble_gatts_stop_service): UUID: 91bad492-b950-4226-aa2b-4ede9fa42f59, handle: 0x28
D (24641) FreeRTOS: Semaphore taking: name: StopEvt (0x3ffdde04), owner: <N/A> for stop
D (24649) FreeRTOS: Semaphore taken:  name: StopEvt (0x3ffdde04), owner: stop
D (24657) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_STOP_EVT
D (24664) BLEUtils: GATT ServerEvent: ESP_GATTS_STOP_EVT
D (24669) BLEUtils: dumpGattServerEvent: *** NOT CODED ***
D (24675) BLEServer: >> handleGATTServerEvent: ESP_GATTS_STOP_EVT
D (24681) BLEService: << stop()
D (24681) BLECharacteristic: >> handleGATTServerEvent: ESP_GATTS_STOP_EVT
D (24684) BLEService: >> executeDelete()
D (24691) BLECharacteristic: << handleGATTServerEvent
D (24700) BLEServer: << handleGATTServerEvent
D (24695) FreeRTOS: Semaphore taking: name: DeleteEvt (0x3ffddd54), owner: <N/A> for executeDelete
D (24714) FreeRTOS: Semaphore taken:  name: DeleteEvt (0x3ffddd54), owner: executeDelete
D (24725) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_DELETE_EVT
D (24730) BLEUtils: GATT ServerEvent: ESP_GATTS_DELETE_EVT
D (24735) BLEUtils: dumpGattServerEvent: *** NOT CODED ***
D (24741) BLEServer: >> handleGATTServerEvent: ESP_GATTS_DELETE_EVT
D (24747) BLECharacteristic: >> handleGATTServerEvent: ESP_GATTS_DELETE_EVT
D (24747) BLEService: << executeDelete
D (24754) BLECharacteristic: << handleGATTServerEvent
D (24763) BLEServer: << handleGATTServerEvent
D (25207) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (25208) BLEUtils: [status: 0, bd_addr: 66:b9:b8:1d:f6:cd, min_int: 0, max_int: 0, latency: 0, conn_int: 6, timeout: 500]
D (25217) BLEServer: BLEServer ... handling GAP event!
D (25401) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (25401) BLEUtils: [status: 0, bd_addr: 66:b9:b8:1d:f6:cd, min_int: 0, max_int: 0, latency: 0, conn_int: 39, timeout: 500]
D (25408) BLEServer: BLEServer ... handling GAP event!
```

nRF connect is only receiving two `ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT` events.

Weijian-Espressif, 2018-08-23:
@chegewara ,  If you delete the service, the server will automatically send a service change indication during the Bluetooth connection. You can refer to gatt_proc_srv_chg (void) in gatt_main.c in line 1131.

chegewara, 2018-08-23:
@Weijian-Espressif well, its not sending or nRF connect is not updating services (i cant confirm it with any other app or wireshark). As you can see `ESP_GATTS_DELETE_EVT` is fired, so im guesssing my esp32 test app is working.

Weijian-Espressif, 2018-08-31:
hi @chegewara 
here is the test log and test code:

gatt_client:

I (1252) GATTC_DEMO: stop scan successfully
I (1282) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 1
I (1282) GATTC_DEMO: REMOTE BDA:
I (1282) GATTC_DEMO: 30 ae a4 00 48 a6 
I (1282) GATTC_DEMO: open success
I (1442) GATTC_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (2682) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (2682) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (2682) GATTC_DEMO: start handle 44 end handle 65535 current handle value 44
I (2692) GATTC_DEMO: service found
I (2702) GATTC_DEMO: UUID16: ee
I (2702) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (2702) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
E (2762) GATTC_DEMO: ESP_GATTC_SRVC_CHG_EVT, receive service changed!!!!
E (2762) GATTC_DEMO: write descr failed, error status = 85
I (3122) GATTC_DEMO: update connection params status = 0, min_int = 6, max_int = 6,conn_int = 6,latency = 0, timeout = 600
I (3372) GATTC_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400

gatt_server:
 (155) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (235) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (515) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (515) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (525) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (525) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (525) GATTS_DEMO: the gatts demo char length = 3

I (535) GATTS_DEMO: prf_char[0] =11

I (535) GATTS_DEMO: prf_char[1] =22

I (545) GATTS_DEMO: prf_char[2] =33

I (545) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (555) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (565) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (565) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (575) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (585) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

I (6175) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 30:ae:a4:80:41:56:
I (6175) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote 30:ae:a4:80:41:56:
I (6355) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (7555) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (7555) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (7635) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 47

I (7635) GATTS_DEMO: GATT_WRITE_EVT, value len 2, value :
I (7645) GATTS_DEMO: 01 00 
I (7645) GATTS_DEMO: notify enable
E (7655) BT_GATT: Active Service Found 
E (7655) GATTS_DEMO: GATTS_TAGESP_GATTS_DELETE_EVT!!!!!
I (8035) GATTS_DEMO: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 600
I (8285) GATTS_DEMO: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 32,latency = 0, timeout = 400

Weijian-Espressif, 2018-08-31:
[test_code.zip](https://github.com/espressif/esp-idf/files/2339181/test_code.zip)

chegewara, 2018-09-19:
@Weijian-Espressif thanks for test code. It seems to work with esp32 client and server code, but nRF connect is not getting or not properly handling this event. 
Ive been exchanging few emails with Nordic semiconductor team and they point me that characteristic does not have CCC descriptor so its not possible to register for indications and thats the reason android cant register and receive service changed indications.

Since its working on esp32 i am assuming it is handled in controller layer without CCC descriptor.

chegewara, 2018-09-20:
From core specs:
> 7.1 SERVICE CHANGED
The «Service Changed» characteristic is a control-point attribute (as defined in
[Vol 3] Part F, Section 3.2.6) that shall be used to indicate to connected devices
that services have changed (i.e., added, removed or modified). The
characteristic shall be used to indicate to clients that have a trusted relationship
(i.e. bond) with the server when GATT based services have changed when
they re-connect to the server. See Section 2.5.2.
This Characteristic Value shall be configured to be indicated, using the Client
Characteristic Configuration descriptor by a client. Indications caused by
changes to the Service Changed Characteristic Value shall be considered lost
if the client has not enabled indications in the Client Configuration
Characteristic Descriptor.
The Service Changed Characteristic Value is two 16-bit Attribute Handles
concatenated together indicating the beginning and ending Attribute Handles
affected by an addition, removal, or modification to a GATT-based service on
the server. A change to a characteristic value is not considered a modification
of the service. If a change has been made to any of the GATT service definition
characteristic values other than the Service Changed characteristic value, the
range shall also include the beginning and ending Attribute Handle for the
GATT service definition.
There is only one instance of the Service Changed characteristic within the
GATT service definition. A Service Changed characteristic value shall exist for
each client with a trusted relationship.
If the list of GATT based services and the service definitions cannot change for
the lifetime of the device then this characteristic shall not exist, otherwise this
characteristic shall exist.
If the Service Changed characteristic exists on the server, the Characteristic
Value Indication support on the server is mandatory.
The client shall support Characteristic Value Indication of the Service Changed
characteristic.
The Service Changed characteristic Attribute Handle on the server shall not
change if the server has a trusted relationship with any client.

If i understand this correctly then characteristic 0x2a05 shall have CCC descriptor but at the moment it does not have or it is not discoverable by nRF connect or esp32 client app:

> This Characteristic Value shall be configured to be indicated, using the Client
> Characteristic Configuration descriptor by a client. Indications caused by
> changes to the Service Changed Characteristic Value shall be considered lost
> if the client has not enabled indications in the Client Configuration
> Characteristic Descriptor.

Weijian-Espressif, 2018-09-20:
@chegewara  Thanks.  I will add ccc for  characteristic 0x2a05.   By the way, our ESP32 uses the bluedroid source code,  we test android and iOS , they can receive service change indication.

chegewara, 2018-09-20:
@Weijian-Espressif Thanks for support, and there is no rush. It is not required at the moment and i dont think it will be required in near future. It can be added in v 4 if at all.

chegewara, 2018-09-25:
From `BLUETOOTH SPECIFICATION Version 5.0 | Vol 3, Part G` page 2227:

> For clients that have a trusted relationship (i.e. bond) with the server, the
attribute cache is valid across connections. For clients with a trusted
relationship and not in a connection when a service change occurs, the server
shall send an indication when the client reconnects to the server (see Section
7.1). For clients that do not have a trusted relationship with the server, the
attribute cache is valid only during the connection. Clients without a trusted
relationship shall receive an indication when the service change occurs only
during the current connection.

Is it driver (bluedroid) implemented or suppose to be application implemented?

chegewara, 2018-10-01:
@Weijian-Espressif Thanks for adding CCC descriptor, but there is one issue. I am testing it with esp_gatt_server example and nRF connect reporting insufficient permission during read/write CCC on 0x2a05.

chegewara, 2018-10-04:
Fixed, thanks.

closed 2018-10-04 by chegewara