espressif/esp-idf #2290: lock is not released

chegewara, 2018-08-12:
Sorry if i misunderstand something, but here i see that lock is aquired in this line:
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L965

but is released only when bt controller cant be enabled, here:
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L993-L1007

Should not be released also in case of enabling controller without errors, in here:
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L1010

igrr, 2018-08-12:
Power management lock is taken here to prevent APB frequency from being reduced while BT is used. So the usage appears to be correct.

At some point, BT code will support reducing APB frequency in BT modem sleep mode. In this case PM lock will also be released on entry to the modem sleep mode.

closed 2018-08-12 by igrr

chegewara, 2018-08-12:
Ok, thanks