espressif/esp-idf #2292: esp_bt_mem_release issue with ESP_BT_MODE_BTDM as parameter

chegewara, 2018-08-12:
Calling one of those or both to release memory works:
```
esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_mem_release(ESP_BT_MODE_BLE);
```
but calling:
`esp_bt_mem_release(ESP_BT_MODE_BTDM);`
is causing crash in line:
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L800

PS what is the point of having two times the same code here(just asking):
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L796-L805
and here:
https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/components/bt/bt.c#L819-L828

dhrishi, 2018-08-12:
@chegewara If you check, mem_start and mem_end point to different regions in all these calls:
1. BTDM BSS and Data
2. BT BSS and Data

chegewara, 2018-08-12:
Its not the point. When i try only this command then it crash:
`esp_bt_mem_release(ESP_BT_MODE_BTDM);`
But i can use separate both commands together to release all bt memory and it works:
```
esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_mem_release(ESP_BT_MODE_BLE);
```
There is no need to build special test app for this, i am using this:
```
app_main(){
esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_mem_release(ESP_BT_MODE_BLE);
}
```
I am assuming that 
`esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);` +`esp_bt_mem_release(ESP_BT_MODE_BLE);`
is the same as `esp_bt_mem_release(ESP_BT_MODE_BTDM);`
but again i may be wrong. 
Only i know this command is causing crash `esp_bt_mem_release(ESP_BT_MODE_BTDM);`

chegewara, 2018-08-12:
@dhrishi I see what you mean, my bad sorry.

dhrishi, 2018-08-12:
@chegewara No. Calling the below APIs in succession
esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_mem_release(ESP_BT_MODE_BLE);

is not equivalent to calling esp_bt_mem_release(ESP_BT_MODE_BTDM);

If ESP_BT_MODE_CLASSIC_BT or ESP_BT_MODE_BLE mode is passed to esp_bt_mem_release(), it just releases the controller memory. It does not maintain the state of your previous calls and hence doesn't call the conditions which check for ESP_BT_MODE_BTDM.

Although, if ESP_BT_MODE_BTDM is passed then apart from freeing the controller memory, it releases BTDM and BT library data and BSS to heap. 

For more info and details about appropriate usage of this API, please refer its documentation.

FayeY, 2018-08-20:
Is this issue solved now?

chegewara, 2018-08-20:
I dont know, is it solved? Is it issue at all?

dhrishi, 2018-08-20:
It may not be considered as an issue, as the API has been used incorrectly or under wrong conditions. Although, an internal MR is already submitted which should handle this error case. 
I will update when it gets merged. @chegewara you can then give it a try and if it fixes your issue then we will close this.

dhrishi, 2018-08-26:
@chegewara Can you try with latest ESP IDF and see if you don't see a crash anymore?

chegewara, 2018-08-28:
@dhrishi when i use this code then all works fine:
```
esp_bluedroid_deinit();
 esp_bt_controller_disable();
 esp_bt_controller_deinit();
 esp_bt_mem_release(ESP_BT_MODE_BTDM);
```
and no, i app still crash if i use only ` esp_bt_mem_release(ESP_BT_MODE_BTDM);`. Logs:
```
I (3076) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (3095) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
D (3114) heap_init: New heap initialised at 0x3ffc1048
I (3130) heap_init: At 3FFC1048 len 0001EFB8 (123 KiB): DRAM
I (3150) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (3169) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (3190) heap_init: New heap initialised at 0x4008e010
I (3205) heap_init: At 4008E010 len 00011FF0 (71 KiB): IRAM
I (3225) cpu_start: Pro cpu start user code
D (3261) clk: RTC_SLOW_CLK calibration value: 3461850
D (1069) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (1070) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (1071) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (1076) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (4) cpu_start: Starting scheduler on APP CPU.
D (1100) heap_init: New heap initialised at 0x3ffe0440
D (1105) heap_init: New heap initialised at 0x3ffe4350
I (1111) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (1119) spiram: Allocating block of size 32768 bytes
D (1124) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (1130) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]

D (1136) BTDM_INIT: Release DRAM [0x3ffb0000] - [0x3ffb6388]

D (1142) BTDM_INIT: Release DRAM [0x3ffb8000] - [0x3ffb9a20]

D (1147) BTDM_INIT: Release DRAM [0x3ffbdb28] - [0x3ffbdb5c]

D (1153) BTDM_INIT: Release BT BSS [0x3ffbfdc8] - [0x3ffbfdcc]

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400e673b
0x400e673b: esp_bt_mem_release at D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c:917 (discriminator 1)

file: "D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c" line 917
func: esp_bt_mem_release
expression: heap_caps_add_region(mem_start, mem_end)

Backtrace: 0x4008a0fc:0x3ffbafd0 0x4008a544:0x3ffbaff0 0x400e673b:0x3ffbb010 0x400d5159:0x3ffbb040 0x400d514f:0x3ffbb060 0x400d0b56:0x3ffbb080
0x4008a0fc: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008a544: _esp_error_check_failed at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:676

0x400e673b: esp_bt_mem_release at D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c:917 (discriminator 1)

0x400d5159: SampleServer() at D:/msys32/home/imper/ble_test/main/SampleServer.cpp:102

0x400d514f: app_main at D:/msys32/home/imper/ble_test/main/main.cpp:41
```

I should read docs more often.

chegewara, 2018-08-28:
Here is log when i use full bt deinit and app crash:
```
D (1259) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]

D (1265) BTDM_INIT: Release DRAM [0x3ffb0000] - [0x3ffb6388]

D (1271) BTDM_INIT: Release DRAM [0x3ffb8000] - [0x3ffb9a20]

D (1277) BTDM_INIT: Release DRAM [0x3ffbdb28] - [0x3ffbdb5c]

D (1283) BTDM_INIT: Release BTDM BSS [0x3ffc0b38] - [0x3ffc13d8]

D (1289) BTDM_INIT: Release BTDM Data [0x3ffbdbb4] - [0x3ffbdbd4]

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400e7423
0x400e7423: esp_bt_controller_mem_release at D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c:1130

file: "D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c" line 896
func: esp_bt_controller_mem_release
expression: heap_caps_add_region(mem_start, mem_end)

Backtrace: 0x400920b8:0x3ffbafa0 0x40092504:0x3ffbafc0 0x400e7423:0x3ffbafe0 0x400e7445:0x3ffbb010 0x400d5dee:0x3ffbb040 0x400d5ddb:0x3ffbb060 0x400d15b2:0x3ffbb080
0x400920b8: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x40092504: _esp_error_check_failed at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:676

0x400e7423: esp_bt_controller_mem_release at D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c:1130

0x400e7445: esp_bt_mem_release at D:/msys32/home/imper/esp/esp-idf/components/bt/bt.c:1130

0x400d5dee: SampleServer() at D:/msys32/home/imper/ble_test/main/SampleServer.cpp:102

0x400d5ddb: app_main at D:/msys32/home/imper/ble_test/main/main.cpp:41

0x400d15b2: main_task at D:/msys32/home/imper/esp/esp-idf/components/esp32/cpu_start.c:473
```

dhrishi, 2018-08-28:
@chegewara Thanks for your response. A couple of things here:
1. Initially (with no logs) I thought the issue would have been that start and end are same, the region being empty, and hence heap_caps_add_region() should not be called. This is fixed with: eb181220384d1e974fb4c8beb8a618bd4f2862c0
2. With the logs you have provided for both cases (I assume that you have still not enabled classic BT), the address range which cause the crash: 
(0x3ffbfdc8, 0x3ffbfdcc) and (0x3ffbdbb4, 0x3ffbdbd4) seem to be invalid (Please refer file soc_memory_layout.h) and hence it has asserted from heap_caps_add_region () on purpose.
Let me know if the explanation helps.

chegewara, 2018-08-28:
@dhrishi One word of explanation. I did like you suggested before, ive read docs and tried this code (before updating esp-idf):
```
esp_bluedroid_disable();
 esp_bluedroid_deinit();
 esp_bt_controller_disable();
 esp_bt_controller_deinit();
 esp_bt_mem_release(ESP_BT_MODE_BTDM);
```
It worked, so like i said i have to read docs more often.

Now, you asked me to test with only one command `esp_bt_mem_release(ESP_BT_MODE_BTDM);`, so i updated to latest commit and here is the problem. Test app crash when i am using only `esp_bt_mem_release(ESP_BT_MODE_BTDM);` or when i am doing full bt deinit, and yes i have turned on both classic bt and ble.

negativekelvin, 2018-08-28:
> (0x3ffbfdc8, 0x3ffbfdcc) and (0x3ffbdbb4, 0x3ffbdbd4) seem to be invalid 
@dhrishi no they are just too small to create a new heap

dhrishi, 2018-08-28:
@negativekelvin Yes. That issue has arisen due to latest optimization in BT.
Releasing BTDM data to heap gives this error as the size is too less (around 32 bytes) to create a new heap. I am already working on it so that esp_bt_mem_release handles it correctly.

negativekelvin, 2018-08-28:
@dhrishi What it should do is try to combine the bss,data sections and create 2 heap regions not 4

FayeY, 2018-09-13:
This issue should have been solved in the latest IDF, please have a try. Thanks.

closed 2018-09-13 by FayeY

negativekelvin, 2018-09-13:
@dhrishi @TianHao-Espressif can this fragmentation be improved?? i think yes.

```
D (262) BTDM_INIT: .data initialise [0x3ffae6e0] - [0x3ffaff10]         D (1122) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]
I (750) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (262) BTDM_INIT: .bss initialise [0x3ffb0000] - [0x3ffb09a8]          D (1122) BTDM_INIT: Release DRAM [0x3ffb0000] - 
D (272) BTDM_INIT: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]
D (272) BTDM_INIT: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]
D (282) BTDM_INIT: .bss initialise [0x3ffb2730] - [0x3ffb6388]                                                          [0x3ffb6388]
I (761) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
D (292) BTDM_INIT: .bss initialise [0x3ffb8000] - [0x3ffb9a20]          D (1132) BTDM_INIT: Release DRAM [0x3ffb8000] - [0x3ffb9a20]
I (772) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
D (292) BTDM_INIT: .bss initialise [0x3ffbdb28] - [0x3ffbdb5c]          D (1132) BTDM_INIT: Release DRAM [0x3ffbdb28] - [0x3ffbdb5c]
I (778) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
                                                                        D (1162) BTDM_INIT: Release BT Data [0x3ffbdb60] - [0x3ffbdcf8]
                                                                        D (1142) BTDM_INIT: Release BTDM Data [0x3ffbdcf8] - [0x3ffbdd18]
                                                                        
                                                                        D (1152) BTDM_INIT: Release BT BSS [0x3ffc0b40] - [0x3ffc9c30]
                                                                        D (1142) BTDM_INIT: Release BTDM BSS [0x3ffc9c30] - [0x3ffca4d0]

I (789) heap_init: At 3FFCC580 len 00013A80 (78 KiB): DRAM
I (796) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (802) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (814) heap_init: At 40091EF0 len 0000E110 (56 KiB): IRAM
```