espressif/esp-idf #2295: Lockup after using esp_aes_crypt_cbc function caused by DPORT ACCESS ECO

comcomservices, 2018-08-14:
## Environment

- Development Kit:      Custom
- Core (if using chip or module): PYCOM W01
- IDF version Latest as of today (30545f4)
- Development Env:	 Make
- Operating System:	Ubuntu
- Power Supply:        3.3V

## Problem Description

Our software crashes at random due to ECO 3.1 from the errata 
https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf

The workaround says that it is automatically handled in IDF 3.1 or newer which we are using..
"This issue is automatically worked around in ESP-IDF V3.0 and newer. "  

Currently our setup uses LWIP on core 0 and http server, etc... and core 1 dedicated to reading an ADC which reads and writes the SPI fifo registers directly.

When using the `esp_aes_crypt_cbc` functions on core 0 (http server) it crashes the program at random (presumably when we are reading the SPI registers at the same time on core 1 that's reading the adc)

### Expected Behavior

Using the aes and sha crypto functions should not cause a crash with this bug since it's supposedly been patched and operation should continue as normal..  Reads will halt the other CPU as needed, etc...

### Actual Behavior

Core locks up and a watchdog timer resets the system

### Steps to repropduce

1.  Setup a task pinned to core 0 that utilises the aes crypto and sha hashing functions
2. Setup a task pinned to core 1 that reads and writs the SPI registers directly
3. Wait a random amount of time until the core locks up..

For what it's worth the code that is writing/reading the spi fifo registers on core 1 it run out of IRAM in interrupt context from a custom allocated interrupt with the following flags
```
 esp_err_t err =  esp_intr_alloc(ETS_GPIO_INTR_SOURCE, ESP_INTR_FLAG_LEVEL3 | ESP_INTR_FLAG_IRAM , gpio_isr_handler, NULL, &ret_handle);
```

### Code to reproduce this issue

## Debug Logs

Crash 1
```
 (9) boot: PRO CPU has been reset by WDT.
W (9) boot: WDT reset info: PRO CPU PC=0x40081bf2
0x40081bf2: esp_dport_access_stall_other_cpu_start at /home/jon/Programming/flexsc2/esp-idf/components/esp32/dport_access.c:100

W (10) boot: WDT reset info: APP CPU PC=0x400993d6
0x400993d6: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:185

I (11) boot: ESP-IDF v3.2-dev-444-gb85e8e1-dirty 2nd stage bootloader
I (17) boot: compile time 13:25:01
I (30) boot: Enabling RNG early entropy source...
I (30) boot: SPI Speed      : 40MHz
I (30) boot: SPI Mode       : DIO
```
Crash 2
```
W (9) boot: PRO CPU has been reset by WDT.
W (9) boot: WDT reset info: PRO CPU PC=0x40081bea
0x40081bea: esp_dport_access_stall_other_cpu_start at /home/jon/Programming/flexsc2/esp-idf/components/esp32/dport_access.c:100

W (10) boot: WDT reset info: APP CPU PC=0x400993d6
0x400993d6: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:185

I (11) boot: ESP-IDF v3.2-dev-444-gb85e8e1-dirty 2nd stage bootloader
I (17) boot: compile time 13:25:01
I (32) boot: Enabling RNG early entropy sou

```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2285513/sdkconfig.txt)

comcomservices, 2018-08-14:
Disabling hardware AES does seem to resolve the issue.. Just a note I would suggest that the option to disable hardware AES be removed from the subtree for MBEDTLS as I would have probably tried that before but thought that was only for the MBEDTLS library... 

Yes it's happening due to the 20second task watchdog timer.. as you describe.. The interrupt watchdog was disabled much earlier due to debugging something else and just hasn't been enabled again..

The ISR was running at 16k times per second, depending on the configuration it may run up to 64k per second.  It usually runs for 20-40us.  The SPI transfer complete register is polled in a loop during one section which may last for as much as (27 bytes at 20MHZ)  (Usually however it is only polled once as we are doing processing while the transfer completes) 

Also the SPI is setup with the standard IDF library.. clocking, modes etc...

```cpp

// SPI read function

static void IRAM_ATTR spiReadADC()
{
    int i;

    int wlen = 7;
    int blen = ((28 * 8) - 1);

    if(adcConfig.sampleRate >= 32000){ // ADC uses 16bit sample size beyond 32k sps, we set the frame size accordingly
        wlen = 5;
        blen = ((20 * 8) - 1);
    }

    ADC_SPI_DEV.user.usr_mosi_highpart=0;

    ADC_SPI_DEV.mosi_dlen.usr_mosi_dbitlen = blen;
    ADC_SPI_DEV.miso_dlen.usr_miso_dbitlen = blen;

    for(i=0; i<wlen; i++) {
        ADC_SPI_DEV.data_buf[i] = ((uint32_t*)result_transaction_txbuffer)[i];    //copy buffer to spi fifo
    }

    ADC_SPI_DEV.cmd.usr = 1; // Kickoff transmission

    if(adcConfig.sampleRate >= 8000){ // Process last reading while waiting for buffer to fill..
        processAdcResult(result_transaction_rxbuffer+1,0); // Process Received Measurement Bytes from ADC, offset the 24bits of status (3 bytes) + RDATA COMMAND (1 byte)
    }

    while(ADC_SPI_DEV.cmd.usr); // Wait for spi to be done

    for(int i=0; i<wlen; i++) {
        ((uint32_t*)result_transaction_rxbuffer)[i] = ADC_SPI_DEV.data_buf[i];//copy spi fifo to buffer
    }

    return;
}

/// Actual interrupt handler

uint32_t cp0_regs[18];

static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    // Enter Critical
    portENTER_CRITICAL_ISR(&afeCriticalMutex);

    getTimeUs(&currTimeUs);

    //GPIO intr process
    uint32_t gpio_num = 0;

    //read status to get interrupt status for GPIO0-31
    uint32_t gpio_intr_status;
    gpio_intr_status = GPIO.status;

    //read status1 to get interrupt status for GPIO32-39
    uint32_t gpio_intr_status_h;
    gpio_intr_status_h = GPIO.status1.intr_st;

    if (gpio_intr_status_h & BIT(GPIO_ADC_DRDY - 32)) {// good for GPIO_NUM < 32

        if(adc_state == 1){

            adc_state = 2;

            /// Save FPU context and state: Workaround to utilize FPU in interrupt context
            uint32_t cp_state = xthal_get_cpenable();

            if(!cp_state) {
                xthal_set_cpenable(1);
            }

            xthal_save_cp0(cp0_regs);   // Save FPU registers

            GPIO.out_w1tc = (1 << GPIO_ADC_CS);
            spiReadADC();
            GPIO.out_w1ts = (1 << GPIO_ADC_CS);

            if(adcConfig.sampleRate < 8000) // If speed is faster then this we process while we're reading the sample
                processAdcResult24(result_transaction_rxbuffer+1,0); // Process Received Measurement Bytes from ADC, offset the RDATA COMMAND (1 byte)

            /// Restore FPU context and state: Workaround to utilize FPU in interrupt context
            xthal_restore_cp0(cp0_regs);

            if(cp_state) {
                // Restore FPU registers

            } else {
                // turn it back off
                xthal_set_cpenable(0);
            }

            getTimeUs(&adcTotalTime);

            adcTotalTime = adcTotalTime - currTimeUs;

            adc_state = 1;

        }
    }

    if (gpio_intr_status_h & BIT(GPIO_ENC_PHA - 32)) { // falling edge on PHA // good for GPIO_NUM < 32

        if((uint64_t)(currTimeUs - lastEncoderChangeUs) > (uint64_t)1000  ){

            BaseType_t xHigherPriorityTaskWoken = pdFALSE; // xHigherPriorityTaskWoken must be initialised to pdFALSE.

            int dir = ((GPIO.in1.data >> (GPIO_ENC_PHB - 32)) & 0x1);

            ///  if( dir)
            ///  xEventGroupSetBitsFromISR(wifiEventGroup, ENCODER_ROTATE_CW_BIT,&xHigherPriorityTaskWoken); // Request an update
            ///   else
            ///  xEventGroupSetBitsFromISR(wifiEventGroup, ENCODER_ROTATE_CCW_BIT,&xHigherPriorityTaskWoken); // Request an update

            lastEncoderChangeUs = currTimeUs;

        }

    }

    if (gpio_intr_status_h & BIT(GPIO_ENC_SW - 32)) { // good for GPIO_NUM < 32

        if(currTimeUs - lastEncoderChangeUs > 1000){
            BaseType_t xHigherPriorityTaskWoken = pdFALSE; // xHigherPriorityTaskWoken must be initialised to pdFALSE.
            ///  xEventGroupSetBitsFromISR(wifiEventGroup, ENCODER_PRESS_BIT,&xHigherPriorityTaskWoken); // Request an update

            //   GPIO.status1_w1tc.intr_st =  BIT(GPIO_ENC_SW - 32);

            lastEncoderChangeUs = currTimeUs;
        }

    }

    // Clear all gpio interrupts.. This function handles everything anyway
    GPIO.status1_w1tc.intr_st = 0xFFFFFFFF;
    GPIO.status_w1tc = 0xFFFFFFFF;

    // Exit critical
    portEXIT_CRITICAL_ISR(&afeCriticalMutex);

}
```

projectgus, 2018-08-15:
> Disabling hardware AES does seem to resolve the issue.. Just a note I would suggest that the option to disable hardware AES be removed from the subtree for MBEDTLS as I would have probably tried that before but thought that was only for the MBEDTLS library...

Sorry, I didn't realise you were calling esp_aes_crypt_cbc() directly rather than via the mbedTLS API.

There's something unexpected here because, as you were right to expect, changing this mbedTLS setting doesn't change anything about `esp_aes_crypt_cbc()` if it's called directly (ie `#include hwcrypto/aes.h`). The setting changes whether the mbedTLS APIs (`mbedtls_aes_crypt_cbc()`) are mapped directly to `esp_aes_crypt_cbc()` or to the mbedTLS software implementation.

It does change some WiFi library behaviour, though (which uses mbedTLS).

The same applies for SHA (using hwcrypto API will always use hardware, but if you use the mbedTLS API then you can switch between software and hardware SHA with a configuration option.)

Are you including any mbedTLS headers in the source files which are calling `esp_aes_crypt_cbc()`?

Can you please refactor to use the mbedTLS APIs and verify any difference with hardware AES and/or SHA disabled in menuconfig?

comcomservices, 2018-08-15:
Just an update.. Unfortunately I had to revert back to using the ESP aes library as the mbedtls library does not seem to operate correctly on data located in the SPIRAM.. Running the encrypt followed by the decrypt function you will get different results..   It could be related to something else but the crypt_cbc functions did not return any error and everything worked before.. Also the mbedtls functions are working fine on the other core that is not working with spiram.. same key and context and they work 100%.. Just the ones that work with the spiram fail... 

Anyway... Tomorrow If i have time i will be able to move all of the functions on core 0 to use the mbedtls functions.. and the one that works with the spiram I will be able to move it to core1 so hopefully no lockups due to ECO 3.1 .. We will use both libraries side by side...

projectgus, 2018-08-15:
Hi @comcomservices ,

Just to check a couple more details:

* When you say mbedTLS is not working correctly, this is with hardware acceleration disabled? That's very unusual! Is it possible there's some other bug, which is manifesting differently when the configuration is changed (for example, buffer overflow somewhere, or a failure to call `mbedtls_aes_init()` (equivalent call would have been `esp_aes_init()`?)
* Are all of the crashes you're seeing on devices using SPIRAM?

comcomservices, 2018-08-15:
Yes with hardware acceleration disabled..

This is not a bug in the usage as we adopted all the functions you from their esp_aes_xxx to the mbedtls_aes_xxx functions including the context object data type...

Also all of the mbedtls functions operating on normal memory are working flawlessly.. also encryption's works flawlessly from spiram, de encryption does not however return the expected results..  its deff not buffer overflow either as we are working on only the start of a very large 3mb buffer which would ever write to the end after many days.. we also have been using the trace function extensively with comprehensive stack checking and have no errors or leaks..

Yes all of the crashes have been on these devices.. all of our devices use spiram

comcomservices, 2018-08-15:
I’m going to do some more testing on all of this tomorrow and will get back to you with more solid concrete info... it’s been a long day here

sebirdman, 2018-10-24:
I'm seeing a similar error when I make an HTTPS request. It doesn't seem to happen as often as @comcomservices 

My IDF is at 22da5f6de
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400812f7  PS      : 0x00060334  A0      : 0x80116c55  A1      : 0x3ffe6cb0
0x400812f7: esp_dport_access_reg_read at /idf/components/esp32/dport_access.c:270

A2      : 0x00000000  A3      : 0x3ffeeb39  A4      : 0x00000020  A5      : 0x3ff01030
A6      : 0x0000004d  A7      : 0xed6ccaf8  A8      : 0x15122500  A9      : 0x00060323
A10     : 0xffffffe6  A11     : 0x00000001  A12     : 0x0000000c  A13     : 0x3ffe6e70
A14     : 0x0000000d  A15     : 0x3ffe83e1  SAR     : 0x00000008  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x400812f7:0x3ffe6cb0 0x40116c52:0x3ffe6cd0 0x4016d7f2:0x3ffe6cf0 0x4016d1b1:0x3ffe6d10 0x4016f91e:0x3ffe6d50 0x4016fb95:0x3ffe6da0 0x4016fbf3:0x3ffe6dc0 0x4016d661:0x3ffe6e20 0x401661ad:0x3ffe6e50 0x40166601:0x3ffe6ef0 0x40166742:0x3ffe6f10 0x4016678f:0x3ffe6f30 0x40166d07:0x3ffe6f50 0x4018b108:0x3ffe6f70 0x4018b40e:0x3ffe6fa0 0x4018b6ed:0x3ffe6fe0 0x4018b791:0x3ffe7000 0x4018a43d:0x3ffe7030 0x4018a4da:0x3ffe7090 0x400e6f22:0x3ffe70b0 0x400e6fd7:0x3ffe70e0 0x400e710c:0x3ffe7250
0x400812f7: esp_dport_access_reg_read at /idf/components/esp32/dport_access.c:270

0x40116c52: DPORT_REG_READ at /idf/components/esp32/hwcrypto/aes.c:347
 (inlined by) esp_aes_block at /idf/components/esp32/hwcrypto/aes.c:136
 (inlined by) esp_aes_crypt_ecb at /idf/components/esp32/hwcrypto/aes.c:194

0x4016d7f2: aes_crypt_ecb_wrap at /idf/components/mbedtls/mbedtls/library/cipher_wrap.c:131

0x4016d1b1: mbedtls_cipher_update at /idf/components/mbedtls/mbedtls/library/cipher.c:85

0x4016f91e: mbedtls_gcm_starts at /idf/components/mbedtls/mbedtls/library/gcm.c:324

0x4016fb95: mbedtls_gcm_crypt_and_tag at /idf/components/mbedtls/mbedtls/library/gcm.c:456

0x4016fbf3: mbedtls_gcm_auth_decrypt at /idf/components/mbedtls/mbedtls/library/gcm.c:484

0x4016d661: mbedtls_cipher_auth_decrypt at /idf/components/mbedtls/mbedtls/library/cipher.c:85

0x401661ad: ssl_decrypt_buf at /idf/components/mbedtls/mbedtls/library/ssl_tls.c:8159

0x40166601: ssl_prepare_record_content at /idf/components/mbedtls/mbedtls/library/ssl_tls.c:8159

0x40166742: mbedtls_ssl_read_record_layer at /idf/components/mbedtls/mbedtls/library/ssl_tls.c:8159

0x4016678f: mbedtls_ssl_read_record at /idf/components/mbedtls/mbedtls/library/ssl_tls.c:8159

0x40166d07: mbedtls_ssl_read at /idf/components/mbedtls/mbedtls/library/ssl_tls.c:8159

0x4018b108: iot_tls_read at /idf/components/aws_iot/port/network_mbedtls_wrapper.c:359 (discriminator 12)

0x4018b40e: _aws_iot_mqtt_internal_read_packet at /idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:364

0x4018b6ed: aws_iot_mqtt_internal_cycle_read at /idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:569

0x4018b791: aws_iot_mqtt_internal_wait_for_read at /idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:633

0x4018a43d: _aws_iot_mqtt_internal_publish at /idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_publish.c:230

0x4018a4da: aws_iot_mqtt_publish at /idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_publish.c:287

0x400e6f22: prv_publish_message at /app/components/aws_mqtt_client/aws_mqtt_client.c:297

0x400e6fd7: prv_send_enqueued_msg at /app/components/aws_mqtt_client/aws_mqtt_client.c:297

0x400e710c: prv_mqtt_task at /app/components/aws_mqtt_client/aws_mqtt_client.c:297
```

comcomservices, 2018-10-25:
@sebirdman

Looks very similar...  I wish I could tell you that we found a solution to it but unfortunately it would still randomly happen...  we spent months and months chasing all sorts of crashes and just when we thought we had a stable system we would run into something new...  having fallen behind deadlines we finally made the tough decision to throw away much of our work and switch platforms...  We have some of the most skilled embedded software developers in Canada working for us and we worked very hard to get to the bottom of issues but it seemed never ending... I would advice you to make sure that crashes are handled properly and that you can recover from them without hiccup... honestly the development system is amazing, one of the best I’ve used.. I’m very hopeful that things will straighten out but since our product was for industrial control we just couldn’t keep working on something that wasn’t finished yet...  I think it will be another year or so before all the issues are worked through but handling crashes is always important and should be implemented anyway.... 

Sorry @espressif for being a bit down on things, please feel free to delete this if you wish... I personally have allot of faith in the product and will definitely consider it for future products since our team knows it so well.. it was no surprise that the product and tools are under development, I think we just underestimated the impact it would have to us...  you guys have all been super helpful though

closed 2018-11-18 by igrr