espressif/esp-idf #2296: new https ota example giving mbed tls handshake error -0x2700

akbarhash, 2018-08-14:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DOIT Board
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    30545f4c // Latest IDF
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

The simple OTA example is not working even after many different trials.
Tried using python server and openssl server
works with http by modifying the code slightly

### Expected Behavior
should not give certificate error
### Actual Behavior

E (6804) TRANS_SSL: mbedtls_ssl_handshake returned -0x2700
E (6804) HTTP_CLIENT: Connection failed, sock < 0
E (6804) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6814) simple_ota_example: Firmware Upgrades Failed

### Steps to reproduce
install ota example
generate cert
setup server and example using cert
run the example

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
void simple_ota_example_task(void * pvParameter)
{
    ESP_LOGI(OTA_TAG, "Starting OTA example...");

    /* Wait for the callback to set the CONNECTED_BIT in the
       event group.
    */
    xEventGroupWaitBits(wifi_event_group, WIFI_CONNECTED_BIT, false, true, portMAX_DELAY);
    ESP_LOGI(OTA_TAG, "Connected to Wifi ! Start to Connect to Server....");
    
    esp_http_client_config_t config = {
        .url = CONFIG_FIRMWARE_UPGRADE_URL,
        .cert_pem = (char *)server_cert_pem_start,
        .event_handler = _http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);
    if (ret == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(OTA_TAG, "Firmware Upgrades Failed");
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
E (6804) TRANS_SSL: mbedtls_ssl_handshake returned -0x2700
E (6804) HTTP_CLIENT: Connection failed, sock < 0
E (6804) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6814) simple_ota_example: Firmware Upgrades Failed

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

akbarhash, 2018-08-14:
``D (15950) HTTP_CLIENT: Begin connect to: https://192.168.0.87:8070
D (15970) TRANS_SSL: Connect to 192.168.0.87:8070
D (16180) TRANS_SSL: Performing the SSL/TLS handshake...
I (16180) mbedtls: ssl_tls.c:6954 => handshake

I (16180) mbedtls: ssl_cli.c:3384 client state: 0

I (16190) mbedtls: ssl_tls.c:2649 => flush output

I (16190) mbedtls: ssl_tls.c:2661 <= flush output

I (16200) mbedtls: ssl_cli.c:3384 client state: 1

I (16200) mbedtls: ssl_tls.c:2649 => flush output

I (16210) mbedtls: ssl_tls.c:2661 <= flush output

I (16210) mbedtls: ssl_cli.c:770 => write client hello

D (16220) mbedtls: ssl_cli.c:808 client hello, max version: [3:3]

D (16230) mbedtls: ssl_cli.c:703 client hello, current time: 1534228243

D (16230) mbedtls: ssl_cli.c:817 dumping 'client hello, random bytes' (32 bytes)

D (16240) mbedtls: ssl_cli.c:817 0000:  5b 72 77 13 6e 02 f5 d4 b3 45 11 a5 80 e0 15 34  [rw.n....E.....4

D (16250) mbedtls: ssl_cli.c:817 0010:  1e 6f b1 9b fc 8f ca db 1f c2 94 d7 7b 4c 49 f2  .o..........{LI.

D (16260) mbedtls: ssl_cli.c:870 client hello, session id len.: 0

D (16270) mbedtls: ssl_cli.c:871 dumping 'client hello, session id' (0 bytes)

D (16270) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02c

D (16280) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c030

D (16290) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 009f

D (16290) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0ad

D (16300) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c09f

D (16310) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c024

D (16310) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c028

D (16320) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 006b

D (16330) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c00a

D (16330) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c014

D (16340) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 0039

D (16350) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0af

D (16350) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0a3

D (16360) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02b

D (16370) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02f

D (16370) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 009e

D (16380) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0ac

D (16390) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c09e

D (16390) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c023
[... 425 lines cut ...]

I (18090) mbedtls: ssl_tls.c:2674 ssl->f_send() returned 7 (-0xfffffff9)

I (18100) mbedtls: ssl_tls.c:2701 <= flush output

I (18110) mbedtls: ssl_tls.c:3100 <= write record

I (18110) mbedtls: ssl_tls.c:4381 <= send alert message

D (18120) mbedtls: ssl_tls.c:4925 ! Certificate verification flags 10004

I (18130) mbedtls: ssl_tls.c:4934 <= parse certificate

I (18130) mbedtls: ssl_tls.c:6964 <= handshake

E (18140) TRANS_SSL: mbedtls_ssl_handshake returned -0x2700
D (18150) TRANS_SSL: Cleanup mbedtls
I (18150) mbedtls: ssl_tls.c:7592 => write close notify

I (18160) mbedtls: ssl_tls.c:7608 <= write close notify

E (18160) HTTP_CLIENT: Connection failed, sock < 0
D (18170) simple_ota_example: HTTP_EVENT_DISCONNECTED
E (18170) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (18180) simple_ota_example: Firmware Upgrades Failed``

jitin17, 2018-08-14:
@akbarhash Your certificate seems to be invalid. Can you try following the instructions given in README.md located at `examples/system/ota/README.md`.

akbarhash, 2018-08-14:
I have done everything again.
I have a doubt in this case what should be 
`Common Name` of server certificate should be host-name of your server.

jitin17, 2018-08-14:
If your firmware upgrade URL looks something like `https://<host-ip-address>:<host-port>/<firmware-image-filename>` this, say `https://192.168.0.3:8070/hello-world.bin`, then  `host-ip-address` will be the `Common Name`, say `192.168.0.3`.

akbarhash, 2018-08-14:
`D (13450) HTTP_CLIENT: Begin connect to: https://192.168.0.87:8070
D (13460) TRANS_SSL: Connect to 192.168.0.87:8070
D (13470) TRANS_SSL: Performing the SSL/TLS handshake...
I (13480) mbedtls: ssl_tls.c:6954 => handshake

I (13480) mbedtls: ssl_cli.c:3384 client state: 0

I (13490) mbedtls: ssl_tls.c:2649 => flush output

I (13500) mbedtls: ssl_tls.c:2661 <= flush output

I (13500) mbedtls: ssl_cli.c:3384 client state: 1

I (13510) mbedtls: ssl_tls.c:2649 => flush output

I (13510) mbedtls: ssl_tls.c:2661 <= flush output

I (13520) mbedtls: ssl_cli.c:770 => write client hello

D (13520) mbedtls: ssl_cli.c:808 client hello, max version: [3:3]

D (13530) mbedtls: ssl_cli.c:703 client hello, current time: 13

D (13540) mbedtls: ssl_cli.c:817 dumping 'client hello, random bytes' (32 bytes)

D (13540) mbedtls: ssl_cli.c:817 0000:  00 00 00 0d a1 d4 aa e0 09 2f 60 97 1f 05 54 ec  ........./`...T.

D (13550) mbedtls: ssl_cli.c:817 0010:  a2 a7 cc 52 69 62 fd 13 9b fb 3d aa 93 4a f7 24  ...Rib....=..J.$

D (13560) mbedtls: ssl_cli.c:870 client hello, session id len.: 0

D (13570) mbedtls: ssl_cli.c:871 dumping 'client hello, session id' (0 bytes)

D (13580) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02c

D (13580) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c030

D (13590) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 009f

D (13600) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0ad

D (13600) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c09f

D (13610) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c024

D (13620) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c028

D (13620) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 006b

D (13630) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c00a

D (13640) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c014

D (13640) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 0039

D (13650) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0af

D (13660) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0a3

D (13660) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02b

D (13670) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c02f

D (13680) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: 009e

D (13680) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c0ac

D (13690) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c09e

D (13700) mbedtls: ssl_cli.c:918 client hello, add ciphersuite: c023
[... 424 lines cut ...]

I (15390) mbedtls: ssl_tls.c:2674 ssl->f_send() returned 7 (-0xfffffff9)

I (15400) mbedtls: ssl_tls.c:2701 <= flush output

I (15400) mbedtls: ssl_tls.c:3100 <= write record

I (15410) mbedtls: ssl_tls.c:4381 <= send alert message

D (15420) mbedtls: ssl_tls.c:4925 ! Certificate verification flags 10000

I (15420) mbedtls: ssl_tls.c:4934 <= parse certificate

I (15430) mbedtls: ssl_tls.c:6964 <= handshake

E (15430) TRANS_SSL: mbedtls_ssl_handshake returned -0x2700
D (15440) TRANS_SSL: Cleanup mbedtls
I (15440) mbedtls: ssl_tls.c:7592 => write close notify

I (15450) mbedtls: ssl_tls.c:7608 <= write close notify

E (15460) HTTP_CLIENT: Connection failed, sock < 0
D (15460) simple_ota_example: HTTP_EVENT_DISCONNECTED
E (15470) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (15470) simple_ota_example: Firmware Upgrades Failed`

akbarhash, 2018-08-14:
```openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070
Enter pass phrase for ca_key.pem:
Using default temp DH parameters
ACCEPT
139642142949824:error:14094413:SSL routines:ssl3_read_bytes:sslv3 alert unsupported certificate:../ssl/record/rec_layer_s3.c:1399:SSL alert number 43
ACCEPT
139642142949824:error:14094413:SSL routines:ssl3_read_bytes:sslv3 alert unsupported certificate:../ssl/record/rec_layer_s3.c:1399:SSL alert number 43
ACCEPT```
server side logs

akbarhash, 2018-08-14:
got it to work the issue was i was using 1024 rsa had changed it during testing. now working with 2048 rsa and common name as 192.168.0.87

akbarhash, 2018-08-14:
suggestion would be to include this bit in the readme.md
thanks jitin

closed 2018-08-14 by akbarhash

Abhijit123Gupta, 2019-06-04:
how do you change the rsa?

Sharjeel77474, 2023-10-09:
Try using chain certificate format... not single certificate format. I also faced the  same error because my certificate was single certificate.. redownload the certificate in chain certificate format and if your code is fine, i hope you won't see the error again.

Sharjeel77474, 2023-10-09:
Try using chain certificate format... not single certificate format. I also faced the same error because my certificate was single certificate format.... redownload the certificate in chain certificate format and if your code is fine, i hope you won't see the error again.