espressif/esp-idf #2297: eclipse for windows, makefile & eclipse_make.py

dsptech, 2018-08-14:
NOT AN ERROR OR SYSTEM ISSUE

To solve the issue #2170 and other problems as in #18 (invalid paths detection, or not necessary paths found) I added a makefile target to extract search paths, and compiler settings too, directly from components / esp-idf configurations (allowing the target call from the built-in setting provider).

I'm dealing now on eclipse_make.py because it's output message "running make ...<path>" cause a false include path detection.
Is there a way to suppress that output without changing the pyton code ? 
(note: I cannot suppress the standard and error outputs because them are used by the provider).

igrr, 2018-08-23:
@dsptech Probably easiest would be to change Python code to either not print the full path, not print this message at all, or replace slashes in the path with something else. Another option might be to print to `stderr` instead — perhaps this will not confuse Eclipse build output parser?

dsptech, 2018-08-23:
Hi @igrr ,
I agree, any output to stderr will continue to confuse the parser. Suppressing the message at all is the better way.

My original intention was to reduce the changes in the library due the increasing of the required checks at every update but, as you said, changing the pyton code is the easiest way. 
So, I will do it. 

Thank you.
Regards.

closed 2018-09-04 by dsptech