espressif/esp-idf #2298: Bluetooth Controller: Suggested Default Data Length not used for new connections

mringwal, 2018-08-14:
During Bluetooth init, BTstack sends the LE Write Suggested Default Data Length Command
 and specifies 251 octets and 2120 ms. However, when creating an outgoing connection, the LE Data Length does not change/no negotiation happens.

I expect that setting the default data length is enough to trigger DLE negotiation for new connections. This is also observed with other controllers: the EM9301, CYW20704, and Nordic nRF52/SoftDevice.

The bundled Bluedroid stack explicitly sends the LE Set Data Length Command for new connections, I guess this fact hides this issue. I've tried this as well and the Data Length Update works when this command is send.

Is this considered a bug by Espressif? If yes, will Espressif fix it?

Weijian-Espressif, 2018-08-23:
Hi @mringwal , ESP32 will ESP32 auto-negotiation data length if remote device feature support ,
you can refer to line 3655 in **btm_ble_read_remote_features_complete()** of btm_ble_gap.c.

mringwal, 2018-08-23:
Hi @Weijian-Espressif -  I'm referring to the Bluetooth Controller implementation in the Bluetooth Library not the Host Stack.

I expect the Bluetooth Controller to automatically trigger DLE negotiation. btm_ble_gap.c is part of Bluedroid / the Bluetooth Host Stack. If it is triggered in the host stack, it is not automatically started by the Bluetooth Controller.

(Of course, we can add a workaround in our stack, but I think the behaviour is not correct and like to know that's the case and if it gets fixed, so that we won't need to add such a workaround)

Weijian-Espressif, 2018-08-31:
We found a compatibility issue before, so I moved it to the host. There are no plans to change now.

mringwal, 2018-08-31:
Thanks for the updates and the clarification. 

I'll add an option to BTstack to start the negotiation after connect to work around this problem in the ESP32 Bluetooth Controller implementation.