espressif/esp-idf #2299: How to correctly configure mcpwm or led_pwm above 200kHz frequency

fangms, 2018-08-15:
Hello,
I want two PWM frequency outputs at 200KHz, and the two PWM levels should be reversed.

Code to reproduce this issue:
  mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, GPIO_PWM0A_OUT);
  mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0B, GPIO_PWM0B_OUT);
  mcpwm_config_t pwm_config;
  pwm_config.frequency = 200000;    // 200kHz
  pwm_config.cmpr_a = 50.0;       //duty cycle of PWMxA = 50.0%
  pwm_config.cmpr_b = 50.0;       //duty cycle of PWMxb = 50.0%
  pwm_config.counter_mode = MCPWM_UP_COUNTER;
  pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
  mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);
  mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_1); 
  mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_0); 
  mcpwm_deadtime_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_BYPASS_FED, 1, 1);  

I expected that this leads to a 200 kHz frequency, But the result is a 167 kHz frequency,Why the factor? How do I configure the correct 200kHz frequency PWM?

negativekelvin, 2018-08-15:
Opposite of this issue

https://github.com/espressif/esp-idf/issues/2255

costaud, 2018-08-17:
Do you just need a 200Khz clock? 

Do you need to adjust the duty cycle? If so, what resolution do you need ?

michprev, 2018-09-19:
@costaud 

> Do you just need a 200Khz clock?
> 
> Do you need to adjust the duty cycle? If so, what resolution do you need ?

The right question is why MCPWM driver does not support this when hardware allows it?

koobest, 2018-10-29:
Hi, @fangms in mcpwm.c, you can modify the following two lines:
`
#define MCPWM_CLK_PRESCL 2       //MCPWM clock prescale 
#define TIMER_CLK_PRESCALE 14      //MCPWM timer prescales
`
in this configuration, it can output 198Khz of pwm.

Alvin1Zhang, 2019-03-14:
@fangms has your issue been resolved? @koobest has the driver been configurable now? Thanks

kaidokert, 2019-12-16:
Would be really nice if this was runtime-configurable. Just ran into an issue where with a pretty standard 20khz PWM, there isn't much resolution left to control duty with the default prescalers.

collin80, 2020-06-30:
They said they'd support configuring the PWM prescaler values at runtime but they have not, in fact, done so. If I weren't so lazy I'd do it myself and submit the pull request. But, instead I think I'll do what everyone else did and change the hard coded values locally and be done with it. But, really somebody should fix this issue and submit a pull request so it quits happening. Will that someone be me, I don't know yet.

closed 2021-05-07 by tom-borcin