espressif/esp-idf #2301: Implementation of fstat() and stat() with respect to stdin/out/err

ammaree, 2018-08-15:
## Environment
- Development Kit:      [Custom]
- Core (if using chip or module): [ESP32-WROOM-32U]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [external 5V]

## Problem Description
I am trying to identify the buffering status of stdin/out/err using fstat() or stat()

### Expected Behavior
[f]stat() should return values that make sense for some/all the fields in struct stat.

### Actual Behavior
values returned are not consistet and/or does not make sense.

### Steps to reproduce

### Code to reproduce this issue

```cpp
struct stat fs ;
int32_t iRV = fstat(stdin->_file, &fs) ;
printf( "dev=%d ....", fs.st_dev,  ...) ;

```
## Other items if possible

igrr, 2018-08-23:
Hi @ammaree,

What is the return value of `fstat`? If it is -1, then you need to check `errno` to determine the error cause. It should be `ENOSYS`, indicating that this function (fstat) is not supported for this file descriptor.

Now, it's entirely reasonable suggestion to actually implement `fstat` for UART VFS. It's probably not going to return any particularly useful values, but at least will give the impression of working.
However there is no buffering happening at the level of `write` or `read` functions of this VFS.

Buffering does happen (by default) in stdio layer. You can change buffering for the given stream using `setvbuf` function. However as far as I know there is no standard (or non-standard) function to query the buffer size of a stream. If you don't mind peeking into the private members of FILE structure, you can get this information: https://github.com/espressif/newlib-esp32/blob/4ddc99d47677b94c115cafcaa4a6939042ea8c4f/newlib/libc/include/sys/reent.h#L185-L186. However as you see, at the layer where operations on file descriptors are performed (e.g. `fstat`) it's not possible to access this private state of the corresponding FILE structure.

ammaree, 2018-08-23:
Hi @igrr thanks for the response. Correct, the errno is ENOSYS, as I found out in #2318 when trying to use dup2() or close()/reopen() to do redirection of stdio.

Ultimately this issue can be closed, provided we can resolve the feature request in #2318 that would remove the need for fstat(), dup2() and reopen()

ammaree, 2018-08-27:
Will follow progress on #2318

closed 2018-08-27 by ammaree