espressif/esp-idf #2305: localtime(_r) inversed behavior

xarrange, 2018-08-16:
## Environment

- Core (if using chip or module): [ESP-WROOM32]
- IDF version c1fdd450
- Power Supply:         [USB]

## Problem Description

Hours value that is returned by localtime(_r) function is less then UTC time when time zone offset is positive and vice versa.
When UTC time is 16:05 and timezone is UTC+6:00 localtime should return 22:05, but it returns 10:05
and when timezone is UTC-6:00 localtime returns 22:05, while correct is 10:05

### Code to reproduce this issue

```cpp
void test_task(void* pvParameters) {	
	struct tm l;	
	time_t stamp = 1534435540;

	// No TZ set
	ets_printf("TZ=%s\n",getenv("TZ"));
	gmtime_r(&stamp, &l);
	ets_printf("UTC: %0u:%0u:%0u - %0u.%0u.%0u\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
	localtime_r(&stamp, &l);
	ets_printf("Local: %0u:%0u:%0u - %0u.%0u.%0u\n\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
	
	// Positive TZ: east to Greenwich (e.g. China)
	setenv("TZ", "UTC+6:00", 1);	
	ets_printf("TZ=%s\n",getenv("TZ"));		
	tzset();	
	gmtime_r(&stamp, &l);
	ets_printf("UTC: %0u:%0u:%0u - %0u.%0u.%0u\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
	localtime_r(&stamp, &l);
	ets_printf("Local: %0u:%0u:%0u - %0u.%0u.%0u\n\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
	
	// Negative TZ: west to Greenwich (e.g. US/Canada)
	setenv("TZ", "UTC-6:00", 1);	
	ets_printf("TZ=%s\n",getenv("TZ"));
	tzset();	
	gmtime_r(&stamp, &l);
	ets_printf("UTC: %0u:%0u:%0u - %0u.%0u.%0u\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
	localtime_r(&stamp, &l);
	ets_printf("Local: %0u:%0u:%0u - %0u.%0u.%0u\n\n",l.tm_hour,l.tm_min,l.tm_sec,l.tm_mday,l.tm_mon+1,l.tm_year+1900);
		
	while(1){
		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
}

void app_main(void)
{	
	xTaskCreate(test_task, "test", 2048, 0, 7, NULL);		
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
TZ=<null>
UTC: 16:5:40 - 16.8.2018
Local: 16:5:40 - 16.8.2018

TZ=UTC+6:00
UTC: 16:5:40 - 16.8.2018
Local: 10:5:40 - 16.8.2018

TZ=UTC-6:00
UTC: 16:5:40 - 16.8.2018
Local: 22:5:40 - 16.8.2018

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

closed 2018-08-17 by xarrange

marcogmaia, 2019-07-15:
I have the same problem, why you closed the issue? Have you solved?

marcogmaia, 2019-07-15:
Oh, from rtfm [GNU TZ-variable](https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html) it says 

> The offset specifies the time value you must add to the local time to get a Coordinated Universal Time value.

So it's alright, our bad.