espressif/esp-idf #2306: Bluetooth visibility in dual mode

crespum, 2018-08-17:
## Environment
- IDF version: I've tried both 53509c7b (v3.1-rc) and latest v3.0.3
- Development Env:	Make
- Operating System:	 Fedora
- Power Supply: USB

## Problem Description
My application uses Bluetooth dual mode, to be precise it uses A2DP and a BLE beacon, but I'm experiencing some issues with the visibility of the different modes: 

I want the beacon to be always visible but not the A2DP profile. The latter should only be visible for an small amount of time. The A2DP profile should remain connectable for the devices which have already been bond.

To achieve this I've tried the function `esp_bt_gap_set_scan_mode`, which according to the docs should "set discoverability and connectability mode for _legacy_ bluetooth". However, after calling `esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE)`, the BLE advertisement completely disappears.

Why does `esp_bt_gap_set_scan_mode` affect the BLE stack?

### Code to reproduce this issue
To test the issue you can apply the following patch to [a2dp_gatts_coex](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_gatts_coex) example. The advertisement should be visible the first 30 seconds and disappear right after the logs print `Beacon disappears now`.

```c
102c102
<     .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
---
>     .own_addr_type      = BLE_ADDR_TYPE_RANDOM,
258a259
>         esp_ble_gap_config_local_privacy(true);
380,381c381,382
<         // esp_ble_conn_update_params_t conn_params = {0};
<         // memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
---
>         esp_ble_conn_update_params_t conn_params = {0};
>         memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
623c624
<         esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE);
---
>         esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
705,708d705
< 
<     vTaskDelay(30000 / portTICK_PERIOD_MS);
<     ESP_LOGI(BT_BLE_COEX_TAG, "Beacon disappears now");
<     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
```

TianHao-Espressif, 2018-09-03:
@crespum, thank you. It's real a problem. It's due to that when call esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE), it will also set hci cmd to disable advertising. 

We will fix it soon.

Alvin1Zhang, 2018-09-17:
@crespum Have you verified the fix patch? Does the patch resolve your issue? Thanks.

crespum, 2018-11-21:
This issue has been fixed in [v3.1.1](https://github.com/espressif/esp-idf/releases/tag/v3.1.1). Thanks @igrr!

closed 2018-11-21 by crespum