espressif/esp-idf #2310: Segfault when running mconf

billr1, 2018-08-19:
## Environment

- Development Kit:      ESP32-DevKit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM-32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 020ade65
- Development Env:		Make
- Operating System:		Linux (Mageia 6)
- Power Supply:         USB

## Problem Description
Segfault when running mconf

This happens when I run 'make menuconfig' for the hello_world example.

### Expected Behavior
Expect to see the config menu.

### Actual Behavior
MENUCONFIG
^[[?1049h^[[1;24r^[(B^[[m^[[4l^[[?7h^[[39;49m^[[?1h^[=/bin/sh: line 1:  5011     Segmentation fault    (core dumped)

### Steps to repropduce

1. make clean
2. make menuconfig

### Code to reproduce this issue
n/a

## Debug Logs

```
$ gdb /home/billr/esp/esp-idf/tools/kconfig/mconf
(gdb) run /home/billr/esp/esp-idf/Kconfig
Program received signal SIGSEGV, Segmentation fault.
                                                    0x00007ffff7bbd992 in waddch_literal (ch=<optimized out>, win=0x73f470)
    at ../../ncurses/base/lib_addch.c:389
389	    line->text[x++] = ch;
(gdb) bt
#0  0x00007ffff7bbd992 in waddch_literal (ch=<optimized out>, win=0x73f470)
    at ../../ncurses/base/lib_addch.c:389
#1  waddch_nosync (ch=32, win=0x73f470) at ../../ncurses/base/lib_addch.c:441
#2  waddch (win=0x73f470, ch=32) at ../../ncurses/base/lib_addch.c:547
#3  0x0000000000415a7a in attr_clear ()
#4  0x0000000000415b33 in dialog_clear ()
#5  0x0000000000415e99 in init_dialog ()
#6  0x000000000040422b in main ()
(gdb)  l
384	    });
385	
386	    /*
387	     * Single-column characters.
388	     */
389	    line->text[x++] = ch;
390	    /*
391	     * This label is used only for wide-characters.
392	     */
393	    if_WIDEC(
(gdb) q

$ ldd /home/billr/esp/esp-idf/tools/kconfig/mconf
	linux-vdso.so.1 (0x00007ffcf1283000)
	libncurses.so.6 => /lib64/libncurses.so.6 (0x00007f3ac90f5000)
	libtinfo.so.6 => /lib64/libtinfo.so.6 (0x00007f3ac8ebb000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f3ac8b08000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f3ac8904000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3ac931f000)

$ rpm -qa |grep ncurses
lib64ncurses6-6.0-8.3.mga6
lib64ncurses-devel-6.0-8.3.mga6
...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

Other programs built with the same ncurses library (such as the Linux kernel version of mconf) work fine.

projectgus, 2018-08-28:
Hi @billr1,

Thanks for providing all this detail.

Just to check one possibility, menuconfig has never worked on this system? Otherwise a "make clean", "make menuconfig" might rebuild the mconf binary successfully if something significant changed in the system library config.

Otherwise, I'm not familiar with Mageia or Mandriva Linux, so this may be the wrong thing to look at, but I'm surprised that it's linking ncurses not ncursesw. Are these equivalent on your system?

Does Mageia use pkgconfig? If you run
```
pkg-config --libs ncurses
pkg-config --libs ncursesw
```
... do you see different libraries?

Similarly, if you run
```
$IDF_PATH/tools/kconfiglxdialog/check-lxdialog.sh -ldflags
```
What does this print?

What LOCALE is set, normally? If you use `LOCALE=C make menuconfig`, does this work?

Angus

billr1, 2018-08-28:
Mageia is a fork of the old MandrIva, nee Mandrake Linux. It is an rpm-based distro, and yes, uses pkg-confg. Here's the answers to the info you requested.

The esp-idf menuconfig has never worked. I did a 'make clean; make' several times and it builds without complaints.

$ pkg-config --libs ncurses
-lncurses -ltinfo
$ pkg-config --libs ncursesw
Package ncursesw was not found in the pkg-config search path.
$ urpmq -y ncursesw  <-- Mageia's equivalent of 'dnf search' on Fedora
lib64ncurses-devel
lib64ncurses5
lib64ncurses6
lib64ncursesw-devel
lib64ncursesw5
lib64ncursesw6
  ...

So, I installed the lib64ncursesw-devel and lib64ncurses6 packages, did a make clean; make menuconfig - and it worked!

So, I guess it turns out the be user error, not having the version of ncurses installed. Thank you so much for your help. This issue can be closed now.

closed 2018-08-28 by billr1