espressif/esp-idf #2311: Heap leak due to wifi connections and disconnections

menesesleonardo, 2018-08-20:
## Environment

- Core : [ESP32-Wrover-I | ESP-WROOM32]
- IDF version (3.0.3-dirty, with esp_error_to_name from master)

- Development Env: [Make | Eclipse]
- Operating System: [Windows | Ubuntu]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

For my deployment, I want to get some sensor data and send it to the web. However, the environment doesn't have a reliable wifi connection, so I opted to create an offline mode where data is still saved and when wifi connection is available, retrieve it. There's no problem there.

The sequence goes as follows.
1. disconnection.
2. make 10 connection attempts.
3. "kill" the whole connection.
4. create again the connection.
5. make another 10 connection attempts.
6. "kill" again the connection and resign for offline mode.
7. start long delay
8. create again the connection.
REPEAT FROM 2!

The whole process works fine. However, heap memory is being lost between each killing and recreation. You can see from long delay to long delay that around 250 bytes are lost each time. I tried every single function to stop and kill wifi and tcpip in "wifi killer" as well as every single one to create a connection in "wifi enforcer". 

To reproduce this problem, just create a hotspot in your phone and then stop it, or put a fake SSID and password  in connection parameters and see the magic happen.

All help is appreciated!!
(please don't mind the function names! ;-p)

https://gist.github.com/menesesleonardo/0ae9d476220729b314f6b0b293a17b90

TimXia, 2018-08-21:
@menesesleonardo, do you mean memory leak due to TCP connection and disconnection? If so, please refer to the answer below.
https://github.com/espressif/esp-idf/issues/2134#issuecomment-403424523

menesesleonardo, 2018-08-23:
hey there. I set maximum sockets to 5 and long delay to 10 minutes. After 200 minutes memory is still decreasing.

liuzfesp, 2019-11-19:
Hi @menesesleonardo does this issue still exist?

liuzfesp, 2020-04-15:
Hi @menesesleonardo the TCP connection create/deleting is a basic regression test in our automatic test case suites, we never see the issue as you pointed out.

I guess what you observed is the memory used for sockets lock. According our current implementation, for specific socket ID, LWIP creates a FreeRTOS lock the 1st time we create the socket and never freed, that's not memory leak. 

For example, for socket ID X, when LWIP creates socket with ID X, it allocates a FreeRTOS lock for it, when socket ID X is closed, the lock for the socket will NOT be freed. Then when we reuse the socket ID X and create socket again, we don't need to create the lock again.

I would like to close this issue, please feel free to reopen it if this issue still bother you.

closed 2020-04-15 by Alvin1Zhang