espressif/esp-idf #2314: BLE. gatt_rsp_timeout retry discovery primary service

KryvosheyaIvan, 2018-08-21:
Hello! I use BLE on esp32. In the past I connected with CC2640 GATT server. But then the developer change something and now I cant connect to it, though android application does. As a matter of fact, I do not know what has changed. But I need to solve problem.

I connect to GATT server (CC2640) as a client. I use the following sequence of actions:
first I use esp_ble_gattc_open() and then when I get the appropriate event - esp_ble_gattc_send_mtu_req(). There is OPEN event, but there are no MTU_CFG event.
Of course, as a basis I took standard gatt_client example.

There are loggs:

`I (34061) F_LOCK_x: OPEN_EVT[0]
I (34061) F_LOCK_x: Conn id handler 0
I (34061) F_LOCK_x: mtu 23
I (34061) F_LOCK_x: status 0
I (34061) F_LOCK_x: f0 f8 f2 6f 76 da
I (34061) F_LOCK_x: Lock opened!

E (39061) BT: gatt_rsp_timeout retry discovery primary service
E (44061) BT: gatt_rsp_timeout retry discovery primary service
E (49061) BT: gatt_rsp_timeout disconnecting...
E (49061) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (49061) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
E (49071) BT: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0016
E (49081) BT: bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0016
E (49091) BT: bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x0016
E (49091) BT: bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0016
E (49101) F_LOCK_x: CLOSE_EVT

I could not find what "gatt_rsp_timeout retry discovery primary service" mean?
I guess, that gatt_server does not respond in time, but I am not sure. Also, by means of android app, I found out that the advertising interval was changed from 100ms to 400ms. I guess CC2640 developer changed some parameters to reduce power consumption. I thought that, using esp_ble_gap_set_prefer_conn_params() or/and esp_ble_gap_update_conn_params() could help. But for now it is not succeeded.

I will appreciate any help.

Weijian-Espressif, 2018-08-31:
Hi KryvosheyaIvan:
     From the log,  ESP32 don't receive response of CC2640 when discover service . I have a BLE board of CC2640. Can you provide your code of CC2640.

Weijian-Espressif, 2018-08-31:
hi @KryvosheyaIvan please tell me the test procedure, thanks.

KryvosheyaIvan, 2018-08-31:
@Weijian-Espressif , it looks like firmware on CC2640 was broken. When it was updated then everything worked fine. Thanks for your will to help!

closed 2018-08-31 by KryvosheyaIvan

locpq, 2019-04-08:
Hi @KryvosheyaIvan 
Hi have probblem same this, but i can connect and read/write data from/to cc2640r2. However with esp32, after esp_ble_gattc_send_mtu_req() return 0 (ok). But can't go to ESP_GATTC_CFG_MTU_EVT.

Thanks and best regards,

KryvosheyaIvan, 2019-04-08:
@locpq , as a matter of fact, current problem was partially solved by changing some value in cc2640 concerning MTU_SIZE. After that kind of action, connection was pretty fine. Unfortunately, for now, I cant tell what exactly variable (#define) was changed.

locpq, 2019-04-09:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Dear Ivan,I fix, i use newest stack for cc2640r2. The probblem from service table when i port from older stack.Thanks you so much,                                                                                                                                                                        Sent from my BlackBerry — the most secure mobile device                                                                                                                                                                           From: notifications@github.comSent: 8 April 2019 18:16To: esp-idf@noreply.github.comReply to: reply@reply.github.comCc: loc.phamquoc@gmail.com; mention@noreply.github.comSubject: Re: [espressif/esp-idf] [TW#25826] BLE. gatt_rsp_timeout retry discovery primary service (IDFGH-931) (#2314)  @locpq , as a matter of fact, current problem was partially solved by changing some value in cc2640 concerning MTU_SIZE. After that kind of action, connection was pretty fine. Unfortunately, for now, I cant tell what exactly variable (#define) was changed.