espressif/esp-idf #2315: wifi off deep sleep current

denyip, 2018-08-22:
I tried to run a simple wifi program, init wifi ->then connect to router->wifi stop->deep sleep, the deep sleep current is around 1.x mA. However, if I don't run the wifi procedure, just start and then deep sleep, the current is around 5.x uA, is it normal? any procedure missing on the wifi routine b4 entering deep sleep mode?

igrr, 2018-08-22:
Yes, you need to shut down Wi-Fi driver before entering deep sleep. Please see https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#wifi-bt-and-sleep-modes.

denyip, 2018-08-23:
I have run esp_wifi_stop(), and it successfully entered event SYSTEM_EVENT_STA_STOP, and then I run esp_deep_sleep(1000000LL * deepSleepSec), but the result is still around 1.5mA, if I do not start wifi, but just run esp_deep_sleep, the current is 5.x uA.

igrr, 2018-08-23:
Hi @denyip, unfortunately I wasn't able to reproduce your result. I have tested with `simple_wifi` example from IDF, and added two lines to the `SYSTEM_EVENT_STA_GOT_IP` event handler:

```c++
    esp_wifi_stop();
    esp_deep_sleep_start();
```

With this, once the device connects to the AP, it goes into deep sleep, and the current is 5.5uA as expected. In this example, WiFi is being used in Station mode. Can you give a hint what your WiFi configuration is? Perhaps you can try the same example with above mentioned modification, and see whether you get same current readings?

denyip, 2018-08-24:
Hi Ivan,
It may due to the rtc gpio. Below is my code. I found that when I turn on the wifi and then turn off and and then init the rtc gpio and then deep sleep, the current is 1.5mA. 
However, if I comment the setup rtc_gpio part, but keep the wifi part, the sleep current can keep in 4-5uA, vice versa if I comment the wifi part but just keep the rtc_gpio part, the current can keep 4-5uA as well. The wifi part is following exactly the same as IDF example, Am I setting wrong on the gpio?

void app_main() {
	ESP_ERROR_CHECK(nvs_init());
	gpio_init();
        //wifi init
	init_connect_wifi();
	delay_ms(2000);
	esp_wifi_stop();
	ESP_LOGI(TAG, "Going sleep");
        //setup rtc gpio
	ESP_ERROR_CHECK(rtc_gpio_init(GPIO_CONFIG_BTN ));
	ESP_ERROR_CHECK(rtc_gpio_set_direction(GPIO_CONFIG_BTN,RTC_GPIO_MODE_INPUT_ONLY));
	ESP_ERROR_CHECK(rtc_gpio_pullup_en(GPIO_CONFIG_BTN ));
	ESP_ERROR_CHECK(rtc_gpio_pulldown_dis(GPIO_CONFIG_BTN ));
	ESP_ERROR_CHECK(rtc_gpio_hold_dis(GPIO_CONFIG_BTN ));
	ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(GPIO_CONFIG_BTN , 0));
	delay_ms(2000);
	esp_deep_sleep(1000000LL * 180);
}

void gpio_init(void) {
	gpio_config_t io_conf;
	io_conf.intr_type = GPIO_PIN_INTR_NEGEDGE;
	io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
	io_conf.mode = GPIO_MODE_INPUT;
	io_conf.pull_down_en = 0;
	io_conf.pull_up_en = 1;
	gpio_config(&io_conf);
	gpio_set_intr_type(GPIO_CONFIG_BTN, GPIO_INTR_NEGEDGE);
        gpio_evt_queue = xQueueCreate(1, sizeof(uint32_t));
	xTaskCreate(btn_task, "btn_task", 2048, NULL, 10, NULL);
        gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
	gpio_isr_handler_add(GPIO_CONFIG_BTN, gpio_isr_handler,
			(void*) GPIO_CONFIG_BTN);
}

void init_connect_wifi(void) {
	tcpip_adapter_init();
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	wifi_config_t wifi_sta_config = { .sta = { .ssid = "xxxxxx", .password =
	"xxxxxx", }, };
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_sta_config));
	ESP_ERROR_CHECK(esp_wifi_start());
}

static esp_err_t event_handler(void *ctx, system_event_t *event) {

	switch (event->event_id) {
	case SYSTEM_EVENT_STA_START:
		esp_wifi_connect();
		break;

	case SYSTEM_EVENT_STA_CONNECTED:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_CONNECTED");
		break;

	case SYSTEM_EVENT_STA_GOT_IP:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
		ESP_LOGI(TAG, "got ip:%s\n",
				ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
		break;
	case SYSTEM_EVENT_STA_DISCONNECTED:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED");
		esp_wifi_connect();
		break;

	case SYSTEM_EVENT_STA_STOP:
		ESP_LOGI(TAG,"wifi stopped");
		break;
	default:
		break;
	}
	return ESP_OK;
}

denyip, 2018-08-31:
hi @igrr , any update on this?

StefanStsc, 2018-11-23:
I've exactly the same problem. The call to esp_sleep_enable_ext0_wakeup increases the deep sleep current by 1mA.

igrr, 2018-11-23:
As a workaround, you can call adc_power_off() function before entering deep sleep. We are working on a fix.

StefanStsc, 2018-11-23:
This actually works. What also works is to use ext1 instead of ext0.

Oliv4945, 2019-04-17:
@igrr , I confirm the workaround, thank you.
If it is not easy to fix in the code,  a comment in the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html) would be nice, I spend lot of time before discovering this issues :)

XxKavosxX, 2020-06-08:
Please, help-me.  I tried a lot of strategies using

esp_wifi_stop();
esp_wifi_deinit(); 

....but no one else worked. The current after use esp_wifi_start() function is aways 

68~79 mA in normal operation
68~79 mA in deep_sleep()

Wifi still ON!

Using deep_sleep() example (without wifi) the current is:

35mA in normal operation
10.5mA in deep_sleep

Thanks for your help!

I'm using esp-idf v4.0 and devBoard kit V1

AxelLin, 2022-02-18:
> As a workaround, you can call adc_power_off() function before entering deep sleep. We are working on a fix.

@igrr 
This issue is still open, so I'm wondering if the fix is available or not (since it has been years since your reply).

igrr, 2022-02-18:
I think this was fixed in Wi-Fi library at some point (i wasn't able to find the exact commit) which introduced another two issues: https://github.com/espressif/esp-idf/issues/1977 and https://github.com/espressif/esp-idf/issues/4585. Those issue were fixed before v4.3 release in 85ac572e, with backports in e7435a75, 2d7b596c, 65517844, a2142eae. It's probably safe to close this one.

closed 2022-02-18 by igrr