espressif/esp-idf #2316: How to send 9-bit data using spi？

qwert1213131, 2018-08-22:
I have a display and need to use 9-bit data. How do I set it up? Thank you!

ginkgm, 2018-08-27:
Hi @qwert1213131 ,
The data phase of ESP32 master is always in units of bytes (8bits), however, we have three phases whose length is configurable. They are (in order of their apperance):
CMD, ADDR, DUMMY(not available for full duplex mode)

We can flexibly use them to form a format of [D/CX] [CMD-8bits] [D/CX] [other data]
e.g.
cmd: 9 bits, the MSB to be the first DCX 
addr: 1 bit, the second DCX.

But if you need every single byte to be of 9-bit length, I'm afraid it's not available for ESP32.

qwert1213131, 2018-08-28:
Hi @ginkgm ，

Thank you very much for your reply. I want to be able to send 9 bits of data repeatedly, so referring to your suggestion, I use a 1-bit address and 8-bit data, but mcu starts to restart indefinitely. It seems that you have to consider another screen that supports 8-bit spi

ginkgm, 2018-08-28:
@qwert1213131 
the MCU refers to ESP32? are you using half duplex mode or full duplex mode?

Could you please try use 1-bit command and no address? maybe it's not valid to use address without command in half duplex mode?

qwert1213131, 2018-08-30:
@ginkgm 

The past should be a problem with the program that caused the reboot.
Using esp32
In full-duplex mode, both addresses and commands can work. Currently, it is [1-bit command or address] [8-bit data].
But the screen refresh is very slow, because every time you send it, you have to follow this format, only one byte and one byte.

I will try again if there is a suitable method.
Thanks very much！

ginkgm, 2018-08-30:
@qwert1213131 
I'm not quite sure if this works, but have you ever tried to send several bytes of data with only one D/CX bit?

qwert1213131, 2018-08-31:
@ginkgm 

Currently 1-bit command, one byte of data, is already working, but the frame refresh rate is very slow.
With the 1-bit command, there are still many problems with multiple bytes of data, which should not meet the operational requirements of the ST7789 spi-3 mode.
thank you very much.
I decided to change a spi screen, using the spi-4 mode like ili9341.

closed 2018-08-31 by FayeY