espressif/esp-idf #2317: Feature request: UART2 log output

Curclamas, 2018-08-22:
### User Story
As a developer I want to be able to choose all UARTs as log output.

### Requirements

- [ ] UART2 should be selectable for log output in menuconfig

### Background

For us as we build dataloggers which usese basically all UARTs with different RS458 drivers etc. this feature would be very helpful for us and increase our flexibility.

As of @igrr there are some reasons (https://github.com/espressif/esp-idf/pull/1217#issuecomment-341650197) which prevent this. It would be very nice if we could still use UART2 even with those limitations (you can add warnings regarding these limitations in menuconfig/docs). 
What do you think?

ammaree, 2018-08-22:
We have a similar requirement but would prefer to have the facility to completely retarget log output. Our situation is as follows:
Devices are deployed in remote environments and normally not accessed at all.
For diagnostics purposes we provide reasonably secure HTTPS and Telnet type access.
At the time of access it is important to be able to see the historic messages.
In order to achieve this we would like to redirect (retarget) all console character output to a buffer.
When a connection is established (HTTPS or Telnet) the historic messages in the buffer must immediately be displayed on the console, after which interactive session continues.

Currently we are partially achieving this by replacing the fprintf(stdout,...) functionality and intercepting the output for stdout/err, redirecting this to our buffer.
We have tried using dup2() or close/reopen() functionality but these are not supported.
We have write the normal _write() _read() mechanisms but those seem not to be supported.

A better option might be to simply provide a mechanism to transparently retarget all stdio functions, allowing the full esp_vfs_??() functionality to continue functioning?

Does this make sense?

Curclamas, 2018-08-22:
@ammaree basically something similar like you tried is already possible and [documented](https://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html#_CPPv219esp_log_set_vprintf14vprintf_like_t) for the IDF. 

My use case is just that I'd love to be able to use the UART2 with the standard `menuconfig` so I can avoid having to tinker with that in my application code. (Since @igrr already said it should be easily possible with some caveats)

igrr, 2018-08-22:
@ammaree Your request is different from support of UART2 for log output which is requested here. Please open a separate issue.

ammaree, 2018-08-22:
Correct, except all startup messages prior to this call would be missing, and often those are of importance. Similarly the messages from the wifi library follow a slightly different route through various lib_printf.c functions.

Retargeting at a character level specifically for stdin/out/err should  work everywhere and continue to allow vfs and higher level redirection.

AshUK, 2021-01-22:
+1 We requires this for the following setup

UART0 - Logging output
UART1 - GSM/Cellular Modem
UART2 - Modbus/RS485 Connection

We would like to switch logging output to UART2 reasoning being there is a restriction that when both UART1 and UART2 are receiving data. If there is a hardware overflow on either UART1 or UART2 both hardware FIFO buffers are cleared. As we are only using logging to output debugging info it nots critical to application run time should it TX/RX buffers get cleared.

See sections _14.3.3 UART RAM_ and  _Register 14.9: UART_CONF0_REG (0x20) - UART_RXFIFO_RST_ in the ESP32 technical documentation