espressif/esp-idf #2318: Feature request: stdio (in/out/err) retargeting

ammaree, 2018-08-22:
Our situation is as follows:

Devices are deployed in remote environments and normally not accessed at all. For diagnostics purposes we provide reasonably secure HTTPS and Telnet type access. At the time of access it is important to be able to see the historic messages.

In order to achieve this we would like to redirect (retarget) all console character output to a buffer. When a connection is established (HTTPS or Telnet) the historic messages in the buffer must immediately be displayed on the console, after which interactive session continues.

Currently we are partially achieving this by replacing the fprintf(stdout,...) functionality and intercepting the output for stdout/err, redirecting this to our buffer. We have tried using dup2() or close/reopen() functionality but these are not supported. We have write the normal _write() _read() mechanisms but those seem not to be supported.

We have tried using esp_log_set_vprintf() but all all startup messages prior to this call would be missing, and often those are of importance. Similarly the messages from the wifi library follow a slightly different route through various lib_printf.c functions.

A better option might be to simply provide a mechanism to transparently retarget all stdio functions, allowing the full esp_vfs_??() functionality to continue functioning?

Thanks

igrr, 2018-08-24:
Hi @ammaree,

Does

> We have tried using esp_log_set_vprintf() but all all startup messages prior to this call would be missing, and often those are of importance.

mean that you need a solution which would work at link time? If the solution works at link time, does it mean that the buffer for storing the messages would be statically allocated (as dynamic allocation is not available early in startup process)?

Also please clarify whether you looking to redirect messages from:

* bootloader
* interrupt handlers in the application

ammaree, 2018-08-25:
We would like the functionality to work at link time so all messages are captured hence buffer will be statically allocated. If possible would like to capture all messages including from bootloader and interrupt handlers.

Supporting the bootloader I would need some advice on since the bootloader would be running before the app has been loaded, hence the buffer would need to be known to and shared by bootloader and app.
 
Interrupt handler messages should not be a problem since the processing overhead to store the messages should be minimal

ammaree, 2018-08-27:
@igrr 

If any more information or feedback required, please let us know.
We are currently holding back on rolling out further OTA updates pending this request. so feedback appreciated.

Thanks

igrr, 2018-08-28:
Retargeting bootloader output  into a buffer and persisting this buffer until application startup might be tough, unless you override bootloader component and customize the linker script to reserve some fixed range of DRAM for your buffer. Then you will have to do the same for the app. Otherwise the buffer declared in the bootloader will likely end up in a completely different spot in memory than in the application. Another option is to use RTC memory as a buffer, if your app isn't using it for other purpose.

With regards to the interrupts, main issue here is not overhead but locking. Interrupts can happen on both cores, and the code which prints output into a buffer must not take locks (mutexes) while inside the interrupt. You can possibly use critical sections (portENTER_CRITICAL) but with any significant amount of logging, this is going to cause interrupts to be disabled for a nontrivial amount of time, possibly affecting your application performance. So if you do want to capture the logs coming from interrupts, you likely need to employ a two-stage approach. Use one buffer for collecting logs from interrupts only, and use a critical section to protect access to it. Then use the second (larger) buffer to collect logs from tasks. When writing to this buffer, check if there is any data in the interrupt log buffer, and memcpy it into the task log buffer if there is. This way you only enter critical sections while logging from interrupts (hopefully this is a rare situation) or when memcpy-ing between the two buffers (hopefully this happens fast).

For the actual overriding of console output, there are two places where output gets written to UART: in UART VFS and in ets_printf uart output callback. The former is responsible for all output sent to stdout and stderr, the latter prints in rare special cases (such as ets_printf invoked from ESP_EARLY_LOGx, panic handler, brownout detector interrupt, task watchdog interrupt, and so on).
To override output from UART VFS at compile time, you can modify `uart_write` function in vfs_uart.c. Overriding output from ets_printf at compile time is not possible, so you will need to install output handlers at run time, once in the bootloader and once in the app. In both cases this has to be done early at startup. You can call `ets_install_putc2` to set an additional callback for ets_printf (this function is defined in rom/ets_sys.h).

ammaree, 2018-08-28:
Hi @igrr 

Thanks for the feedback, much appreciated. An update on progress.
a) Got the redirection for stdout resolved, going to our buffer.
b) moved the buffer into RTC slow RAM, no problem

Since we do not use the ULP or any sleep mode We expected the RTC slow RAM to be 8KB in size as per the SOC_RTC_DATA_HIGH/LOW definitions in "soc.h" but the linker complained that the rtc_slow_seg was overflowing by 4096 bytes.

Upon checking in esp32.ld we saw that len is defined as 0x1000 less the ULP reserved memory.
Is this correct or should the size be 0x2000?

We also thought of usiung the RTC fast IRAM but since the TRM states this is only accessible from PRO CPU it would require us pinning a number of key tasks (HTTP, TelNET and CONTROL) to the same CPU. Any way around this restriction.

Once we have the above resolved we will look at the bootloader redirection, might have more questions then.

ammaree, 2018-09-06:
HI @igrr 

In Q2 above I have found the problem,

I am calling esp_timer_impl_get_time() before  esp_timer_impl_init() has been called, causing the system too hang.

Since I need the best/accurate runtime in uS from my "syslog" module, how can I check whether esp_timer_impl_init() has been run?

Q1 I still need clarity on.

Thanks

igrr, 2018-09-06:
The lowest overhead (although slightly low level) way of checking whether esp_timer is initialized would be:
```c++
if (REG_GET_BIT(FRC_TIMER_CTRL_REG(1), FRC_TIMER_ENABLE) != 0) {
    // can call esp_timer_get_time
}
```

With regards to Q1, the space starting from 0x1000 is reserved for ESP-IDF components (such as WiFi stack). At the moment it happens to be unused, but this might not be the case in the future.

ammaree, 2018-09-06:
@igrr 
Thanks for the feedback, test works like a charm. redirection of stdout/err to our syslog functionality and from there into the RTC slow RAM buffer is working perfectly.

For now we are overriding the 0x1000 in the esp32.ld with 0x2000 to get full use of the available buffer. Maybe you should consider adding an additional #define we can use that will automatically reduce the available space once you start using this SRAM for Wifi?

As far as the bootloader redirection is concerned, we will leave that till a later stage.

Will close this for now.

closed 2018-09-06 by ammaree

someburner, 2022-03-17:
@ammaree would you mind sharing how you did the bootloader logs save?

@igrr I am also trying to do this but not sure if my approach is right. I'm thinking of using a bootloader hook like this:

```c
RTC_NOINIT_ATTR int len = 0;
RTC_NOINIT_ATTR char rtc_buf[2048];

IRAM_ATTR void my_putc(char c)
{
    // esp_rom_uart_putc('a');
    esp_rom_uart_putc(c);
    if(rtc_len < 4096) {
        rtc_buf[rtc_len] = c;
        rtc_len++;
    }
}

void bootloader_after_init(void) {
    esp_rom_install_channel_putc(1, my_putc);

    ESP_LOGI("HOOK", "This hook is called AFTER bootloader initialization");
    ESP_LOGI("HOOK", "test = %d", rtc_len);
    ESP_LOGI("HOOK", "rtc_buf = %p", &rtc_buf);
}
```

This seems to sorta work, except I can't figure out a way to link `rtc_buf` in both the hook and my app_main. But manually printing from the pointer it works:

```c
    uint32_t addr = 0x3fff1b9c;
    char * ptr = (char *)addr;
    ESP_LOGI("rtc", "rtc_buf = %s", ptr);
```

Also I seem to be missing the logs after App CPU up but maybe my buffer isn't big enough.

ammaree, 2022-03-19:
@someburner 
The buffer we are using is the largest portion of the RTC slow RAM
Bootloader logs we only changed the macros to a slightly different format, was not able to save them to the buffer.
We ended up only doing the application logs which we intercept using our own log replacement component. Messages are  written to the RTC buffer and selectively, based on the log level, to a cloud host if WIFI+IP active. 

If no network link available messages are written to a FLASH based file and sent to host once connection has been established...