espressif/esp-idf #2320: Unable to get DHCP IP after OTA

vonnieda, 2018-08-22:
ESP-IDF: v3.0 (ee295c175b3d7ca0c387e950e473001d92b44764)
Module: WROVER

After performing an OTA update and calling esp_restart() the chip restarted, connected to WiFi AP but was not able to get a DHCP IP. I performed another esp_restart() and the problem persisted. Finally I power cycled the chip and it got an IP immediately, as normal.

During this issue, I noticed log messages I've never seen before:

```
I (4471) wpa: PTK has been installed, it may be an attack, ignor it.
I (4471) wpa: GTK has been installed, it may be an attack, ignor it.
I (5451) wpa: PTK has been installed, it may be an attack, ignor it.
I (5451) wpa: GTK has been installed, it may be an attack, ignor it.
```

sdkconfig attached:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2311747/sdkconfig.txt)

This issue sounds similar to what was happening in: https://github.com/espressif/esp-idf/issues/1273

Thanks,
Jason

negativekelvin, 2018-08-22:
It looks like false positive KRACK detection due to keys not cleared on restart and ignoring keys causing failed handshake

TimXia, 2018-08-23:
@vonnieda Could you please tell us the brand, firmware version and configurration of the WiFi AP? Did you use the OTA example?

vonnieda, 2018-08-23:
Hi @TimXia, the AP is a UniFi AP AC LITE. The firmware version is 3.4.14.3413. I don't see a way to export the device configuration, so if you have any specific questions about how it's configured, let me know and I will check. It's otherwise pretty basic WPA2 2.4GHz.

I'm not using the OTA example, but my code is quite similar to it. Here's my code with some private stuff removed. If you need the exact code please tell me an email address I can send it to.

```
static int firmware_download_callback(request_t *req, char *data, int len)
{
	esp_err_t err = esp_ota_write(context->ota_handle, data, len);
}

static void update(update_context_t* context) {
	// Find a partition to write the update to
	context->target_partition = esp_ota_get_next_update_partition(NULL);
	if (context->target_partition == NULL) {
	    return;
	}

	esp_err_t err;
	err = esp_ota_begin(context->target_partition, OTA_SIZE_UNKNOWN,
			&(context->ota_handle));
	if (err != ESP_OK) {
	    return;
	}

	// Download the firmware
	status_code = req_perform(req);

	// Finish the OTA process.
	err = esp_ota_end(context->ota_handle);
	if (err != ESP_OK) {
		esp_ota_end(context->ota_handle);
	    return;
	}

	// Everything looks good, so apply the update.
	err = esp_ota_set_boot_partition(context->target_partition);
	if (err != ESP_OK) {
	    return;
	}

	vTaskDelay(10000 / portTICK_PERIOD_MS);

	esp_restart();
}
```

zhangyanjiaoesp, 2018-09-14:
@vonnieda Can you provide the capture packets information? The packets should be the wireless packets, not the Ethernet packets captured by the Wireshark.

zhangyanjiaoesp, 2018-09-17:
@vonnieda you can ues `OmniPeek` to capture wireless packets

zhangyanjiaoesp, 2019-03-18:
@vonnieda Has this problem been solved?

vonnieda, 2019-03-18:
Unfortunately I don't have a good way to know. We've only seen this issue once in the field and have not been able to reproduce it.

Alvin1Zhang, 2019-03-18:
@vonnieda Feel free to reopen this if you run into the issue again. Thanks.

closed 2019-03-18 by Alvin1Zhang