espressif/esp-idf #2322: unable to set password in gatt_security_server example

Milan991, 2018-08-23:
Hi everyone,

I want to implement BLE security using password authentication. My device don't have Display or Keyboard and I want to connect to it using Android. My idea is to enter six digit key on Android side which I manually set in ESP firmware.

For start I'm using gatt_security_server example from ESP-IDF and I can't find the way to manually set BLE password, every time ESP generates random six digit password.

Is there a way to set manual password and to disable random password generation in gatt_security_server example?

Thank you all for advises in advance.

Weijian-Espressif, 2018-08-31:
@Milan991 I have added an API to set static passkey, it will be merged into IDF next week.

Milan991, 2018-08-31:
@Weijian-Espressif Great, thank you.

chegewara, 2018-09-11:
@Weijian-Espressif thanks for that feature

chegewara, 2018-09-13:
@Weijian-Espressif 
I have to add, probably its how it suppose to be, but this only works with 6 digit passkey.

Weijian-Espressif, 2018-09-14:
The spec define:
Numeric value (passkey) entered by user. Valid values are decimal 000000 - 999999.

for example:
uint16_t passkey = 123;
 esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint16_t));

for some phones , it is ok that you enter 123, but some other phones maybe need you enter 000123.

chegewara, 2018-09-14:
@Weijian-Espressif I did test and for me works only 0 or 000000 and all 6 digit values. Ive been trying 1234 and 12345 but could not pair/bond with nRF connect android smartphone.

I did not try 001234 on smartphone because with dynamic passkey when i have had 5 digit passkey it was enough to validate without leading 0. I will try with leading 0's and leave feedback.

PS i am using uint32_t instead, but it should not be an issue.

lettierimarco, 2018-11-07:
I 'm using the example server gatt security...with iphone and android with NRF app it connects without prompt passkey... already checked if i've the last commits

chegewara, 2018-11-07:
Im guessing you have this setting not changed:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L557

It needs to be changed to `ESP_IO_CAP_DISPLAY` if i remember.

lettierimarco, 2018-11-07:
do you mean ESP_IO_CAP_OUT?

chegewara, 2018-11-07:
Yes, thats the one.

lettierimarco, 2018-11-29:
but why static passkey changes?
i receive the passkey Notify number overwrite my passkey

lettierimarco, 2018-11-29:
i've done as you suggested:
```
	esp_ble_auth_req_t auth_req = ESP_LE_AUTH_BOND;     //bonding with peer device after authentication
	esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;           //set the IO capability to No output No input
	uint8_t key_size = 16;      //the key size should be 7~16 bytes
	uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	//set static passkey
	uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passKey, sizeof(uint32_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
```
but the passkey changes!

chegewara, 2018-11-29:
I dont know, i just copy pasted this part of code to my (Kolban's) library and it works like expected.