espressif/esp-idf #2323: Connect attempt only with first AP on network with multiple APs

mikaelkanstrup, 2018-08-23:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [-]
- Core (if using chip or module): [ESP32]
- IDF version: 30545f4
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [external 5V]

## Problem Description

When connecting to ESS networks, i.e. enterprise networks with multiple APs a connection attempt is only done towards the first found AP. I have tried both wifi_config.sta.scan_method:s (SCAN_FAST and SCAN_ALL) but both have the same problem.

When using the SCAN_ALL method all channels are scanned and the one AP with best RSSI is selected. If the connect attempt fails for any reason there's no new connection attempt towards the second, third, ... best APs. This is a problem as enterprise networks might block connection attempts towards a certain AP due to load balancing (or other reasons). In such cases ESP32 never succeed to connect.

### Expected Behavior
1. Scan all channels to find all APs/BSSes
2. Select the one with best RSSI and try to connect to it
3. If connection attempt fail, mark AP as temporarily unavailable and retry with next best AP
4. Repeat from 3 until all APs have been checked.

### Actual Behavior
1. Scan all channels to find all APs/BSSes
2. Select the one with best RSSI and try to connect to it
3. If connection attempt fail ESP32 is just stuck without wifi connection

### Steps to reproduce

1. Setup a simple test network with hostapd consisting of at least two APs/BSSes with same SSID
2. Make sure the APs have significantly different RSSI level. This can be done for example by lowering the tx power of one of the APs with:
`iwconfig txpower command`
3. Limit the number of STAs allowed to 0 on the AP with best RSSI. This can be done with `max_num_sta=0' in hostapd.conf file.
4. Modify the wpa2_enterprise example code setting scan_method to SCAN_ALL
5. Connect to the network using the wpa2_enterprise example code.
6. Notice that ESP32 never connects to AP with lower RSSI

### Code to reproduce this issue
esp-idf wpa2_enterprise example code with the only modification to set SCAN_ALL scan_method.

## Debug Logs

```
I (251) example: Setting WiFi configuration SSID Kanstrup...
I (261) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (331) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (331) wifi: mode : sta (30:ae:a4:56:f1:c8)
I (2751) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
I (3731) wifi: state: init -> auth (b0)
I (3731) wifi: state: auth -> init (1100)
I (3731) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1
I (4341) example: ~~~~~~~~~~~
I (4341) example: IP:0.0.0.0
I (4341) example: MASK:0.0.0.0
I (4341) example: GW:0.0.0.0
I (4341) example: ~~~~~~~~~~~
I (6341) example: ~~~~~~~~~~~
I (6341) example: IP:0.0.0.0
I (6341) example: MASK:0.0.0.0
I (6341) example: GW:0.0.0.0
I (6341) example: ~~~~~~~~~~~
I (8341) example: ~~~~~~~~~~~
I (8341) example: IP:0.0.0.0
I (8341) example: MASK:0.0.0.0
I (8341) example: GW:0.0.0.0
I (8341) example: ~~~~~~~~~~~
I (10341) example: ~~~~~~~~~~~
I (10341) example: IP:0.0.0.0
I (10341) example: MASK:0.0.0.0
I (10341) example: GW:0.0.0.0
I (10341) example: ~~~~~~~~~~~
I (12341) example: ~~~~~~~~~~~
I (12341) example: IP:0.0.0.0
I (12341) example: MASK:0.0.0.0
I (12341) example: GW:0.0.0.0
I (12341) example: ~~~~~~~~~~~
I (14341) example: ~~~~~~~~~~~
I (14341) example: IP:0.0.0.0
I (14341) example: MASK:0.0.0.0
I (14341) example: GW:0.0.0.0
I (14341) example: ~~~~~~~~~~~
I (16341) example: ~~~~~~~~~~~
I (16341) example: IP:0.0.0.0
I (16341) example: MASK:0.0.0.0
I (16341) example: GW:0.0.0.0
I (16341) example: ~~~~~~~~~~~
I (18341) example: ~~~~~~~~~~~
I (18341) example: IP:0.0.0.0
I (18341) example: MASK:0.0.0.0
I (18341) example: GW:0.0.0.0
```

## Other items if possible

Air sniffer showing packets from/to the two APs and ESP32.

TimXia, 2018-08-27:
@mikaelkanstrup If first connection fails, you can call esp_wifi_connect to connect again.

mikaelkanstrup, 2018-08-28:
@TimXia Well the example code already calls esp_wifi_connect on SYSTEM_EVENT_STA_DISCONNECTED so the problem is not really about a missing call to the connect function. I tried adding this block of code to the wpa2_enterprise_example_task just see if it had any effect:

`if (xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, 0, 1, 1) != CONNECTED_BIT) {
    ESP_LOGI(TAG, "DISCONNECTED. Calling esp_wifi_connect again");
    esp_wifi_connect();
}`

But the behavior is still the same. Even rebooting the system won't solve this scenario. At each startup the same BSS will be selected as "best" AP, then fail and never try any of the other BSSes of the ESS network.

I guess one could implement the mechanism manually in the application code by scanning all channels, keeping a list of BSSes, call connect with bssid and channel set in the wifi_sta_config and repeat until all BSSes have been checked but for sure this feels like something the core implementation should handle instead.

As you are already familiar with wpa_supplicant (and integrate parts of the implementation) you could have a look at the blacklisting and perhaps the background scanning mechanism to see if something similar would be usable for ESP-IDF.

XinDeng11, 2018-08-30:
@mikaelkanstrup same reason I have explained in the https://github.com/espressif/esp-idf/issues/2324

mikaelkanstrup, 2018-08-30:
@XinDeng11 I'm not so sure this same root cause. For this case the EAP procedure has not even started.

After calling esp_wifi_connect ESP-IDF internal code scans all channels and selects the best BSS to try connecting to. In this scenario this attempt fails already after the initial authentication frames where AP respond with an authentication frame with status code "Association defined because AP is unable to handle additional STAs (0x0011)".

After connect attempt fails with first BSS I would expect ESP-IDF internal code to go back to its list of valid BSSes and select the BSS with second best RSSI and make another connect attempt. If again this fails continue until all valid BSSes have been tried. If all fail then send a DISCONNECTED event to inform application code.

As it is right now there's no internal retry mechanism and there's no event sent to application code either.

XinDeng11, 2018-09-18:
hi @mikaelkanstrup sorry for later reply, I think why cause that happen is because some of APs are in the same channel, we have found the root cause, and will merge it later, if you are urgent, I will give you a lib to verify it before we merge the code

XinDeng11, 2018-09-19:
@mikaelkanstrup Please use the lib in the attachment, thanks
[2018-09-19_155241.zip](https://github.com/espressif/esp-idf/files/2396004/2018-09-19_155241.zip)

XinDeng11, 2018-09-21:
@mikaelkanstrup How about the problem, is the patch can solve the problem?

mikaelkanstrup, 2018-09-21:
@XinDeng11 Sorry for the delay. I've been testing with the libs provided in attachement 2018-09-19_155241.zip and unfortunately the problem is not solved.

I notice the following behavior:

**Network with two APs:**
When STA is rejected by first AP no SYSTEM_EVENT_STA_DISCONNECTED is sent to application and STA never tries to connect to second AP

**Network with only one AP:**
When STA is rejected by AP a SYSTEM_EVENT_STA_DISCONNECTED is sent to application. The         esp_wifi_connect call found in example code event_handler function trigger new connect attempts.

So there's something stopping ESP-IDF internals from sending the SYSTEM_EVENT_STA_DISCONNECTED event when more than one AP present on the network. I first tried having the two APs on same channel and then tried having them on different channels but behavior was same.

XinDeng11, 2018-09-21:
@mikaelkanstrup can you send me the log?

XinDeng11, 2018-09-21:
it should be work, because we have tried it, and we can see it connect to the second one, when they are in different channels, it need set use the all_channel_scan in scan_method

mikaelkanstrup, 2018-09-21:
@XinDeng11 I was not able to capture an air sniffer log this time and from esp32 logs there's nothing of interest. It just keeps printing the status logs coming from example_task loop.

It's great that you're confident that you've found the root cause. I might have done something wrong when trying to use the libs you sent. I put them into components/esp32/libs folder, did a make clean before building the example code again so was pretty sure I used the provided libs. And I did set WIFI_ALL_CHANNEL_SCAN so that's not the problem. 

But anyway I will unfortunately not have time to test further until after the weekend so feel free to continue the steps towards having your fix integrated on master. On Monday I'll either make another attempt with the libs or if fix is already integrated I'll use that instead. Will make sure to capture and send you all the logs.

XinDeng11, 2018-09-25:
Hi@mikaelkanstrup, actullay the status code is 17, it is mean the connected station number has reached the limitation,  but it still a problem of us, we lost the case may happen in a same ssid enviroment, may be you can try with the lib in the attachment, thanks
[2018-09-25_204842.tar.gz](https://github.com/espressif/esp-idf/files/2415356/2018-09-25_204842.tar.gz)

mikaelkanstrup, 2018-09-25:
@XinDeng11 The fix you've done in the latest libs attached solves this issue. ESP now successfully retries with the second AP if first one fails. Thanks a lot for your time spent investigating and solving this problem!

XinDeng11, 2018-09-26:
@mikaelkanstrup OK， got it. We will merge the fix to master later, after it merge, we will let you know

closed 2018-10-08 by igrr