espressif/esp-idf #2325: esp_err.h: add a macro to check a condition and print an error

nicola-lunghi, 2018-08-23:
in esp_err.h is defined the ESP_ERROR_CHECK macro like this

https://github.com/espressif/esp-idf/blob/020ade652d9b6b0b87b0bebe176a150cc4407749/components/esp32/include/esp_err.h#L106-L112

https://github.com/espressif/esp-idf/blob/020ade652d9b6b0b87b0bebe176a150cc4407749/components/esp32/panic.c#L655-L666

could be useful to have also a macro that check return value or a condition, print an error and 
return the value

this is defined multiple time in the codebase

https://github.com/espressif/esp-idf/blob/020ade652d9b6b0b87b0bebe176a150cc4407749/components/driver/i2c.c#L34-L37

https://github.com/espressif/esp-idf/blob/020ade652d9b6b0b87b0bebe176a150cc4407749/components/driver/i2s.c#L41-L44

https://github.com/espressif/esp-idf/blob/020ade652d9b6b0b87b0bebe176a150cc4407749/components/driver/can.c#L31-L41

Or something like that.. (we need a function that print an error without panic :-))

```
#define ESP_ERROR_CHECK_AND_RETURN(x) do {                                         \
        esp_err_t __err_rc = (x);                                       \
        if (__err_rc != ESP_OK) {                                       \
            _print_failed_error(__err_rc, __FILE__, __LINE__,       \
                                    __ASSERT_FUNC, #x);                 \
           return __err_rc;
        }                                                               \
    } while(0);
#endif
```

nicola-lunghi, 2018-08-23:
also noticed now that after the while(0) there shouldnt be a semicolon
```
  } while(0)
```

dobairoland, 2018-10-10:
Hello @nicola-lunghi! Thank you for your suggestion and pointing out the mistake with `while`.

I implemented a macro called `ESP_ERROR_CHECK_WITHOUT_ABORT` which can be used also as
`esp_err_t rc = ESP_ERROR_CHECK_WITHOUT_ABORT( some_function() );`. This macro uses GCC macro expression. This issue will be closed after my commit is pushed to master.

The removal of semicolon after `while(0)` will take place later (in the next major release) because it can broke third-party libraries if `ESP_ERROR_CHECK` is used without semicolon. My commit corrects those issues in IDF.

Please note that I didn't implemented a macro with `return` as you suggest. The reason is that it is almost impossible to do that in C in a general way which will be clear and useful for all developers. There is `NDEBUG`, `CONFIG_OPTIMIZATION_ASSERTIONS_SILENT` and maybe other macros which should affect the naming and behavior. It is not clear when the return should be performed on those silent and semi-silent occasions. However, if you have a suitable proposal for this, please feel free to submit a pull request.

closed 2018-10-18 by igrr

dobairoland, 2018-10-25:
@nicola-lunghi Do you agree with this solution?

nicola-lunghi, 2018-10-25:
Yes! sorry about the delay

Thanks
Nick