espressif/esp-idf #2329: memory leak in wifi_init and wifi deinit sequence

sjoteppagol, 2018-08-24:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [* ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ *] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ *] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //94ec3c8e

- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [|external 5V]

## Problem Description

//Detailed problem description goes here.
I am facing memory leak problem in esp32 while working with wifi, after performing ESP_ERROR_CHECK( esp_wifi_stop()); and 
ESP_ERROR_CHECK( esp_wifi_deinit()); the device is not completely releasing heap. 
As you can observe in the attached log file,  Heap size ENDING OF THE CALL at the starting is **Heap size ENDING OF THE CALL  84180** where as after running for some time it is reduced to Heap size ENDING OF THE CALL  78232. 
On each iteration device is losing ~ **270 bytes.** 

### Expected Behavior
The heap should not decrease.
### Actual Behavior
Device is losing 270 bytes on each iterations.

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
/******************************************************************************
* Function : aws_shadow_disconnect
* Description : Code  to handle the  WiFi Connect and AWS Shadow 
*******************************************************************************/
```cpp
AWS_TASK_HANLER_STATE_CODE aws_shadow_disconnect(void)
{
    IoT_Error_t rc = FAILURE,rc1 = SUCCESS;
    init_done = 0;
    rc = aws_iot_shadow_disconnect(&mqttClient);
    aws_iot_error_log(rc);  //aws_error_log
    if(SUCCESS != rc)
    {
        ESP_LOGE(TAG, "Disconnect error %d", rc);
        //return AWS_SHADOW_DISCONNECT;           /* Function return aws_sahdow Disconnect STATE  */
    }
    ESP_LOGI(TAG, "vEventGroupDelete");
    vEventGroupDelete(wifi_event_group);
    ESP_LOGI(TAG, "esp_wifi_disconnect");
    ESP_ERROR_CHECK( esp_wifi_disconnect()); 
    ESP_LOGI(TAG, "esp_wifi_stop");
    ESP_ERROR_CHECK( esp_wifi_stop());
    ESP_LOGI(TAG, "esp_wifi_deinit");
    ESP_ERROR_CHECK( esp_wifi_deinit());
    /* Need delay before the de-init happens */
[... 2272 lines cut ...]
E (484062) spi_slave: spi_slave_initialize(81): host already in use
I (484062) APP_DEBUG: TLC RESET, Wrong SPI_SYNC_CODE
I (484072) APP_DEBUG: System Power Up, Days: 1 and Time: 12:08:00
I (484072) APP_DEBUG: Heap size ENDING OF THE CALL  71604

I (485052) APP_DEBUG: System Power Up, Days: 1 and Time: 12:08:01
I (485062) APP_DEBUG: Heap size ENDING OF THE CALL  71604

E (485452) aws_iot: failed! mbedtls_net_connect returned -0x52
I (485452) IOT_AWS_ERROR:  server is unknown ?? 

E (485452) APP_DEBUG: aws_iot_shadow_connect returned error -23
E (485452) APP_DEBUG: aws_iot_shadow_init returned error -23
I (486052) APP_DEBUG: System Power Up, Days: 1 and Time: 12:08:02
I (486062) APP_DEBUG: Heap size ENDING OF THE CALL  77976

I (486462) APP_DEBUG: AWS_TASK aws_shadow_disconnect(); !!!! : 

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2018-08-24:
https://github.com/espressif/esp-idf/issues/2311

sjoteppagol, 2018-09-03:
Hello negativekelvin,
I have gone through your provided link, in the sdkconfig max number of open sockets are set to 10.
Which is better way to avoid these kind of memory leaks !? Whether to give some delay before trying to re-connect or reducing max number of open sockets.?

Alvin1Zhang, 2019-03-15:
@sjoteppagol Hi, does your issue still exist? Thanks.

sjoteppagol, 2019-03-20:
@Alvin1Zhang  I haven't tested it in recent ESP-IDF, but now we are not repeatedly calling esp_wifi_stop function. Instead of that first we are making sure that we have proper internet connection by pinging to any website. If yes then only we are calling esp wifi stop function. 

Thank you
 Shridhar

jack0c, 2022-01-05:
This should be fixed in version after v3.3, If you still find the issue, please reopen it.

closed 2022-01-05 by jack0c