espressif/esp-idf #2330: crypto/base64.h not guarded by extern C

nkolban, 2018-08-25:
Am working on a C++ project and tried to use base64_encode() as defined here:

https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/include/crypto/base64.h

However, it does not have the:

```
extern "C" {
}
```

guard and hence results in base64_encode not found at link time.  This issue asks for the inclusion of C++ guards in the header file.

projectgus, 2018-08-28:
Hi @nkolban,

We don't really consider wpa_supplicant's internal crypto functions as part of IDF's public API. We haven't stopped anyone from using them, but they could change or be removed at some point in the future.

I would recommend `#include "mbedtls/base64.h"` and then use mbedtls_base64_encode/mbedtls_base64_decode.

Will keep this issue open, but probably the resolution will be to move the non-public wpa_supplicant headers into a private header directory somewhere.

nkolban, 2018-08-28:
Thank you sir ... I tried to follow my thinking back in case we had an opportunity to clarify something further.   I needed base64 function and found the following post in the forum:

https://www.esp32.com/viewtopic.php?t=2461

From a summary read, I heard that base64 functions were implemented in ROM.  From there I looked for the signature definitions of the ROM functions.

That said, I'll change my logic to use mbed and also make note of those in my book of notes for the future.  Thank you sir.   I am more than satisfied.  Feel free to close the issue as desired from my perspective.

X-Ryl669, 2019-01-14:
For other finding this issue by searching, please also notice that wpa_supplicant's base64 implementation is wrong. It inserts '\n' in the middle of the base64 encoded stream to limit the lines to 73 chars. It's not following RFC requirement here.

`mbedtls_base64_encode` is correct however.

projectgus, 2019-07-31:
This is fixed as of c139683024 this header is no longer publicly visible to other components.

Solution, as noted elsewhere in the thread, is to `#include mbedtls/base64.h` and call `mbedtls_base64_encode()`.

closed 2019-07-31 by projectgus