espressif/esp-idf #2331: ethernet: infinite loop if phy cannot be initialized

loboris, 2018-08-25:
Both [phy_lan8720_init()](https://github.com/espressif/esp-idf/blob/master/components/ethernet/eth_phy/phy_lan8720.c#L91) and [phy_tlk110_init()](https://github.com/espressif/esp-idf/blob/master/components/ethernet/eth_phy/phy_tlk110.c#L97) calls [esp_eth_smi_wait_value()](https://github.com/espressif/esp-idf/blob/master/components/ethernet/emac_main.c#L243) in an infinite loop which, in case phy cannot be initialized for some reason, blocks the whole system, eventually causing task watchdog timeout.

I think there should some kind of timeout, the function should exit with failed status which should propagate to `esp_eth_enable()` function, returning `ESP_FAIL`.

```
...
E (784552) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (785552) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (786552) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - mp_task (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: IDLE
E (787552) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (788552) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (789552) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
...
```

negativekelvin, 2018-08-25:
Linking with https://github.com/espressif/esp-idf/issues/2141

loboris, 2018-08-26:
Thank you for the link, I hope it will be addressed in the esp-idf.

loboris, 2018-08-27:
The issue is slightly different than the linked one as the blocking occurs on different place in code.

It looks like there are infinite loops in the ethernet componnent in several places.
This is realy a bad behaviour, any function should fail gracefully in all possible scenarios instead of blocking (or resetting) the whole system.
The ethernet component developers should realy look at the code and fix this.

negativekelvin, 2018-08-27:
Yes I know, just suggesting they should be fixed together. Some of the while loops for the internal Mac registers might be ok although maybe not if something kills the peripheral clock.

GOST-UA, 2018-09-12:
I also had same situation ones but bug gone after turn off/on power and never happened again

suda-morris, 2018-09-19:
@loboris Thanks for your advice. The infinite loop problem will be fixed in the newest master branch later.

closed 2018-09-24 by igrr

Mikegyver, 2019-08-27:
hi everyone. i'm facing the same issue but with different ethernet shield.

FYI, i'm using ENC28J60 and following are the error appear on arduino serial monitor.

`E (1033) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000
E (2033) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0x0000
E (2033) emac: Initialise PHY device Timeout`

i did the test using example code in the library from http://www.fisch.lu/junk/ESP32-Ethernet.zip

hope the details above will provide enough information regarding this issue. regards.