espressif/esp-idf #2332: Issue with running gdb on ESP32 Wrover Dev Kit

RA5040, 2018-08-25:
## Environment

- Development Kit:   ESP32-Wrover-Kit

- esp-idf-v3.0.3

 - openocd-esp32-win32-0.10.0-esp32-20180724

- Development Env:		Command line

- Operating System:	Windows

## Problem Description

It appears that to run gdb on the ESP32 DevKit C, it's necessary to do things in this order:

1. Make and flash the program
2. Run OpenOCD
3. Run xtensa-esp32-elf-gdb

If OpenOCD is running before flashing the program nothing works.

The Espressif documentation says that OpenOCD should be running, then the program should be flashed, then gdb should be run: https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-command-line

So there's either a bug in the code or a bug in the documentation

### Expected Behavior

OpenOCD should be running before flashing and starting gdb

### Actual Behavior

If OpenOCD is running the gdb fails

### Steps to reproduce

Using a ESP32WroverKit with built-in JTAG, follow the Espressif instructions to debug with JTAG in Windows to the letter, build and flash one of the IDF examples (for example blink), and you should see that dbg does not work.

Flash the program, then run OpenOCD, then run the xtensa gdb and hardware debugging should work.

Here is the gdb log:
===============
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-host_pc-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
0x400d1ffa in app_main () at C:/msys32/home/Robert/esp/blink/main/blink.c:45
45 vTaskDelay(1000);
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active) APP_CPU: PC=0x00000000
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active) APP_CPU: PC=0x40000400
Hardware assisted breakpoint 1 at 0x400d1fdb: file C:/msys32/home/Robert/esp/blink/main/blink.c, line 43.
Target halted. PRO_CPU: PC=0x400D1FDB (active) APP_CPU: PC=0x400D1816
[... 129 lines cut ...]
32	/* Blink off (output low) */
33	gpio_set_level(BLINK_GPIO, 0);
34	vTaskDelay(1000 / portTICK_PERIOD_MS);
35	/* Blink on (output high) */
36	gpio_set_level(BLINK_GPIO, 1);
37	vTaskDelay(1000 / portTICK_PERIOD_MS);
38	}
39	}
40	
41	void app_main()
42	{
43	xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
44	while(true){
45	vTaskDelay(1000);
46	}
47	}

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2320892/sdkconfig.zip)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[blink.zip](https://github.com/espressif/esp-idf/files/2320891/blink.zip)

RA5040, 2018-08-26:
I was asked to run this again with -d3 on openocd. I attach the log.
[OpenOCDLog.zip](https://github.com/espressif/esp-idf/files/2321462/OpenOCDLog.zip)

igrr, 2018-09-05:
This issue was moved to espressif/openocd-esp32#58

closed 2018-09-05 by igrr